pulp-platform/pulpino #92: Compiling AES in C

federicoVS, 2017-07-14:
Hi I am kind of a newbie and I need PULPino to run an aes program written in C (from https://github.com/kokke/tiny-AES128-C). When I run it using `make tiny-AES128-C.vsim` is gives me the following errors:

```
../../libs/Arduino_lib/core_libs/libArduino_core.a(main.cpp.o): In function `initVariant':
/pulpino/sw/libs/Arduino_lib/core_libs/inc/main.cpp:36: undefined reference to `setup'
/pulpino/sw/libs/Arduino_lib/core_libs/inc/main.cpp:37: undefined reference to `loop'
/pulpino/sw/libs/Arduino_lib/core_libs/inc/main.cpp:40: undefined reference to `loop'
collect2: error: ld returned 1 exit status
make[3]: *** [apps/tiny-AES128-C/tiny-AES128-C.elf] Error 1
make[2]: *** [apps/tiny-AES128-C/CMakeFiles/tiny-AES128-C.elf.dir/all] Error 2
make[1]: *** [apps/tiny-AES128-C/CMakeFiles/tiny-AES128-C.vsim.dir/rule] Error 2
make: *** [tiny-AES128-C.vsim] Error 2
```
If I compile it normally (using gcc) it works.

Any idea how can I fix this?

be4web, 2017-08-20:
That is difficult to say from your output, are you sure you included all necessary files and headers in your CMakeLists?

closed 2017-09-14 by FrancescoConti

Nagatejapallapu, 2021-08-12:
Did you find the solution for the "undefined reference" error. If yes, Please help. Thanks