pulp-platform/pulpino #94: make all fpga

federicoVS, 2017-07-18:
I run `make all` in the fpga folder (while in `tcsh`) and I get the following errors:

```
make -C ips/xilinx_clock_manager/ clean all
make[1]: Entering directory `/home/user/dir/pulpino/fpga/ips/xilinx_clock_manager'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_clock_manager.xpr
rm -rf xilinx_clock_manager.hw
rm -rf xilinx_clock_manager.data
rm -rf xilinx_clock_manager.runs
rm -rf xilinx_clock_manager.srcs
rm -rf xilinx_clock_manager.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
rm -rf .Xil
vivado -mode batch -source tcl/run.tcl
make[1]: vivado: Command not found
make[1]: *** [all] Error 127
make[1]: Leaving directory `/home/user/dir/pulpino/fpga/ips/xilinx_clock_manager'
make: *** [ips/xilinx_clock_manager/ip/xilinx_clock_manager.dcp] Error 2
```
Can someone help me out?

darketik, 2017-07-18:
If your Xilinx vivado tool is installed in "/opt/Xilinx/Vivado/2016.4" for example. 
You have to run this command in your shell before executing make commands:
source /opt/Xilinx/Vivado/2016.4/settings64.csh
This will setup Vivado environnement.

gautschimi, 2017-07-18:
you have to add the vivado installation directory to your path. it 
doesnt recognize the command "vivado"

closed 2017-08-11 by FrancescoConti