pulp-platform/pulpino #97: SPI transfer timed out

AmirAliEbrahimi, 2017-07-27:
Hello,
I'm trying to run PULPino on MYIR Z-turn Board (it's a xc7z020clg400-1 Zynq evaluation board),The problem is when I run `./spiload helloworld.stim.txt` , I get this error:
```sh
Device has been reset
spidev spi32766.0: SPI transfer timed out
Write Error: Connection timed out 
Segmentation fault
```
BTW I'm change the devicetree , the original one makes the kernel hangs at `Starting kernel ...`
Any idea how can I fix this?
Thanks

gautschimi, 2017-07-28:
Hm looks strange. I'd try to run it again and specify a timeout:

./spiload helloworld.stim.txt --timeout=30

(for a 30 seconds timeout)

but I'm not sure it helps sorry. I'd also check for error messages and 
warnings in the log files when you created the bitstream but also the 
kernel etc. Maybe you get some hint on what went wrong.

best,
Michael

wang1991jun, 2017-08-10:
i want to use SPI.but i don't how to play.it diffrent with before i work.youcan tell me how to init

closed 2017-09-14 by FrancescoConti