pulp-platform/pulpino #98: section '.text' does not fit in region 'instrram'

ChristianPalmiero, 2017-08-01:
Hi everyone!
I am trying to run the following application on the RI5CY-based PULPino platform.

`#include  <stdio.h>
#include  <string.h>
#include  <stdlib.h>

int main (int argc, char **argv)
{
  char buf[100];
  
  snprintf(buf, sizeof(buf), "Foo %x %x");
  buf[sizeof(buf)-1] = 0;
  printf("Buffer size is: (%d)\nData input: %s\n", strlen(buf), buf) ;
	
  return 0;
}`

I am using the snprintf function in a wrong way because I would like to perform a format string attack (I am working on a security-based project).

When I try to execute

`make applicationName`

I get the following error

> /stuff/ri5cy_gnu_toolchain/install/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: format_string_1v.elf section `.text' will not fit in region `instrram'
/stuff/ri5cy_gnu_toolchain/install/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: region `instrram' overflowed by 10192 bytes
collect2: error: ld returned 1 exit status

How can I force the linker to allocate a larger instruction ram section?

FrancescoConti, 2017-08-11:
Hi Federico, you can hack the linker script in sw/ref/link.common.ld - but that won't do you much good if you don't increase also the physical size of the instruction memory (32 kB if I'm not wrong).

ChristianPalmiero, 2017-08-11:
Thank you. I will find a way to work around this, I am not even sure that the sscanf function is fully supported in PULPino. 
Btw, I am going through the qprintf.c library file (sw/libs/string_lib/src) in order to study the PULPino implementation of the prinft function. The %n parameter (number of bytes written so far) is not supported, is it?

be4web, 2017-08-20:
The printf is a very basic implementation - no `%n` is not supported.

ellenFan, 2018-06-06:
@ChristianPalmiero Hey, I run into the same problem (instrRAM overflow while running mi testbench). Have you managed to solve this problem? Could I share your migration benchmark? 

Thank you

ChristianPalmiero, 2018-06-07:
@ellenFan No, I haven't manage to solve the issue :( I think there's no way to work it out

FrancescoConti, 2018-06-07:
@ellenFan @ChristianPalmiero you can increase the physical instruction memory by overriding the value of the `INSTR_RAM_SIZE` (in bytes) in the `core_region` instantiation https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/rtl/pulpino_top.sv#L168 .

Then you have to modify the linker script accordingly -- as suggested in my initial response.

ellenFan, 2018-06-07:
@FrancescoConti Hi, I have already increased the physical instruction memory and now is trying to hack the  linker script. According to Pulpino Datasheet, Instruction memory takes on address space from 0x0 to 0x00008000, with BOOTROM following from 0x0008 0000 to 0x0008 0200. Let's say our physical instruction memory is increased to 686280 bytes (670 megabytes). Should I move the address space assigned to BOOTROM as well as DATA Memory forward to make room for the new large instruction memory? What if the address space is not large enough for the new large instruction memory? 

Thank you so much and I would really appreciate your response.

ellenFan, 2018-06-07:
[image]
May I use the address space between Boot ROM and DATA Memory (from 0x008 0200 to 0x0010 0000)?There are similar address spaces not filled in on the graph above, what are they reserved for?

Thank you so much

ellenFan, 2018-06-07:
@ChristianPalmiero @FrancescoConti What if we increase our instrram's size to order of megabytes that could probably exceed the address space provided?

FrancescoConti, 2018-06-08:
Hi @ellenFan . In general, you want to use the available memory space in a well-aligned manner, so that you can decode addresses with the minimum amount of bits in interconnects and demuxes, minimizing the complexity of this blocks. For example, in PULPino you could identify the instruction memory by checking only three bits, `addr[31]==1'b0`, `addr[20]==1'b0`, `addr[19]==1'b0`. The space you propose to reuse cannot be addressed in this way, so I think this is not a good idea.

Given the memory map of PULPino, you should be able to increase the instruction memory to 512kB easily (mapped from `0x0000_0000` to `0x0008_0000`). It might even work without any further hardware change (beyond increasing the physical size of the memory).

If you really want/need to change the memory map e.g. to move memory spaces around, you need to change more than just the linker script. Have a look at the interconnect diagram:
[image]
The memory map is defined essentially 1) by the AXI interconnect see (https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/rtl/pulpino_top.sv#L314), 2) by the core instruction demux (here https://github.com/pulp-platform/pulpino/blob/master/rtl/instr_ram_wrap.sv). This one currently works in the following way: if the instruction RAM occupies N bits in your address space, the ROM will be mapped at addresses starting with the bit N+1 = 1. This is hardwired in the way the demux works in that specific instance.