pulp-platform/pulpino #101: Out-of-bounds memory access in sha test

niosHD, 2017-08-12:
The [data array](https://github.com/pulp-platform/pulpino/blob/master/sw/apps/bench/sha/sha_test.c#L14) in the sha test (i.e., `sw/apps/bench/sha`) has a size of 256 bytes. However, [during hash computation](https://github.com/pulp-platform/pulpino/blob/master/sw/apps/bench/sha/sha_test.c#L26) up to 16 additional bytes get processed because an offset is added to the base pointer in the loop without adopting the corresponding length.

Regards,
Mario

be4web, 2017-08-20:
Thanks for the pointer. Would you consider to submit a PR to fix this issue? Thanks

niosHD, 2017-08-23:
Sure, I can look into it. However, I doubt that I have time to properly fix the issue within the next weeks.

FrancescoConti, 2018-01-20:
Fixed as of https://github.com/pulp-platform/pulpino/tree/6370c5f

closed 2018-01-20 by FrancescoConti