pulp-platform/pulpino #106: Peripherals and memory

fathomtreelimited, 2017-08-20:
Hi!

The description and diagrams suggest you have i2s on the APB bus.  Do you have any specifications for this audio interface?

For memory support please take a look at the octal SPI (166Mhz) solution from Cypress called HyperBus.  They have 64Mbit ram and 512Mbit flash together in a tiny FPGA package.  Bus speed is 333MB/sec.

http://www.mouser.com/ds/2/100/S71KL256SC0_S71KL512SC0,%20HYPERFLASH%20AND%20HYPERRAM%20M-1082255.pdf

Royalty free IP available.

http://www.cypress.com/documentation/software-and-drivers/hyperbus-master-interface-controller-ip-intellectual-property

Best regards,
David.

be4web, 2017-08-20:
Thanks for the pointer - we are aware of Cypress's HyperRam which is indeed interesting for us. Regarding I2S, we do have an I2S peripheral, but I do not think that we have open-sourced it yet. So I think the data on the website might be a little unaligned with reality. Sorry for that.

closed 2017-08-20 by be4web

fathomtreelimited, 2017-08-20:
Ok, thanks.  HyperBus and the Macronix parts are great little parts. Do you have any documentation you can share about the i2s peripheral?