pulp-platform/pulpino #110: unable to run vsim simulation

ashan8k, 2017-08-28:
ashan@e420:~/git/pulp/pulpino/sw/build$ make helloworld.vsimc
[  0%] Built target bench
[  0%] Built target crt0
[  0%] Built target string
[ 25%] Built target sys
[ 25%] Built target Arduino_core
[ 50%] Built target Arduino_separate
[ 50%] Built target helloworld.elf
[ 50%] Generating vectors/stim.txt
[ 50%] Built target helloworld.stim.txt
[ 75%] Built target helloworld.links
[ 75%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
Error loading design
apps/helloworld/CMakeFiles/helloworld.vsimc.dir/build.make:57: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsimc' failed
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc] Error 12
CMakeFiles/Makefile2:949: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all] Error 2
CMakeFiles/Makefile2:956: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
Makefile:376: recipe for target 'helloworld.vsimc' failed
make: *** [helloworld.vsimc] Error 2

FrancescoConti, 2017-08-28:
Can you run this with the command `make helloworld.vsimc VERBOSE=1` and copy-paste here the tail of your log?

ashan8k, 2017-08-28:
Hi Francesco,
Thanks for quick response. here is the verbose output.
--------------------------------------------------------------------------------------------------------------------------------------
ashan@e420:~/git/pulp/pulpino/sw/build$ make helloworld.vsimc VERBOSE=1
/usr/bin/cmake -H/home/ashan/git/pulp/pulpino/sw -B/home/ashan/git/pulp/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 helloworld.vsimc
make[1]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
/usr/bin/cmake -H/home/ashan/git/pulp/pulpino/sw -B/home/ashan/git/pulp/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/ashan/git/pulp/pulpino/sw/build/CMakeFiles 4
make -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all
make[2]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/depend
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
cd /home/ashan/git/pulp/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ashan/git/pulp/pulpino/sw /home/ashan/git/pulp/pulpino/sw/libs/bench_lib /home/ashan/git/pulp/pulpino/sw/build /home/ashan/git/pulp/pulpino/sw/build/libs/bench_lib /home/ashan/git/pulp/pulpino/sw/build/libs/bench_lib/CMakeFiles/bench.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/build
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/bench_lib/CMakeFiles/bench.dir/build'.
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
[  0%] Built target bench
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
cd /home/ashan/git/pulp/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ashan/git/pulp/pulpino/sw /home/ashan/git/pulp/pulpino/sw /home/ashan/git/pulp/pulpino/sw/build /home/ashan/git/pulp/pulpino/sw/build /home/ashan/git/pulp/pulpino/sw/build/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
make[3]: Nothing to be done for 'CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
[  0%] Built target crt0
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/depend
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
cd /home/ashan/git/pulp/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ashan/git/pulp/pulpino/sw /home/ashan/git/pulp/pulpino/sw/libs/string_lib /home/ashan/git/pulp/pulpino/sw/build /home/ashan/git/pulp/pulpino/sw/build/libs/string_lib /home/ashan/git/pulp/pulpino/sw/build/libs/string_lib/CMakeFiles/string.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/build
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/string_lib/CMakeFiles/string.dir/build'.
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
[  0%] Built target string
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
cd /home/ashan/git/pulp/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ashan/git/pulp/pulpino/sw /home/ashan/git/pulp/pulpino/sw/libs/sys_lib /home/ashan/git/pulp/pulpino/sw/build /home/ashan/git/pulp/pulpino/sw/build/libs/sys_lib /home/ashan/git/pulp/pulpino/sw/build/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/sys_lib/CMakeFiles/sys.dir/build'.
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
[ 25%] Built target sys
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/depend
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
cd /home/ashan/git/pulp/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ashan/git/pulp/pulpino/sw /home/ashan/git/pulp/pulpino/sw/libs/Arduino_lib/core_libs /home/ashan/git/pulp/pulpino/sw/build /home/ashan/git/pulp/pulpino/sw/build/libs/Arduino_lib/core_libs /home/ashan/git/pulp/pulpino/sw/build/libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build
make[3]: Entering directory '/home/ashan/git/pulp/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build'.
[... cut ...]
CMakeFiles/Makefile2:949: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/all] Error 2
make[2]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
CMakeFiles/Makefile2:956: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
make[1]: Leaving directory '/home/ashan/git/pulp/pulpino/sw/build'
Makefile:376: recipe for target 'helloworld.vsimc' failed
make: *** [helloworld.vsimc] Error 2

ashan8k, 2017-08-28:
in gui mode modelsim console 
# source tcl_files/run.tcl
# vsim -quiet tb -L pulpino_lib -L axi_node_lib -L apb_node_lib -L axi_mem_if_DP_lib -L axi_spi_slave_lib -L axi_spi_master_lib -L apb_uart_sv_lib -L apb_gpio_lib -L apb_event_unit_lib -L apb_spi_master_lib -L fpu_lib -L apb_pulpino_lib -L apb_fll_if_lib -L core2axi_lib -L apb_timer_lib -L axi2apb_lib -L apb_i2c_lib -L zero_riscy_lib -L axi_slice_dc_lib -L riscv_lib -L apb_uart_lib -L axi_slice_lib -L adv_dbg_if_lib -L apb2per_lib "+nowarnTRAN" "+nowarnTSCALE" "+nowarnTFMPC" "+MEMLOAD=PRELOAD" -t ps -voptargs="+acc -suppress 2103" -GTEST="" -gRISCY_RV32F=0 -gZERO_RV32E=0 -gZERO_RV32M=0 -gUSE_ZERO_RISCY=0 
# Start time: 16:38:05 on Aug 28,2017
# ** Warning: (vsim-8311) System Verilog assertions are supported only in Questasim.
# ** Fatal: ** Error: (vsim-3828) Could not link 'vsim_auto_compile.so': cmd = '/home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/ashan@e420_dpi_8127/linuxpe_gcc-4.7.4/vsim_auto_compile.so" /home/ashan/git/pulp/pulpino/vsim/work/_dpi/auto_compile@/linuxpe_gcc-4.7.4/*.o   '
# (vsim-50) A call to system(/home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/ashan@e420_dpi_8127/linuxpe_gcc-4.7.4/vsim_auto_compile.so" /home/ashan/git/pulp/pulpino/vsim/work/_dpi/auto_compile@/linuxpe_gcc-4.7.4/*.o    >'/tmp/questatmp.u3dQg8' 2>&1) returned error code '1'.
# The logfile contains the following messages:
# /home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/../libexec/gcc/i686-pc-linux-gnu/4.7.4/ld: /usr/lib32/crti.o: unrecognized relocation (0x2b) in section `.init'
# /home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/../libexec/gcc/i686-pc-linux-gnu/4.7.4/ld: final link failed: Bad value
# collect2: error: ld returned 1 exit status
# 
# No such file or directory. (errno = ENOENT)
# 
# 
# FATAL ERROR while loading design
# Error loading design
# End time: 16:38:06 on Aug 28,2017, Elapsed time: 0:00:01
# Errors: 1, Warnings: 1

FrancescoConti, 2017-08-28:
Ok, did you compile the platform? What is the output of `make vcompile`?

ashan8k, 2017-08-28:
yes make vcompile is 100% successful
--------------------------------------------------------
--> PULPino compilation complete! 
--> Compiling work.tb... 
Compiling component:  work.tb 

--> work.tb compilation complete! 

--> PULPino platform compilation complete! 

Built target vcompile

ashan8k, 2017-08-28:
in modelsim console i can see 
** Fatal: ** Error: (vsim-3828) Could not link ### 'vsim_auto_compile.so': cmd = '/home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/gcc -shared -fPIC -m32 -B/usr/lib32 -o "/tmp/ashan@e420_dpi_8127/linuxpe_gcc-4.7.4/vsim_auto_compile.so" /home/ashan/git/pulp/pulpino/vsim/work/_dpi/auto_compile@/linuxpe_gcc-4.7.4/*.o '

this path is missing. 
 # /tmp/ashan@e420_dpi_8127/linuxpe_gcc-4.7.4/vsim_auto_compile.so

in vsim/tcl_files/config/vsim.tcl
-------------------------------------------------------------
source tcl_files/config/vsim_ips.tcl

set cmd "vsim -quiet $TB \
  -L pulpino_lib \
  $VSIM_IP_LIBS \
  +nowarnTRAN \
  +nowarnTSCALE \
  +nowarnTFMPC \
  +MEMLOAD=$MEMLOAD \
  -gUSE_ZERO_RISCY=$env(USE_ZERO_RISCY) \
  -gRISCY_RV32F=$env(RISCY_RV32F) \
  -gZERO_RV32M=$env(ZERO_RV32M) \
  -gZERO_RV32E=$env(ZERO_RV32E) \
  -t ps \
  -voptargs=\"+acc -suppress 2103\" \
  $VSIM_FLAGS"

\# set cmd "$cmd -sv_lib ./work/libri5cyv2sim"
eval $cmd <-- I guess Issue is here.

ashan8k, 2017-08-28:
I guess this issue is due to 32bit Modelsim. Is there anyone who is trying to do this with Modelsim 32bit version on Ubunutu 16.04

I have mentioned the issue in stack overflow

[link](https://stackoverflow.com/questions/45922292/pulpino-modelsim-issue-vsim-auto-compile-so-32bit-ubuntu)

ashan8k, 2017-08-30:
Further more in the log file this is visible 
`# The logfile contains the following messages:
# /home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/../libexec/gcc/i686-pc-linux-gnu/4.7.4/ld: /usr/lib32/crti.o: unrecognized relocation (0x2b) in section `.init'
# /home/ashan/intelFPGA_pro/17.0/modelsim_ase/gcc-4.7.4-linux/bin/../libexec/gcc/i686-pc-linux-gnu/4.7.4/ld: final link failed: Bad value
# collect2: error: ld returned 1 exit status
# 
# No such file or directory. (errno = ENOENT)
# 
# 
# FATAL ERROR while loading design
# Error loading design
# End time: 17:39:32 on Aug 30,2017, Elapsed time: 0:00:01
# Errors: 1, Warnings: 1
`

ashan8k, 2017-08-30:
Hi, I found the issue. 

 Modelsim uses it's native gcc. That's I got this issue. I had to change `vsim/tcl_files/config/vsim.tcl` file

I added ` -dpicpppath /usr/bin/gcc` at the end of `cmd` like below 
```
set cmd "vsim -quiet $TB \
  -L pulpino_lib \
  $VSIM_IP_LIBS \
  +nowarnTRAN \
  +nowarnTSCALE \
  +nowarnTFMPC \
  +MEMLOAD=$MEMLOAD \
  -gUSE_ZERO_RISCY=$env(USE_ZERO_RISCY) \
  -gRISCY_RV32F=$env(RISCY_RV32F) \
  -gZERO_RV32M=$env(ZERO_RV32M) \
  -gZERO_RV32E=$env(ZERO_RV32E) \
  -t ps \
  -voptargs=\"+acc -suppress 2103\" \
  $VSIM_FLAGS -dpicpppath /usr/bin/gcc"
```

closed 2017-08-30 by ashan8k

promach, 2018-04-17:
For non-pulpino use, how do we manually set the gcc version used for modelsim to the system-wide gcc version as described in [altera forum question](http://www.alteraforum.com/forum/showthread.php?t=57781&p=237623#post237623) ?

wooolf0455, 2018-12-12:
I have some questions like yours , but I can not deal with it. Could you help me?thanks a lot #264

binjon178, 2019-06-14:
Hi @ashan8k 
I also have similar problem with you, which the error msg is below
[Errlog](https://imgur.com/a/OQUz2ES)

Thank you for help !!!