pulp-platform/pulpino #111: Correction in the datasheet

manojcs26, 2017-09-05:
Hey, 
 I was programming SPI master and was not able to understand the logic used in the function
 ` spi_read_fifo(int *data, int datalen) `

I was stuck with the below line.

        while ((((*(volatile int*) (SPI_REG_STATUS)) >> 16) & 0xFF) == 0);
        data[i] = *(volatile int*) (SPI_REG_RXFIFO);

In the datasheet you have mentioned the bits _[31:12]_ of **SPI_STATUS** register as unused but, 18th and 16th bit are connected to _elements_tx_ and _elements_rx_ respectively which are useful bits to determine the completion of transaction.
 So could you please clarify this in the datasheet....

Thank you
Manoj C S

WaleedOsama, 2017-12-03:
hi ,
i am working like you on this , could you help me 
the only datasheet i know about is the 31 page pdf on their website , i don't see the example you are talking about there , could you please tell me where you found it ?

also it would be very helpful to tell me how you simulate your code and make sure it is actualy working 

thanks

WaleedOsama, 2017-12-06:
anyway , i believe that this line is just checking if the FIFO is empty it waits , if not it returns a word from it . but it is weird that in datasheet it said that they are unused