pulp-platform/pulpino #112: Make vcompile error

f2013619, 2017-09-07:
Hi, when I execute make vcompile it compiles upto a point and gives the following error. Hoe can I solve this issue? Thanks in advance

-> Compiling apb_uart... 
Copying /Software/ModelSim/questa_10.1d/questasim/linux_x86_64/../modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  apb_uart 

** Error: Library std not found.
** Error: VHDL Compiler exiting

make[3]: *** [CMakeFiles/vcompile] Error 1
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
make: *** [vcompile] Error 2

FrancescoConti, 2017-09-08:
It looks like QuestaSim is trying to copy modelsim.ini in a directory in which you don’t have write permissions.
I don’t know why this should be happening, but you can try to create an (empty) one in the local vsim directory of PULPino.

Francesco

Francesco Conti
Integrated Systems Laboratory - ETH Zurich
Energy-Efficient Embedded Systems Laboratory - University of Bologna
f.conti@unibo.it<mailto:f.conti@unibo.it> - fconti@iis.ee.ethz.ch<mailto:fconti@iis.ee.ethz.ch>
www-micrel.deis.unibo.it/~conti

f2013619, 2017-09-08:
Hi, I was able to successfully compile. But now when I am running make helloworld.vsim, I am getting the following error

choudhary@i80studpc03:~/pulpino/sw/build:$make helloworld.vsimc
[  0%] Built target crt0
[  0%] Built target sys
[  0%] Built target Arduino_core
[ 33%] Built target Arduino_separate
[ 33%] Built target bench
[ 33%] Built target string
[ 33%] Built target helloworld.elf
[ 33%] Generating vectors/stim.txt
[ 33%] Built target helloworld.stim.txt
[ 33%] Generating modelsim.ini
make[3]: *** [apps/helloworld/modelsim.ini] Error 1
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.links.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsimc.dir/rule] Error 2
make: *** [helloworld.vsimc] Error 2

f2013619, 2017-09-08:
Also, the make vcompile is successful if I remove the apb_uart file. Otherwise I get this error
Compiling component: apb_uart ** Error: Library std not found. ** Error: VHDL Compiler exiting make[3]: *** [CMakeFiles/vcompile] Error 1 make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2 make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2 make: *** [vcompile] Error 2

I will be really grateful if you could help me out.

FrancescoConti, 2017-09-08:
What do you mean by "removing the apb_uart file"? The whole directory from `ips/apb_uart`?

f2013619, 2017-09-08:
No like if I removed the command for apb_uart from vcompile_ips.csh file .

FrancescoConti, 2017-09-08:
Ok, I think the `modelsim.ini` issue is still messing up things. Please try to do the following and report here the full output:

```
cd pulpino
./update-ips.py
cd vsim
cp /Software/ModelSim/questa_10.1d/questasim/linux_x86_64/../modelsim.ini .
vcompile/build_rtl_sim.csh
```

f2013619, 2017-09-11:
Hi, I got the following result 

-> Compiling PULPino Platform... 

--> Compiling PULP IPs libraries... 
--> Compiling axi_node... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  axi_node 

--> axi_node compilation complete! 
--> Compiling apb_node... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  apb_node 

--> apb_node compilation complete! 
--> Compiling axi_mem_if_DP... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  axi_mem_if_DP 

--> axi_mem_if_DP compilation complete! 
--> Compiling axi_spi_slave... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  axi_spi_slave 

--> axi_spi_slave compilation complete! 
--> Compiling axi_spi_master... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  axi_spi_master 

--> axi_spi_master compilation complete! 
--> Compiling apb_uart_sv... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  apb_uart_sv 

--> apb_uart_sv compilation complete! 
--> Compiling apb_gpio... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  apb_gpio 

--> apb_gpio compilation complete! 
--> Compiling apb_event_unit... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
[... 99 lines cut ...]

--> axi_slice_dc compilation complete! 
--> Compiling riscv... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  riscv 

Compiling component:  riscv_regfile_rtl 

Compiling component:  riscv_vip_rtl 

--> riscv compilation complete! 
--> Compiling apb_uart... 
Copying modelsim.ini to /Software/ModelSim/questa_10.1d/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.1d/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.1d/".
Permission denied. (errno = EACCES)
Compiling component:  apb_uart 

** Error: Library std not found.
** Error: VHDL Compiler exiting

f2013619, 2017-09-12:
Hi, If I export MODELSIM the compilation is complete but if afterwards i execute make helloworld.vsim I get the following error
[  0%] Built target string
[  0%] Built target sys
[  0%] Built target Arduino_core
[ 50%] Built target Arduino_separate
[ 50%] Built target bench
[ 50%] Built target crt0
[ 50%] Built target helloworld.elf
[ 50%] Generating vectors/stim.txt
[ 50%] Built target helloworld.stim.txt
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
env: -64: No such file or directory
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make: *** [helloworld.vsim] Error 2
 but I am using questasim_10.2_64 bit setup only. Can you please help me solve this issue?

FrancescoConti, 2017-09-14:
Run this in verbose mode with `make VERBOSE=1 helloworld.vsim`. You should get a lot of usable output for debug.

f2013619, 2017-09-14:
Hi, if I run in verbose mode also I am getting the same error 
env:-64 no such file or directory

This is the verbose output -
/usr/bin/cmake -H/home/choudhary/pulpino/sw -B/home/choudhary/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 helloworld.vsim
make[1]: Entering directory `/home/choudhary/pulpino/sw/build'
/usr/bin/cmake -H/home/choudhary/pulpino/sw -B/home/choudhary/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/choudhary/pulpino/sw/build/CMakeFiles 2
make -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsim.dir/all
make[2]: Entering directory `/home/choudhary/pulpino/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
cd /home/choudhary/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/choudhary/pulpino/sw /home/choudhary/pulpino/sw /home/choudhary/pulpino/sw/build /home/choudhary/pulpino/sw/build /home/choudhary/pulpino/sw/build/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
make[3]: Nothing to be done for `CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
/usr/bin/cmake -E cmake_progress_report /home/choudhary/pulpino/sw/build/CMakeFiles 
[  0%] Built target crt0
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
cd /home/choudhary/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/choudhary/pulpino/sw /home/choudhary/pulpino/sw/libs/sys_lib /home/choudhary/pulpino/sw/build /home/choudhary/pulpino/sw/build/libs/sys_lib /home/choudhary/pulpino/sw/build/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
make[3]: Nothing to be done for `libs/sys_lib/CMakeFiles/sys.dir/build'.
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
/usr/bin/cmake -E cmake_progress_report /home/choudhary/pulpino/sw/build/CMakeFiles 
[  0%] Built target sys
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/depend
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
cd /home/choudhary/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/choudhary/pulpino/sw /home/choudhary/pulpino/sw/libs/Arduino_lib/core_libs /home/choudhary/pulpino/sw/build /home/choudhary/pulpino/sw/build/libs/Arduino_lib/core_libs /home/choudhary/pulpino/sw/build/libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
make[3]: Nothing to be done for `libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build'.
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
/usr/bin/cmake -E cmake_progress_report /home/choudhary/pulpino/sw/build/CMakeFiles 
[  0%] Built target Arduino_core
make -f libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/build.make libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/depend
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
cd /home/choudhary/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/choudhary/pulpino/sw /home/choudhary/pulpino/sw/libs/Arduino_lib/separate_libs /home/choudhary/pulpino/sw/build /home/choudhary/pulpino/sw/build/libs/Arduino_lib/separate_libs /home/choudhary/pulpino/sw/build/libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
make -f libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/build.make libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/build
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
make[3]: Nothing to be done for `libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/build'.
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
/usr/bin/cmake -E cmake_progress_report /home/choudhary/pulpino/sw/build/CMakeFiles  1
[ 50%] Built target Arduino_separate
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/depend
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
cd /home/choudhary/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/choudhary/pulpino/sw /home/choudhary/pulpino/sw/libs/bench_lib /home/choudhary/pulpino/sw/build /home/choudhary/pulpino/sw/build/libs/bench_lib /home/choudhary/pulpino/sw/build/libs/bench_lib/CMakeFiles/bench.dir/DependInfo.cmake --color=
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/build
make[3]: Entering directory `/home/choudhary/pulpino/sw/build'
[... cut ...] rce\ tcl_files/run.tcl\;'
env: -64: No such file or directory
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
make[3]: Leaving directory `/home/choudhary/pulpino/sw/build'
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[2]: Leaving directory `/home/choudhary/pulpino/sw/build'
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make[1]: Leaving directory `/home/choudhary/pulpino/sw/build'
make: *** [helloworld.vsim] Error 2

f2013619, 2017-09-26:
Hi, i solved the issue by generating modelsim.ini file using some other version of modelsim. I guess there was some problem with my previous modelsim. Hence I am closing this issue.

closed 2017-09-26 by f2013619