pulp-platform/pulpino #114: Netlist generation and simulation

FrischAd, 2017-09-14:
Hello,
I generated a netlist using Design Compiler.
I have two issues:
1. In sp_ram_wrap.sv is a module called 'sp_ram_bank' but there is no file containing this module. Where can I find it?
2. I generated the netlist by using the 'sp_ram' and not the 'sp_ram_bank'. Is it possible to simulate this netlist with your provided scripts or alter them easily? If not, how did you verify that the netlist is correct? (Formal verification?)

Thanks in advance.

FrancescoConti, 2017-09-14:
You probably want to replace `sp_ram_bank` with a single ported memory macro  instantiation for your technology of choice if you're doing ASIC synthesis. I'm pretty sure the one in the repo is just a placeholder.
If you use the functional `sp_ram` which can be found in `rtl/components`, you will end up with a memory composed of flip-flops (standard cells), which is likely not what you want.

closed 2017-09-14 by FrancescoConti

FrischAd, 2017-09-14:
Thank you.
Can you explain how you verified the netlist? I would like to use the provided simulations for programs like 'fft' or 'convolution'.

FrancescoConti, 2017-09-14:
I'm not sure I understand the question. As I said in the previous answer, we verified the design at RTL and, targeting technology, with post-synthesis and post-layout gate-level netlists. We also tested several chips post-silicon. We have taped out chips with PULPino in UMC 65nm tech and many other chips using the same IPs within the PULP project in 130nm, 65nm, 40nm and 28nm technologies from several vendors.

We include all the necessary instructions for RTL simulation in the `README.md` file. We do not provide any synthesis script for the reasons I stated before; you can probably reuse many of our RTL scripts but you will have to set up the simulation platform on your own if you want to perform post-synthesis or post-layout simulation on a gate-level netlist.

FrischAd, 2017-09-14:
Thanks again. You answered my question because I want to do the post-synthesis simulation and I wasn't sure how to do this.

f2013619, 2017-10-12:
@FrischAd  Hi, I also want to do a post-synthesis simulation for power analysis in Synopsys prime time for which I require netlist. Can you help me with how did you manage to get the netlist for RTL simulation or how you did the post synthesis simulation.

FrischAd, 2017-10-12:
Hi, if you replace the sp_ram_bank with your own memory macro then it's straight forward. For the synthesis you just have to add all source files, define the top module and then execute your usual synthesis script. If you don't have a memory macro you can use the sp_ram module (of course this isn't perfect but for a first attempt it should be ok).
Right now I'm just synthesizing the riscv core. To perform the post-synthesis simulation you can change the vcompile_riscv.csh file (add your netlist and standard cell library and remove all other parts). Or when you want to do the simulation of the whole platform you can change the name of the  top module in tb.sv and add the netlist to one of the vcompile scripts. Now you can simulate as described in the readme of this repository. Keep in mind that this are some hacks and isn't the proper way but it works =)

f2013619, 2017-10-13:
Hi, Thanks for the reply. Actually I am new to this netlist generation. Can you share with me the netlist file you generated for pulpino or the synthesis script which I can use in DC compiler. I tried doing it but was not able to set the environment correctly. It would be a great help.

FrischAd, 2017-10-13:
I could give you the netlist but I'm not allowed to share the standard cell library. Without this library you can't simulate.
Maybe it's the best for you to read some tutorial about DC to understand what's happening.
This seems to be a short tutorial with a script file: https://www.ese.wustl.edu/~xuan.zhang/ese566_files/tutorials/synopsys_dc_tutorial.pdf

f2013619, 2017-10-13:
Hi, The netlist file would also help. Can you share the netlist file?

FrischAd, 2017-10-13:
I can give you the netlist of the riscv core because right now I don't have a netlist of the whole platform.

f2013619, 2017-10-13:
Yeah okay, that would work.
Thanks a lot.

FrischAd, 2017-10-13:
Just change the file type from .txt to .v (Github does not support .v ....)

[riscv_core.postsyn.txt](https://github.com/pulp-platform/pulpino/files/1382746/riscv_core.postsyn.txt)

f2013619, 2017-10-13:
Thanks a lot .

f2013619, 2017-10-16:
Hi, I was trying to synthesize the platform using dc compiler. But the ffiles are .sv files not .v files. So how did you gave the verilog files as input.

FrancescoConti, 2017-10-16:
Design Compiler supports SystemVerilog input just fine. Just use `analyze -format sverilog filename.sv` instead of `analyze -format verilog filename.v`.

f2013619, 2017-10-16:
Thanks. I am getting the following error while using design compiler. Can you help me resolve it.

Cannot find the design '<pulpino_top.sv>' in the library 'WORK'. (LBR-0)
0
# Designate the design to synthesize
current_design $my_toplevel
Error: Can't find design '<pulpino_top.sv>'. (UID-109)
Error: Current design is not defined. (UID-4)

FrancescoConti, 2017-10-16:
Did you analyze it?

f2013619, 2017-10-16:
yeah I did.

f2013619, 2017-10-16:
This is the script i am using.
[compiledc.txt](https://github.com/pulp-platform/pulpino/files/1387649/compiledc.txt)

FrancescoConti, 2017-10-16:
What is the output of the `elaborate` command?

f2013619, 2017-10-16:
>elaborate pulpino_top -architecture verilog -library DEFAULT
Warning: Can't read link_library file 'vtvt_tsmc180.db'. (UID-3)
Running PRESTO HDLC
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_addr' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_prot' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_region' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_len' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_size' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_burst' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_lock' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_cache' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_qos' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_id' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_user' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_ready' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].aw_valid' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_addr' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_prot' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_region' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_len' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_size' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_burst' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_lock' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_cache' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_qos' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_id' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_user' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_ready' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].ar_valid' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].w_valid' may become connected to an inout port. (VER-735)
Information:  /home/choudhary/Downloads/pulpino/rtl/pulpino_top.sv:181: Variables crossing hierarchy: interface content 'masters[0].w_data' may become connected to an inout port. (VER-735)
[... cut ...] master_APB_BUS__I_soc_ctrl_master_APB_BUS__I_debug_master_APB_BUS__' with
	the parameters "NB_MASTER=9,APB_ADDR_WIDTH=32,APB_DATA_WIDTH=32|((N%clk_i%)(N%rst_ni%)(N%apb_slave%I%WORK/APB_BUS%%APB_ADDR_WIDTH=32,APB_DATA_WIDTH=32)(N%apb_masters%I%WORK/APB_BUS%[9]%%APB_ADDR_WIDTH=32,APB_DATA_WIDTH=32)(N%start_addr_i%)(N%end_addr_i%))". (HDL-193)
Warning: Cannot find the design 'apb_node_wrap' in the library 'WORK'. (LBR-1)
Warning: Design 'pulpino_top' has '21' unresolved references. For more detaile

FrancescoConti, 2017-10-16:
There are several problems:
1) the link library is not readable
```
Warning: Can't read link_library file 'vtvt_tsmc180.db'. (UID-3)
```
2) the `elaborate` command does not seem to be able to find various designs, e.g.
```
Warning: Cannot find the design 'axi2apb32' in the library 'WORK'. (LBR-1)
[...]
Warning: Cannot find the design 'apb_node_wrap' in the library 'WORK'. (LBR-1)
```
3) it seems to me you're trying to `analyze` only the files in the `rtl/` directory, but nothing (or not all) in the `ips/` one.

The consequence, as visible in the last line, is that 
```
Warning: Design 'pulpino_top' has '21' unresolved references. 
```
which is the reason of your following issue.

My suggestion when you deal with DC, especially the first time you synthesize a design, is to open a `dc_shell` and run the commands in the script interactively, up to the point you reach  the `compile` or `compile_ultra` command. That way you can immediately understand problems and usually resolve the least tricky ones.

f2013619, 2017-10-16:
Hi, If I analyze ips it shows a lot of syntax errors like the following 
/home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:35: Syntax error at or near token 'clk_i'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:36: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:36: Syntax error at or near token 'rst_ni'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:38: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:38: Syntax error at or near token 'PADDR'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:39: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:39: Syntax error at or near token 'PWDATA'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:40: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:40: Syntax error at or near token 'PWRITE'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:41: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:41: Syntax error at or near token 'PSEL'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:42: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:42: Syntax error at or near token 'PENABLE'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:43: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:43: Syntax error at or near token 'PRDATA'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:44: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:44: Syntax error at or near token 'PREADY'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:45: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:45: Syntax error at or near token 'PSLVERR'. (VER-294)
Error:  /home/choudhary/pulpino/ips/apb/apb2per/apb2per.sv:50: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  Too many errors; can't continue. (VER-40)

FrancescoConti, 2017-10-16:
(I'm not sure whether the previous comment was deleted, in any case here's a reply): you have a lot of errors due to `Redeclaration of port 'logic' is not allowed. (VER-121)`: double-check you are using the `-format sverilog` flag, because the synthesis tool is treating `logic` as a port name instead of a language keyword.

f2013619, 2017-10-17:
Hey, thanks for all the help.I really appreciate it.  I am down to two unresolved issues. Can you help me with how to resolve them. 
They are as follows-
Building the design 'umcL65_LL_FLL'. (HDL-193)
Warning: Cannot find the design 'umcL65_LL_FLL' in the library 'WORK'. (LBR-1)
Information: Building the design 'sp_ram_bank' instantiated from design 'sp_ram_wrap_RAM_SIZE32768_DATA_WIDTH32_1' with
        the parameters "NUM_BANKS=8,BANK_SIZE=1024". (HDL-193)
Warning: Cannot find the design 'sp_ram_bank' in the library 'WORK'. (LBR-1)
Warning: Unable to resolve reference 'umcL65_LL_FLL' in 'clk_rst_gen'. (LINK-5)
Warning: Unable to resolve reference 'sp_ram_bank' in 'sp_ram_wrap_RAM_SIZE32768_DATA_WIDTH32_1'. (LINK-5)
0

FrancescoConti, 2017-10-17:
For what concerns the second one, you have to replace `sp_ram_bank` with an instantiation of your own favorite memory cut from your technology library (see the initial question of this same issue).

For what concerns the FLL, this is a clock generator IP that is used in our PULP-based chips (see e.g. http://ieeexplore.ieee.org/abstract/document/7934447/ ) but is not released as open source as far as I know. If you want to tape out a chip you will of course need a clock generator IP (not necessarily this one), but if you just want to synthesize for comparisons or validation, than you can probably simply bypass it (remove the IP from `clk_rst_gen.sv` + hardwire `clk_sel_i` to 0 so that the internal clock is tied to the reference clock `clk_i`).

f2013619, 2017-10-17:
Hi,  I am not able to replace the sp_ram_bank with sp_ram. Can you tell me where all I need to replace. I was able to remove the other error. Sorry for the trouble.

FrancescoConti, 2017-10-17:
You must not substitute `sp_ram_bank` with `sp_ram`, which is a functional model for simulation. This would likely result in a synthesized design with all memory implemented with flip-flops, which is 99.9999% not what you want.

What you actually want is to get the library files for memories in your technology. Assuming you are working in an academic or industrial context, you will probably be able to get help on these in the same department where you got the standard cell technology libraries.
Memory libraries are typically distributed in `.db` databases for Design Compiler, and you can add them to your `link_library` to make them linkable by your design. Then you have to instantiate them in your code in place of `sp_ram_bank`; typically you can find the instantiation template from a functional model of the memories distributed together with the libraries. It usually requires some minimum effort to correctly bind the memory cuts with your surrounding logic (in this case, that in `sp_ram_wrap`).

f2013619, 2017-10-18:
Hey. thanks for the information. I am now using the memory lib SRAM4x32.db' but I am not understanding what exact changes I need to do to instantiate them as I dont have the template file. I am getting the other error now -
Error: Can't find inout port 'rstn_i' on reference to 'sp_ram' in 'instr_ram_wrap_RAM_SIZE32768_DATA_WIDTH32'. (LINK-1)
Error: Unable to match ports of cell core_region_i/instr_mem/sp_ram_i ('sp_ram') to 'sp_ram_RAM_SIZE32768_DATA_WIDTH32'. (LINK-25)

saranprasad, 2017-11-03:
@FrischAd 
One help.
Hi,
I want to do analysis and i want to use sp_ram for ram. What all changes are required. I changed in sp_ram_wrap and changed the parameters in components/sp_ram.sv as both side parameters match. The tool crashes when executed compile_ultra when processing ram after a long hang

```
Beginning Pass 1 Mapping
  ------------------------
  Processing 'sp_ram_ADDR_WIDTH15_DATA_WIDTH32_NUM_WORDS32768_1'
dcfarm: detected bad exit for job 3004, name J1 -- terminating all child jobs now...
dcfarm: destroying parent process due to child fatal
```

FrischAd, 2017-11-04:
Hi,
you have to make sure that 'ASIC' is not set and so DC will use sp_ram. Further changes are not required.
I can't help you with this erro message because this didn't happen when I tried it.

saranprasad, 2017-11-05:
Ok.
I removed ASIC flag from config. Still facing same issue (tool crashes while processing sp_ram !!!).
Regression test for behavioural source cloned from github passed for all but cnn_layer.test and cnn_layer.test.

Now as the ASIC flag is removed, it shd go through same flow. But, tool crashes

If we set ASIC and dont make any code changes, netlist is generated successfully with 2 blackbox components for sp_ram and fll. !!!!!!

saranprasad, 2017-11-06:
Hi FrischAd,

Something wrong with memory which I cant figure out now.
One help on post synthesis simulation
If I am synthesising only riscv. 

"Right now I'm just synthesizing the riscv core. To perform the post-synthesis simulation you can change the vcompile_riscv.csh file (add your netlist and standard cell library and remove all other parts). Or when you want to do the simulation of the whole platform you can change the name of the top module in tb.sv "

"pulpino/vsim/vcompile/ips" folder have many csh files. 
I can see something like below: Shall i remove all the lines of vlog and replace it with netlist and same vlog format ? Where should i give standard cell library ? 
How to start sim. for ex: from build make helloworld.vsim will now use netlist ?

```
echo "${Green}Compiling component: ${Brown} riscv ${NC}"
echo "${Red}"
vlog -quiet -sv -suppress 2583 -work ${LIB_PATH}    +incdir+${IP_PATH}/include+incdir+${IP_PATH}/../../rtl/includes ${IP_PATH}/include/apu_core_package.sv || goto error
vlog -quiet -sv -suppress 2583 -work ${LIB_PATH}    +incdir+${IP_PATH}/include+incdir+${IP_PATH}/../../rtl/includes ${IP_PATH}/include/riscv_defines.sv || goto error
vlog -quiet -sv -suppress 2583 -work ${LIB_PATH}    +incdir+${IP_PATH}/include+incdir+${IP_PATH}/../../rtl/includes ${IP_PATH}/include/riscv_tracer_defines.sv || goto error
vlog -quiet -sv -suppress 2583 -work ${LIB_PATH}    +incdir+${IP_PATH}/include+incdir+${IP_PATH}/../../rtl/includes ${IP_PATH}/riscv_alu.sv || goto error
vlog -quiet -sv -suppress 2583 -work ${LIB_PATH}    +incdir+${IP_PATH}/include+incdir+${IP_PATH}/../../rtl/includes ${IP_PATH}/riscv_alu_basic.sv || goto error
```

FrischAd, 2017-11-06:
Hi,
I think you don't desire to have blackboxes because without a behaviour description you can't simulate it.

The pulpino/vsim/vcompile/ips directory is correct and there you can find the vcompile_riscv.csh. 
You add in this file the netlist AND standard cell library using the 'vlog' command (same format). Of course you have to adapt the path to your netlist and standard cell library.
Now you can remove all the other 'vlog' commands.
Start the simulation as always but make sure that you call the 'vcompile' command in the build directory.
The simulation will use the netlist for the riscv core (all the other files are still using the rtl code).

saranprasad, 2017-11-07:
Ok.
Did you do postsyn sim for riscv only. My postsyn simulation is not moving from "Preloading data memory from slm_files/tcdm_bank0.slm". I waited for hours. It is not finishing. Any idea ?
PS: clock gating is bypassed with a behav model (assigned both clocks to same)

FrischAd, 2017-11-10:
In the meantime I did a postsyn simulation for the whole plattform as well.
I had problems with the clock gating, too. I changed the implementation and now it's working.
I have no idea concerning the memory loading problem. You can try to load the memory via the SPI instead of preloading. The helloworld example takes about 3 million ns.

saranprasad, 2017-11-10:
Ok. My post syn sim with riscv (with unresolved reference) going good. May I know the clk freq you are using for simulation

FrischAd, 2017-11-10:
Which references are unresolved?
I used the default 25 MHz.

saranprasad, 2017-11-15:
my cluster_clock_gating is unresolved considering only riscv synthesis.

Actually when i solve it by assigning "clk_o = clk_i", after that post synthesis simulation is stalled (gets paused at the stage i mentione above).
Without that it is passing!! (but some doubts in power estimation)

Regarding frequency ? Is this delay used having a high margin than required ?
Black box in netlist. Do you think it will impact the power or timing in any way ?

saranprasad, 2017-11-15:
can you suggest a trick for cluster_clock_gating. something is wrong with directly assigning it as above !!

FrischAd, 2017-11-15:
The clock_gating module is provided in the directory pulpino/rtl/includes.
So when you don't assign "clk_o=clk_i" then it is passing??

Sorry I don't get it with the frequency.
When there is a block box then there is no power estimation for this module and when this module is responsible for the clock generation then there is no clock at all (which will heavily impact the power estimation).

saranprasad, 2017-11-15:
Yes. When I leave the black box as such, it is passing. but some doubts in the power and timing analysis

saranprasad, 2017-11-15:
have you successfully done post synthesis simulation for riscv only ? (with cluster gate)

FrischAd, 2017-11-15:
Yes, I've done a successfull synthesis for the riscv only and for the whole plattform.
You have to add the cluster clock gating for a correct synthesis.

saranprasad, 2017-11-23:
hi,
there is clk and clk_i in riscv core. Do you know what is the idea behind it ?

FrischAd, 2017-11-24:
clk_i is the input clock and clk is the clock used within the riscv core (which can be clock gated)

saranprasad, 2017-11-24:
within riscv core, this is something which is not getting instantiated !!
The sequential power will be 0 and clock network power will be there. Is that ok. External clock (input) only needs to be given for synthesis and power analysis. Am I right ?

FrischAd, 2017-11-24:
What do you mean by 'not getting instantiated'? Can you be more precise?
clk_i is an input. Clk is an internal signal. Both signals are connected with core_clock_gate_i.

What is sequential power?

FrancescoConti, 2017-11-24:
As stated previously by @FrischAd, you have to provide a clock gating cell that will connect the input `clk_i` to the internal (possibly gated) `clk`.

Sequential power, as defined by Design Compiler, refers to the amount of power spent within sequential blocks (flip-flops, latches) as opposed to that spent inside combinational blocks or in the interconnects. Of course it makes no sense for it to be 0. In general, you shouldn't trust power reports unless they are intuitively reasonable. Moreover, 1) post-synthesis power reports are much more unrealistic (optimistic) than post-layout ones - topological synthesis helps quite a bit 2) power reports obtained without switching activities collected directly with simulations on the netlist are *not reliable* at all and can be wildly incorrect.
I have no idea what you mean by the following assertion:
> External clock (input) only needs to be given for synthesis and power analysis.
In general, you can say that having something "broken" on the clock path will break your synthesis real fast. Registers not reached by a clock will be optimized away (i.e. removed) and power, area and timing results will cease to make any sense. Correctly specifying clocks is one of the keys to correct RTL-to-gates synthesis.

Honestly from your questions it appears you are lacking some preliminary knowledge necessary to be productive in doing synthesis and power analysis. I can recommend you do some preliminary study on textbooks (e.g. some on [this list](https://www.amazon.com/s/ref=nb_sb_noss?url=search-alias%3Daps&field-keywords=digital+vlsi+design&rh=i%3Aaps%2Ck%3Adigital+vlsi+design) contain decent introductions to using Design Compiler among other tools), ask around on StackOverflow or - if you have this possibility - ask more expert people in your group/department to guide you, which is definitely the best option. The PULPino issue tracker is unfortunately _not the proper place_ to learn how to properly work on synthesis.

saranprasad, 2017-11-24:
Ok. I am actually sorry for disturbing @FrischAd , @FrancescoConti 
Yes. Power analysis using switching activity is what i am relying on.

We have these power groups. That is why i guess. 
clock_network: cells in   the   clock_network   excluding   io_pad cells.
register:   latches and flip flops driven by the clock network excluding io_pads and black_boxes.
sequential:   latches   and flip flops clocked by signals other than those in the clock network.   

Thanks
Saran

saranprasad, 2018-01-08:
Hi,

One help. How to use sdf file for  gate-leveel simulation for riscv only. sdfnoerror in the apps .dir folder of app is not helping

Thanks
Saran

saranprasad, 2018-01-08:
I am not able to reduce the clock period to make it close to synthesis time. (I am considering only received return value is 0 as a application is passed).
The simulation continues.

```
# ** Error: (vsim-SDF-3250) /home/prasar00/project/postsynt_riscv_only/pulpino/vsim/..//tb/tb.sv(217): Failed to find INSTANCE 'top_i/core_region_i'.
# ** Warning: (vsim-3770) Failed to find user specified function 'mem_push' in DPI C/C++ source files.
```

while reading sdf file, it can recognize only top_i. Not beyond that. What should I do ?

FrischAd, 2018-01-08:
I'm able to use sdf files. My approach (id_stage):
1. Compile sdf file in the pulpino/vsim/run.tcl: 
     "sdfcom Path_To/file.sdf compiled_file_name.sdf"
    This uses typical delays, you can add the options "-maxdelays" or "-mindelays" to use other delays.
2. In the pulpino/vsim/tcl_files/config/vsim.tcl add the compiled sdf file to cmd:
    "-sdftyp /top_i/core_region_i/CORE/RISCV_CORE/id_stage_i=Path_To/compiled_file_name.sdf"
    If you are using min or max delays then you must change the option sdftyp to sdfmin or sdfmax.

Hope this helps you. 
Another way is to reference the sdf file in the Verilog code (here in tb.v):
"initial $sdf_annotate("Path_To/file_name.sdf", "tb", "sdffile.log");"
See https://studfiles.net/preview/4447478/page:4/
I didn't use this approach so I can't tell you much more than this.

saranprasad, 2018-01-09:
Hi,

Can you tell me why you gave id_stage_i ?
I want to load timing info of all components under riscv. What should I do ?

FrancescoConti, 2018-01-09:
The first approach suggested by @FrischAd should work; if you need to do it for the full core you can adapt what he did, e.g.
`-sdftyp /top_i/core_region_i/CORE/RISCV_CORE=compiled_sdf_file.sdf`

saranprasad, 2018-01-09:
Hi,

I am not getting it. I tried some small modifications here and there. But, it is not working.

```
# ** Warning: (vsim-3448) /home/prasar00/project/riscv_syn_noAging_45nm1/riscv_core.sdf(425881): Setting negative specify check constraint (-3 ps) to zero.
# ** Fatal: (vsim-SDF-3444) Failed to annotate from SDF file "/home/prasar00/project/riscv_syn_noAging_45nm1/riscv.sdf".
```

Thanks
Saran

FrischAd, 2018-01-09:
The id_stage was just an example.
What are your modifications? Have you tried to use an empty sdf file to check that the error is not in the sdf file?

saranprasad, 2018-01-11:
Added the below command in cmd  (vsim.tcl)

```
-sdftyp /top_i/core_region_i/CORE/RISCV_CORE=/home/prasar00/project/riscv_syn_noAging_45nm1/riscv.sdf
```

in run.tcl:
```
sdfcom /home/prasar00/project/riscv_syn_noAging_45nm1/riscv_core.sdf /home/prasar00/project/riscv_syn_noAging_45nm1/riscv.sdf
```

But, I am not able to run the simulation with synthesised clock for riscv.
If simulated with lower clock, I get setup error from the library used and simulation does not stop.

Thanks
Saran

saranprasad, 2018-01-11:
By modifications, I mean:

1. Without compiling -> gave the name of sdf file directly in sdftyp.
2. Added -sdfnoerror in vsim.tcl and did $sdf_annotate with top_i as instance name in the command. 

Changes like this. But, it is not useful. The clock is not synchronised with synthesised one.
issue #146 : I didnt understand what is it about uart and spi

Thanks
Saran

FrancescoConti, 2018-01-11:
You should use either `-sdftyp stuff` in the command line of vsim.tcl, or `$sdf_annotate$` in the testbench, not both at the same time.

That said, SDF annotation may file for many reasons - usually the fatal error at the end is the consequence of some warning or error further up. Try checking the entire log.

PS: I am not sure what you mean by "synthesized clock".

[22 more replies]