pulp-platform/pulpino #115: Error occurs when I clone the subrepositores and update them

kuopinghsu, 2017-09-21:
I got following error message when I run the 'update-ips.py' script,

Your branch is up-to-date with 'origin/verilator'.
ERROR: targets not allowed for ip 'riscv', sub-ip 'riscv_regfile_rtl':
set(['gf22'])
    gf22
Check the src_files.yml file.

I think the problem is at ipstools/SubIPConfig.py line 49. It declare the allowed target is 'gf28', but the file 'ips/riscv/src_files.yml' line 48 is 'gf22'. If I modify 'ipstools/SubIPConfig.py' line 49 to 'gf22', then the error message is gone.

woodyplum, 2017-09-22:
got the same error, can not finish the make vcompile step, i check in a totally new git clone.

davideschiavone, 2017-09-25:
Hello,

thanks for reporting, this should be fixed with this commit now

https://github.com/pulp-platform/IPApproX/commit/e528de74bef7a8ffd1f16469b040b977921cf728

Cheers,
Davide

closed 2017-09-28 by FrancescoConti