pulp-platform/pulpino #116: running a new application on pulpino

f2013619, 2017-09-26:
Hi, Can you please guide me with how to run a new fft application or give me the solution for motion detection exercise that was given by eth zurich in one power point presentation.
Thanks in advance

f2013619, 2017-09-26:
I am getting the following error when I try to run a new application which contains c code for sum of two numbers

Scanning dependencies of target applcatn.elf
[100%] Building C object apps/applcatn/CMakeFiles/applcatn.elf.dir/applcatn.c.o
Linking CXX executable applcatn.elf
/home/choudhary/Dokumente/ri5cy_gnu_toolchain-master/install/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: applcatn.elf section `.text' will not fit in region `instrram'
/home/choudhary/Dokumente/ri5cy_gnu_toolchain-master/install/lib/gcc/riscv32-unknown-elf/5.2.0/../../../../riscv32-unknown-elf/bin/ld: region `instrram' overflowed by 29868 bytes
collect2: error: ld returned 1 exit status
make[3]: *** [apps/applcatn/applcatn.elf] Error 1
make[2]: *** [apps/applcatn/CMakeFiles/applcatn.elf.dir/all] Error 2
make[1]: *** [apps/applcatn/CMakeFiles/applcatn.dir/rule] Error 2
make: *** [applcatn] Error 2

FrancescoConti, 2017-09-28:
The error reports that binary code of your application in the `.text` segment does not fit in the instruction RAM (it overflows it by 29kB).
You are probably including data as well as code in `.text`; you have to reduce the size of your binary.

closed 2017-09-28 by FrancescoConti

f2013619, 2017-11-23:
Whwn I am trying to rum my application which consists of different sub directories. For some of them it shows the following error. Can you tell me how I can remove these errors?
Thank You in advance 

../../../libs/Arduino_lib/core_libs/libArduino_core.a(main.cpp.o): In function `initVariant':
/home/choudhary/pulpino/sw/libs/Arduino_lib/core_libs/inc/main.cpp:36: undefined reference to `setup'
/home/choudhary/pulpino/sw/libs/Arduino_lib/core_libs/inc/main.cpp:37: undefined reference to `loop'
/home/choudhary/pulpino/sw/libs/Arduino_lib/core_libs/inc/main.cpp:40: undefined reference to `loop'
collect2: error: ld returned 1 exit status
make[3]: *** [apps/SeizureDetection/src/seiz_pred.elf] Error 1
make[2]: *** [apps/SeizureDetection/src/CMakeFiles/seiz_pred.elf.dir/all] Error 2
make[1]: *** [apps/SeizureDetection/src/CMakeFiles/seiz_pred.dir/rule] Error 2
make: *** [seiz_pred] Error 2

Nagatejapallapu, 2021-08-12:
Did you find the solution to this? If yes, Please help out. Thank you.

aledolme, 2023-01-04:
Did you find a solution to these errors? I've the same problem compiling my own application.