pulp-platform/pulpino #118: RISC-V Linux support?

alswg1994, 2017-09-28:
Hi, I'm working on FPGA EDAC module directly attachable to one's core.
While I have chosen Pulpino to test this, I need Linux running on its RISCY(RV32IMC+F) core.
Does Pulpino support Linux? If yes, how can I boot Linux on Pulpino itself using spiload or any other?

FrancescoConti, 2017-09-28:
Hi, the PULPino system does not have an MMU and does not support Linux.
It is a microcontroller system comparable to a Cortex-M class commercial MCU.

closed 2017-09-28 by FrancescoConti

alswg1994, 2017-09-28:
There's Linux version  that doesn't need MMU. Is that also not supported?

FrancescoConti, 2017-09-28:
The RI5CY core itself should be enough then. But you would still need an instruction cache, or an instruction memory big enough.

be4web, 2017-09-28:
In theory... You will need to spend significant (!) amounts of work to adapt Linux. Furthermore RI5CY does not implement all necessary privilege levels, neither is the interrupt Controller PLIC comformant or the timer architecture comformant to the riscv privileged specification.

My word of advise: step back and evaluate why you really need Linux. If there is no way around it, I would strongly suggest to try one of the Berkeley cores.

alswg1994, 2017-09-28:
@be4web You mean Rocket right? Thanks!

be4web, 2017-09-28:
Yes, or BOOM

be4web, 2017-09-28:
Where does the verification software run? On an auxiliary core? All FPGA vendors either provide a Linux capable soft or hard core.

alswg1994, 2017-09-29:
@be4web It has to be run on the core that EDAC module attaches to, so yeah, I'll go with Rocket(BOOM seems to have limited support? I think).