pulp-platform/pulpino #119: Issue trying to run helloworld.vsim

rujutak7, 2017-10-06:
Hi,

I not able to run the simulation for helloworld.vsim. I have successfully compiled the RTL and use the compiler from https://github.com/pulp-platform/ri5cy_gnu_toolchain.

When I try to run helloworld.vsim I get the following error.

[  0%] Built target bench
[  0%] Built target crt0
[  0%] Built target string
[  0%] Built target sys
[  0%] Built target Arduino_core
[ 50%] Built target Arduino_separate
[ 50%] Built target helloworld.elf
[ 50%] Generating vectors/stim.txt
[ 50%] Built target helloworld.stim.txt
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
env: -64: No such file or directory
apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make:49: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim' failed
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
CMakeFiles/Makefile2:1169: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
CMakeFiles/Makefile2:1177: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:471: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2

Please advice.
Thanks,
Rujuta

FrancescoConti, 2017-10-06:
You should check that the CMake configuration scripts, e.g. `cmake_configure.riscv.gcc.sh`, reference the correct path to ModelSim. By default this is set by using the `which vsim` command, and most likely this is what fails in your system.

closed 2017-10-06 by FrancescoConti

rujutak7, 2017-10-06:
This worked, Thanks!

riteish4141, 2018-01-10:
Hello,

I am having the same problem and as mentioned I referred the path to ModelSim in configuration scripts.
I have question regarding it as may be I am doing it wrong-
Do I need to reference the correct path to ModelSim in all the configuration scripts in the /pulpino-master/sw/build folder or only to cmake_configure.riscv.gcc.sh ?
My ModelSim is located at /home/riteish/altera/13.1/modelsim_ase/bin so I replaced the VSIM=`which vsim` by VSIM=/home/riteish/altera/13.1/modelsim_ase/bin
Is it right or I am missing on something ?