pulp-platform/pulpino #122: password for git@iis-git.ee.ethz.ch's

saranprasad, 2017-10-17:
Asking for password when executed generate-scripts.py

prasar00@i80studpc06:~/project/pulpino-Pulpino_v2.1:$./generate-scripts.py 
Cloning into 'ipstools'...
git@iis-git.ee.ethz.ch's password: 
Connection closed by 129.132.2.24

FrancescoConti, 2017-10-17:
You should start by running `update-ips.py`, not `generate-ips.py`. Please remove the `ipstools` directory and then run `update-ips.py`, it should work that way.

(That said, there's a probably a bug in `generate-scripts.py` - we will have a look into fixing it)

saranprasad, 2017-10-17:
Please find the output of update-ips
Regarding the probable bug. I am starting to use pulpino. So i cloned the released version (v2.1).

prasar00@i80studpc06:~/project/pulpino-Pulpino_v2.1:$**./update-ips.py** 
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Traceback (most recent call last):
  File "./update-ips.py", line 65, in <module>
    [server, group, remote] = find_server()
  File "./update-ips.py", line 53, in find_server
    print tcolors.ERROR + "ERROR: could not find remote server." + tcolors.ENDC
NameError: global name 'tcolors' is not defined
prasar00@i80studpc06:~/project/pulpino-Pulpino_v2.1:$