pulp-platform/pulpino #123: Server not found in update-ips script

saranprasad, 2017-10-17:
Server not found error coming.
Latest clone is used.

prasar00@i80studpc06:~/project/pulpino-master:$./update-ips.py 
Traceback (most recent call last):
  File "./update-ips.py", line 65, in <module>
    [server, group, remote] = find_server()
  File "./update-ips.py", line 53, in find_server
    print tcolors.ERROR + "ERROR: could not find remote server." + tcolors.ENDC
NameError: global name 'tcolors' is not defined
prasar00@i80studpc06:~/project/pulpino-master:$

FrancescoConti, 2017-10-17:
What is the output of `git remote -v`?

saranprasad, 2017-10-17:
prasar00@i80studpc06:~/project/pulpino-master:$pwd
/home/prasar00/project/pulpino-master
prasar00@i80studpc06:~/project/pulpino-master:$git remote -v
prasar00@i80studpc06:~/project/pulpino-master:$

FrancescoConti, 2017-10-17:
How did you clone the project? The `update-ips.py` script uses the remote repository information  (accessed through the `git remote -v` command) to understand where to fetch stuff from.

PS: when you copy-paste code, please embed it with triple ` characters to enter a verbatim environment. E.g.:

```
    The following is some code:
    ```
    $ cd pulpino
    $ ls
    ```
```

I edited your previous comments in this fashion to make them understandable.

saranprasad, 2017-10-17:
I didnt copy paste the code. Clone/download from github directly. Then extracted the zip file to get 'pulpino_master' folder.

meanwhile, there was one more error like below previously. this is gone after executing 'git init'
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

FrancescoConti, 2017-10-17:
1. What I meant is you copy-pasted code in the previous response, which is why they were (scarcely) readable like this:

prasar00@i80studpc06:~/project/pulpino-master:$pwd
/home/prasar00/project/pulpino-master
prasar00@i80studpc06:~/project/pulpino-master:$git remote -v
prasar00@i80studpc06:~/project/pulpino-master:$

instead of this

```
prasar00@i80studpc06:~/project/pulpino-master:$pwd
/home/prasar00/project/pulpino-master
prasar00@i80studpc06:~/project/pulpino-master:$git remote -v
prasar00@i80studpc06:~/project/pulpino-master:$
```

which is embedded in triple ` chars. Please do it this way.

2. You shouldn't download the .zip archive (which apparently does not include the `.git` subdirectory):  you should directly clone pulpino from GitHub. This is the normal flow for GitHub projects:

```
git clone https://github.com/pulp-platform/pulpino.git
```

Then proceed as in the README.

saranprasad, 2017-10-17:
Ok. Update-ips's was successful. 
Carrying on as README.

closed 2017-10-17 by FrancescoConti

saranprasad, 2017-10-17:
Hi,
error on configas below. compler used is COMPILER=`which riscv32-unknown-elf-gcc`

prasar00@i80studpc06:~/project/pulpino/sw/build:$./cmake_configure.riscv.gcc.sh 
-- The C compiler identification is unknown
-- The CXX compiler identification is GNU 4.8.4
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error: your C compiler: "CMAKE_C_COMPILER-NOTFOUND" was not found.   Please set CMAKE_C_COMPILER to a valid compiler path or name.
CMake Error: your C compiler: "CMAKE_C_COMPILER-NOTFOUND" was not found.   Please set CMAKE_C_COMPILER to a valid compiler path or name.
-- GCC_MARCH= IMXpulpv2
-- USE_ZERO_RISCY= 0
-- RISCY_RV32F= 0
-- ZERO_RV32M= 0
-- ZERO_RV32E= 0
-- PL_NETLIST= 
-- Configuring incomplete, errors occurred!
See also "/home/prasar00/project/pulpino/sw/build/CMakeFiles/CMakeOutput.log".
See also "/home/prasar00/project/pulpino/sw/build/CMakeFiles/CMakeError.log".
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_C_COMPILER= 

-- Generating done
-- Build files have been written to: /home/prasar00/project/pulpino/sw/build
prasar00@i80studpc06:~/project/pulpino/sw/build:$

saranprasad, 2017-10-17:
solved. path was not set