pulp-platform/pulpino #124: make vcompile error "vlib not found"

saranprasad, 2017-10-17:
make vcompile throwing error on axi_node as below
'''
prasar00@i80studpc06:~/project/pulpino/sw/build:$make vcompile
Scanning dependencies of target vcompile

vlib: Command not found.
vlib: Command not found.

--> Compiling PULPino Platform... 

--> Compiling PULP IPs libraries... 
--> Compiling axi_node... 
vlib: Command not found.
vmap: Command not found.
Compiling component:  axi_node 

vlog: Command not found.

make[3]: *** [CMakeFiles/vcompile] Error 1
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
make: *** [vcompile] Error 2

'''

FrancescoConti, 2017-10-17:
`vlib`, `vmap`, `vlog` are not in your path. Check that you have ModelSim or QuestaSim correctly installed and in your `PATH` environment variable.

saranprasad, 2017-10-17:
sourced questasim 10.2 64bit. then on the same terminal if we run vcompile, it starts good. but, some error as below is coming:

1. error in all nodes as this. not sure if there is any permission issue. i am not root.
'''
--> Compiling axi_node... 
Copying /Software/ModelSim/questa_10.2c_5/questasim/linux_x86_64/../modelsim.ini to /Software/ModelSim/questa_10.2c_5/
** Error: (vmap-7) Failed to open ini file "/Software/ModelSim/questa_10.2c_5/" in write mode.
Is a directory. (errno = EISDIR)
** Error: (vmap-20) Cannot access for writing file "/Software/ModelSim/questa_10.2c_5/".
Permission denied. (errno = EACCES)
'''

2. In uart, std not found error and make stopped
'''
Compiling component:  apb_uart 

** Error: Library std not found.
** Error: VHDL Compiler exiting

'''

FrancescoConti, 2017-10-17:
This is an error that happens sometimes with ModelSim/QuestaSim, as it tries to copy modelsim.ini in the wrong place. Try to copy your baseline modelsim.ini in the `pulpino/sim` folder before running `make vcompile`, i.e. in your case

```
cd pulpino/sim
cp /Software/ModelSim/questa_10.2c_5/questasim/linux_x86_64/../modelsim.ini ./modelsim.ini
```

saranprasad, 2017-10-17:
copied modelsim.ini to pulpino/**vsim** .
Both the issue is showing up again on vcompile

be4web, 2017-10-17:
Both issues are strongly suggesting that you have problems with your local Modelsim installation. Please try to fix them and validate that everything is working, by for example, compiling a design stand-alone.

You can't expect us to help everybody using our design to debug their EDA tool installation.

closed 2017-10-17 by be4web

saranprasad, 2017-10-17:
Ok