pulp-platform/pulpino #126: make helloworld error

saranprasad, 2017-10-19:
Build was successful (make vcompile). But make helloworld.vsim error.  and .vsim file dont exist in apps/helloworld

'''
prasar00@i80studpc06:~/project/pulpino/sw/build:$make helloworld.vsim
[  0%] Built target string
[  0%] Built target sys
[  0%] Built target Arduino_core
[ 50%] Built target Arduino_separate
[ 50%] Built target bench
[ 50%] Built target crt0
[ 50%] Built target helloworld.elf
[ 50%] Generating vectors/stim.txt
[ 50%] Built target helloworld.stim.txt
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
env: -64: No such file or directory
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make: *** [helloworld.vsim] Error 2
prasar00@i80studpc06:~/project/pulpino/sw/build:$
'''

saranprasad, 2017-10-19:
```
prasar00@i80studpc06:~/project/pulpino/sw/build:$make helloworld.vsim
[  0%] Built target string
[  0%] Built target sys
[  0%] Built target Arduino_core
[ 50%] Built target Arduino_separate
[ 50%] Built target bench
[ 50%] Built target crt0
[ 50%] Built target helloworld.elf
[ 50%] Generating vectors/stim.txt
[ 50%] Built target helloworld.stim.txt
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
env: -64: No such file or directory
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make: *** [helloworld.vsim] Error 2
prasar00@i80studpc06:~/project/pulpino/sw/build:$
```

FrancescoConti, 2017-10-19:
See my answer to issue #119

closed 2017-10-19 by FrancescoConti

saranprasad, 2017-10-19:
yes. the issue resolved. Thank you