pulp-platform/pulpino #127: illegal instruction when using microriscy configuration for simulation

woodyplum, 2017-10-23:
hi, 
   i just want to run the hellworld test with the microriscy configuration with the cmake_configure.microriscy.gcc.sh script. while the simulation can not finished. after check the waveform it can be found that after the compile the helloworld.read has the result as following.  
    where the instruction in RI5CY "0527b95b          	p.subn	s2,a5,s2,2" still there and the verilog does not support for it and generate a illegal_insn_handler thus the simulation fail. 

Does the microriscy configuration ready for simualtion, or i need more configuration for it to run? thanks. 

------------------------------------------------------------------------------------------------------------------------
000002b8 <__libc_init_array>:
 2b8:	ff010113          	addi	sp,sp,-16
 2bc:	00912223          	sw	s1,4(sp)
 2c0:	001007b7          	lui	a5,0x100
 2c4:	001004b7          	lui	s1,0x100
 2c8:	01212023          	sw	s2,0(sp)
 2cc:	00078793          	mv	a5,a5
 2d0:	00048913          	mv	s2,s1
 2d4:	0527b95b          	0x527b95b
 2d8:	00812423          	sw	s0,8(sp)
 2dc:	00112623          	sw	ra,12(sp)
 2e0:	00048493          	mv	s1,s1
 2e4:	00000413          	li	s0,0 # 0 <__DYNAMIC>
 2e8:	00090a63          	beqz	s2,2fc <__libc_init_array+0x44>
 2ec:	0044a78b          	0x44a78b
 2f0:	00140413          	addi	s0,s0,1
 2f4:	000780e7          	jalr	a5
 2f8:	fe891ae3          	bne	s2,s0,2ec <__libc_init_array+0x34>
 2fc:	001004b7          	lui	s1,0x100
 300:	f91ff0ef          	jal	290 <_fini>
 304:	001007b7          	lui	a5,0x100
 308:	00048913          	mv	s2,s1
 30c:	00078793          	mv	a5,a5
 310:	0527b95b          	0x527b95b

haugoug, 2017-10-23:
It seems you are using the wrong compiler as it is using instructions for another core.
Could you indicate how you compiled the compiler ?
Also if you could provide the full gcc compile and link commands, we can check if they seem right

woodyplum, 2017-10-24:
hi haugoug, 
    thanks for your reply. 

    i just compile the toolchain within the toolchain database which download the same time with the pulpino, following the instruction with the readme.md.  And the compile process report no Error, at least it work for the RI5CY core configuration. 
   after that set the environment RISCV with install directory and add the PATH with install/bin, seems it all be OK for the RI5CY core configuration: cmake_configure.riscv.gcc.sh. the helloworld and other tests all can be run correctly. 
   I have changed nothing in the database from GIT except adding the waveform dump procedure in the tb.sv. 
  
  While when using the cmake_configure.microriscy.gcc.sh script for micro-ri5cy, it report no problem at compile stage, but the simulation hang, thus the illegal instruction been found.  the gcc and link command not changed as the database. 
   how should I find the log for these commands configuration? 

regards,

woodyplum, 2017-10-24:
re-compile the toolchain with 'make MICRORISCY=1', the testcase passed. 
Then for the 4 released configuration i need 4 separated toolchain installation, is that true? 
Why can not just one toolchain and different compile options?   thanks.

haugoug, 2017-10-24:
Yes indeed you need one per configuration. That's just a limitation of the current toolchain as the multilib support has not been activated , but this will be fixed soon.

woodyplum, 2017-10-25:
Thanks, got it.

closed 2017-10-25 by woodyplum