pulp-platform/pulpino #130: Errors thrown when analyze done on rtl files in design_vision

saranprasad, 2017-10-25:
Error is thrown on files wen tried to execute analyze in design vision

Example of error is as below.

```
Compiling source file /home/prasar00/project/pulpino/rtl/components/sp_ram.sv
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:18: Syntax error at or near token 'clk'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:20: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:20: Syntax error at or near token 'en_i'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:21: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:21: Syntax error at or near token 'addr_i'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:22: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:22: Syntax error at or near token 'wdata_i'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:23: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:23: Syntax error at or near token 'rdata_o'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:24: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:24: Syntax error at or near token 'we_i'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:25: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/sp_ram.sv:25: Syntax error at or near token 'be_i'. (VER-294)
Compiling source file /home/prasar00/project/pulpino/rtl/components/rstgen.sv
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:15: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:15: Syntax error at or near token 'rst_ni'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:18: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:18: Syntax error at or near token 'test_mode_i'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:21: Redeclaration of port 'logic' is not allowed. (VER-121)
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:21: Syntax error at or near token 'rst_no'. (VER-294)
Error:  /home/prasar00/project/pulpino/rtl/components/rstgen.sv:22: Redeclaration of port 'logic' is not allowed. (VER-121)
```

FrancescoConti, 2017-10-25:
There are two problems here:
1. you are using the `analyze` without any flag to indicate that the file is in SystemVerilog, hence it doesn't recognize `logic` as a keyword.
2. most importantly, `sp_ram` is not supposed to be synthesized at all. See issue #114 for details.

closed 2017-10-25 by FrancescoConti

saranprasad, 2017-10-26:
Hi,

One help. How can I understand which files need to be synthesized (Analysed) from rtl, ips folders logically otherthan ram part. I am getting lot of warnings, unresolved refernces etc after running a script for synthesis. Plus area for cells shown as 0 even after showing a lot of cells

Thanks
Saran

gautschimi, 2017-10-26:
the systemverilog files in rtl/components/* are just simulation models. 
you should not analyze those files but you should instantiate your own 
memory macros, standard cells, etc. inthere

saranprasad, 2017-10-27:
Ok.
I am starting with ips. and using interactive way to execute commands sequentially (not gui).

```
dc_shell> analyze -library work -format sverilog {/home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv}
Running PRESTO HDLC
Compiling source file /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:49: symbol C_TAG must be a constant or parameter. (VER-260)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:64: symbol C_TAG must be a constant or parameter. (VER-260)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:113: symbol C_FLAG must be a constant or parameter. (VER-260)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:149: symbol C_TAG must be a constant or parameter. (VER-260)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:150: symbol C_TAG must be a constant or parameter. (VER-260)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:49: invalid symbol C_TAG found in expression. (VER-255)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:64: invalid symbol C_TAG found in expression. (VER-255)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:113: invalid symbol C_FLAG found in expression. (VER-255)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:149: invalid symbol C_TAG found in expression. (VER-255)
Error:  /home/prasar00/project/pulpino/ips/fpu/hdl/fpu_v0.1/fpu_shared.sv:150: invalid symbol C_TAG found in expression. (VER-255)
*** Presto compilation terminated with 10 errors. ***
0
dc_shell> 
```

saranprasad, 2017-10-27:
```
dc_shell> analyze -library work -format sverilog /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv
Running PRESTO HDLC
Compiling source file /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv
Error:  /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv:103: The construct 'string' is not supported in synthesis. (VER-700)
Warning:  /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv:641: The construct 'class declaration in this context' is not supported in synthesis; it is ignored. (VER-708)
Error:  /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv:641: Syntax error at or near token '='. (VER-294)
Error:  /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv:642: Syntax error at or near token '='. (VER-294)
Warning:  /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv:666: The construct 'final' is not supported in synthesis; it is ignored. (VER-708)
Error:  /home/prasar00/project/pulpino/ips/riscv/riscv_tracer.sv:742: Syntax error at or near token 'new'. (VER-294)
*** Presto compilation terminated with 4 errors. ***
0
```

saranprasad, 2017-10-27:
One more thing.
how does analyze work. When somultaneously, there is an error and when executed alone, there is no error.
```
dc_shell> analyze -library work -format sverilog {/home/prasar00/project/pulpino/ips/riscv/riscv_register_file_latch.sv /home/prasar00/project/pulpino/ips/riscv/riscv_register_file.sv}
Running PRESTO HDLC
Compiling source file /home/prasar00/project/pulpino/ips/riscv/riscv_register_file_latch.sv
Compiling source file /home/prasar00/project/pulpino/ips/riscv/riscv_register_file.sv
Error:  /home/prasar00/project/pulpino/ips/riscv/riscv_register_file.sv:27: Module 'riscv_register_file' is redefined. (VER-25)
*** Presto compilation terminated with 1 errors. ***
0
```

saranprasad, 2017-10-27:
```
Warning:  /home/prasar00/project/pulpino/ips/riscv/riscv_prefetch_L0_buffer.sv:569: The construct 'assert property' is not supported in synthesis; it is ignored. (VER-708)
```

saranprasad, 2017-10-27:
```
Warning:  /home/prasar00/project/pulpino/ips/riscv/riscv_prefetch_L0_buffer.sv:569: The construct 'assert property' is not supported in synthesis; it is ignored. (VER-708)

Warning:  /home/prasar00/project/pulpino/ips/riscv/riscv_prefetch_L0_buffer.sv:569: The construct 'assert property' is not supported in synthesis; it is ignored. (VER-708)

Warning:  /home/prasar00/project/pulpino/ips/riscv/riscv_prefetch_L0_buffer.sv:569: The construct 'assert property' is not supported in synthesis; it is ignored. (VER-708)
```

be4web, 2017-10-27:
Your style of using the issue function makes it virtually impossible to be answered meaningful. That said (in issue order):

1. You are probably missing a package. While System Verilog makes the impression that compile order doesn't matter this is not the case. Packages need to be compiled before the modules which implement it. Have a look at the sourcecode.

2. riscv-tracer is a simulation module and can't be synthesised (classes can't be synthesised in general by DC)

3. You have to choose **either** the register file based on flip-flops or latches. The module name is the same and you can decide which one to use when analysing.

4. Can be ignored or you put a ifdef around the assertion.
5. Same.

FrancescoConti, 2017-10-27:
Hi, first of all see issue #114 as many of your issues are similar.

Second, some of the files you are analyzing (e.g. `riscv_tracer.sv`) are not meant for synthesis but for simulation.
In general, the same `ipstools` tool that is used to generate simulation scripts can be used to automatically generate `analyze` scripts, based on a simple config file which is in the root of every IP (`src_files.yml`). You can use those scripts as a guideline.

Moreover, while we can help we do not give support for synthesis of PULPino on actual silicon at the moment, as we assume that anyone that wants to do an ASIC synthesis should have the basic competence to do this on his own, and will probably want to adapt PULPino anyways. Hence at the moment you have to write scripts adapted to your particular scenario on your own.
That could change in the future, but at the moment the situation is this.

Last but not least, this is meant to be a forum for issues with PULPino, so if you have doubts about how Design Compiler works, I suggest you [google for it](https://www.google.it/search?ei=dSvzWYvoCan76AT6y76wDw&q=synopsys+design+compiler+tutorial&oq=synopsys+design+compiler+tutorial)

saranprasad, 2017-10-27:
Hi,

Really thanks for the infos and time actually.

regarding 114 -> I understand that i need a library for memory part. Right now it is not available.

Regarding warnings and errors -> Expected ips to be totally synthesized. Got lot of warnings and errors. So, got confused.

DC -> Trying to understand better. Was following a script suggested through #114 itself
https://wustl.app.box.com/v/compiledctcl

saranprasad, 2017-11-03:
Hi,
I want to do analysis and i want to use sp_ram for ram. What all changes are required. I changed in sp_ram_wrap and changed the parameters in components/sp_ram.sv as both side parameters match. The tool crashes when executed compile_ultra when processing ram

```
  Beginning Pass 1 Mapping
  ------------------------
  Processing 'sp_ram_ADDR_WIDTH15_DATA_WIDTH32_NUM_WORDS32768_1'
dcfarm: detected bad exit for job 3004, name J1 -- terminating all child jobs now...
dcfarm: destroying parent process due to child fatal
```

deepika-s517, 2022-09-07:
Hi,
I am getting the same fatal error while running the DC shell synthesis. kindly let know the solution for this below error. 
Beginning Pass 1 Mapping
  ------------------------
  Processing 'fc'
dcfarm: detected bad exit for job 5174, name J1 -- terminating all child jobs now...
dcfarm: destroying parent process due to child fatal