pulp-platform/pulpino #134: vcompile error

javiermacotela, 2017-11-07:
Hi, after I run the ./cmake_configure.riscv.gcc.sh I try to run make vcompile and I get the following error: 

/bin/sh: 1: tcsh: not found
CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 127
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2

How can I fix this?
Thanks for your help!

be4web, 2017-12-03:
My guess would be to install `csh`

closed 2017-12-03 by be4web

vedanjalip, 2020-05-10:
I have installed csh and exported it to the path but still facing the same error.