pulp-platform/pulpino #136: Internal Logic Analyzer on PULPino

Aregaard, 2017-11-28:
Is it possible to implement an Internal Logic Analyzer into the PULPino. Doing a project and want to analyze the actual power consumption and measure specific signals. Like "Report Power" in Vivado. Do you have a suggestion on a solution?

Best Regards
Oskar

FrancescoConti, 2017-11-28:
It's not clear to me what is the issue here... Have you tried and _not_ been able to use ILAs on PULPino?

Aregaard, 2017-11-30:
Sorry for putting up a question without really telling whats the problem. 

I was able to upload our modified PULPino, bitstream file to the Zync Z-7010 board.  To run our test-program we use your guide; Linux on the ARM core and the PULPino as peripheral.

But now I want to upload the PULPino to Zync and the through the ILA run the C-program. While the C-program is running, we want to extract the wave form from the ILA.

Two things I have tried so far:

The First attempt: Include all the files in Vivado (flat hiearchy). (Add files manually into a project). But it generates a lot of errors (trying to fix these errors right now). So maybe this is not a good way of doing it.

The second attempt: Include the ILA-IP in the top_pulpino, modify the scripts, before doing the "make -all" in the FPGA-folder. When the bitstream is generated use vivado to upload it to the FPGA. (Have not tried this yet).

So everything boils down to: Do you prefer one way above another?

FrancescoConti, 2017-11-30:
Ok now I understand better :) in general between the two ways you propose I would go for the second. But more specifically, let me explain an approach to the problem which might be simpler. For the sake of honesty, I have not tried this on PULPino, but I have done it multiple times on our internal projects which use the same IPs and flow and are typically more complex, so I believe it should work the same way.

1. Do a full synthesis run using our scripted flow.
2. Go in `fpga/pulpemu`, where you should find a `pulpemu.xpr` Vivado project. Open it with the Vivado GUI.
3. Open the post-synthesis, pre-implementation view. Add the debug probes directly there via the GUI.
4. Run the implementation. Vivado will complain that the synthesis is out-of-date, *ignore that* and proceed with the place & route and bitstream generation without performing the synthesis again.
5. You should be able to use whatever means you prefer to download the bitstream, and still be able to access the ILA.

closed 2017-12-03 by be4web