pulp-platform/pulpino #137: AXI and APB Interfaces

amohamme, 2017-12-03:
We have a module that has an AXI and APB interface. Is it possible to connect this module to PULPino via its AXI interconnect or APB bus ?

We need to have something similar to the figure below:

[image]

be4web, 2017-12-03:
Yes its possible.

closed 2017-12-03 by be4web

amohamme, 2017-12-04:
Thanks for your reply.
Is there is examples for such cases ? What changes are required (HW/SW) ?

f2013619, 2018-01-16:
@amohamme I am also trying to connect an external module to the axi bus. Can you tell me how you did it? Specifically how you updated the files. I have tried doing it but the module is working correctly but its not taking the inputs from the core.
It would be a great help. Thanks in advance.