pulp-platform/pulpino #139: Booting from SD-card 16GB.

Aregaard, 2017-12-03:
When I run minicom from Ubuntu to the FPGA I get this.

## Booting kernel from Legacy Image at 02080000 ...                             
   Image Name:   Linux-3.18.0-xilinx-dirty                                      
   Image Type:   ARM Linux Kernel Image (uncompressed)                          
   Data Size:    3479888 Bytes = 3.3 MiB                                        
   Load Address: 00008000                                                       
   Entry Point:  00008000                                                       
   Verifying Checksum ... OK                                                    
## Flattened Device Tree blob at 02000000                                       
   Booting using the fdt blob at 0x2000000                                      
   Loading Kernel Image ... OK                                                  
   Loading Device Tree to 1f335000, end 1f33a817 ... OK                         

Starting kernel ...                                                             

��n�+%쵨�,ٵ+�hr/rŅ0,P���vؽr�| �孂����z�*ť-�(���-Ƕ��m�i혷n�(��؇o�m�+�k׭     
�yv;v�Pxt                                                                                
v

What error have I made to create this?

I have made the SD-card bootable with this guide www.wiki.xilinx.com/Prepare+Boot+Medium. No errors at all.
Note: I have created a module that is attached to the APB on the PULPino. Everything according to your flow are without errors.

be4web, 2017-12-03:
This seems to be a mismatch in the baudrate of your UART as soon as Linux starts. I would say you double check your baudrate settings in the device tree. This is not an issue related to PULPino though as it is entirely on the ARM/Xilinx side.

closed 2017-12-03 by be4web

Aregaard, 2017-12-05:
Solution: No it was not the Baudrate. The problem was in the fpga folder "make all"; Be very careful so you set the variables correct. You can also double check your devicetree file, to see what the Zync actually got in baudrate.