pulp-platform/pulpino #142: AXI Interface

amohamme, 2017-12-10:
We have a module that has an AXI interface. It is possible to connect this module to PULPino via its AXI interconnect. How can this AXI peripheral added to the PULPino AXI interconnect. Is there are examples for similar designs ?

We need to have something similar to the figure below:

[image]

Aregaard, 2017-12-11:
Well first you will need to modify the AXI and APB. Check the data cheet for address space that is free. In the PULPino files (dont remember the name of the file) you can find the specific addresses.
Then you will need to write a slave and master to connect to the APB/AXI. (A FSM perhaps?)

Check ARMs documentation about APB and AXI.
In the documentation you can see how the signals are used.

FrancescoConti, 2017-12-11:
At the end of `pulpino_top.sv`, there is the instantiation of the AXI interconnect (`axi_node_intf_wrap`). You have to modify the parameters there, in particular `NB_SLAVE` and the start/end addresses of the slave you want to add.

amohamme, 2017-12-20:
Is there is a limitation on the maximum number of slaves to be added to the AXI Interconnect ?

I was able to add one AXI slave by setting the NB_SLAVE to 4 and adding the start/end corresponding address. This is working fine.
However, when I add one more AXI slave (i.e. NB_SLAVE = 5), it doesn't work and the processor hangs when running the C application. It seems not executing the fist line of the C code.

Any advice please ? Also, is there is any documentation (more than the PULPino datasheet) describing the  AXI Interconnect master/slaves connections and operation with the core ?

be4web, 2017-12-20:
There is no theoretical limit on the number of slaves but a practical one until you can't reach your timing requirements. The AXI standard is pretty extensively documented by ARM itself.

Regarding your problem: It sounds like you have a problem with your memory map (start and end address - maybe accessing two slaves at the same time?). You should double check that and also the address you are requesting.

amohamme, 2017-12-20:
Thanks for the quick response, appreciated.
 
Actually, I don't mean the AXI documentation itself. But I mean documentation for the PULPino usage of the AXI interconnect. How are the masters/slaves connected to the core and what are the masters/slaves connectivity, who is controlling who ?

I am about connecting external module to pulpino. This external module has AXI interface (master/slave).  The master interface will make register reads and writes to a register space inside pulpino. The slave will be controlled by pulpino (from the c code). Below is how it looks like

[image]

**Regarding the issue, let me share more details here. I have two cases:
1- The case of 4 masters/slaves is working fine. 
This is the code added on the pulpino_top.sv file**

> parameter [31:0]    INST_MEM_START      = 32'h0000_0000,
>                     INST_MEM_END        = 32'h000F_FFFF,
>                     DATA_MEM_START      = 32'h0010_0000,
>                     DATA_MEM_END        = 32'h001F_FFFF,
>                     PERIPH_MEM_START    = 32'h1A10_0000,
>                     PERIPH_MEM_END      = 32'h1A1F_FFFF,
>                     AXI_MAST_MEM_START  = 32'h1A20_0000,
>                     AXI_MAST_MEM_END    = 32'h1A2F_FFFF;
> 
> axi_node_intf_wrap
>   #(
>     .NB_MASTER      ( 4                    ),
>     .NB_SLAVE       ( 4                    ),
>     .AXI_ADDR_WIDTH ( `AXI_ADDR_WIDTH      ),
>     .AXI_DATA_WIDTH ( `AXI_DATA_WIDTH      ),
>     .AXI_ID_WIDTH   ( `AXI_ID_MASTER_WIDTH ),
>     .AXI_USER_WIDTH ( `AXI_USER_WIDTH      )
>   )
>   axi_interconnect_i
>   (
>     .clk       ( clk_int    ),
>     .rst_n     ( rstn_int   ),
>     .test_en_i ( testmode_i ),
> 
>     .master    ( slaves     ),
>     .slave     ( masters    ),
> 
>     .start_addr_i ( { AXI_MAST_MEM_START, PERIPH_MEM_START, DATA_MEM_START, INST_MEM_START } ),
>     .end_addr_i   ( { AXI_MAST_MEM_END  , PERIPH_MEM_END  , DATA_MEM_END  , INST_MEM_END   } )
>   );
> 

**2- The case of 5 masters/slaves is not working.
This is the code added on the pulpino_top.sv file**

> parameter [31:0]    INST_MEM_START      = 32'h0000_0000,
>                     INST_MEM_END        = 32'h000F_FFFF,
>                     DATA_MEM_START      = 32'h0010_0000,
>                     DATA_MEM_END        = 32'h001F_FFFF,
>                     PERIPH_MEM_START    = 32'h1A10_0000,
>                     PERIPH_MEM_END      = 32'h1A1F_FFFF,
>                     AXI_MAST_MEM_START  = 32'h1A20_0000,
>                     AXI_MAST_MEM_END    = 32'h1A2F_FFFF,
> 		    AXI_SLAVE_MEM_START = 32'h1A30_0000,
>                     AXI_SLAVE_MEM_END   = 32'h1A3F_FFFF;
> 
> axi_node_intf_wrap
>   #(
>     .NB_MASTER    ( 5                    ),
>     .NB_SLAVE       ( 5                    ),
>     .AXI_ADDR_WIDTH ( `AXI_ADDR_WIDTH      ),
>     .AXI_DATA_WIDTH ( `AXI_DATA_WIDTH      ),
>     .AXI_ID_WIDTH   ( `AXI_ID_MASTER_WIDTH ),
>     .AXI_USER_WIDTH ( `AXI_USER_WIDTH      )
>   )
>   axi_interconnect_i
>   (
>     .clk       ( clk_int    ),
>     .rst_n     ( rstn_int   ),
>     .test_en_i ( testmode_i ),
> 
>     .master    ( slaves     ),
>     .slave     ( masters    ),
> 
>     .start_addr_i ( { AXI_SLAVE_MEM_START, AXI_MAST_MEM_START, PERIPH_MEM_START, DATA_MEM_START, INST_MEM_START } ),
>     .end_addr_i   ( { AXI_SLAVE_MEM_END, AXI_MAST_MEM_END  , PERIPH_MEM_END  , DATA_MEM_END  , INST_MEM_END   } )
>   ); 
> 

**So, I think the address seems to be correct. Also, in the c code I am not trying to access the AXI. It is just empty C code with printf. However, even the printf is not working and the code seems hanging due to the addition of the 5th AXI. Once this 5th AXI is removed, the code works fine. Below is the C code used.**

> #include <stdio.h>
> #include "axi.h"
> #include "spi.h"
> 
> int main()
> {
>   printf("==== Start ====\n");
>   printf("==== End ====\n");
>   
>   return 0;
> }

Appreciating your feedback.

be4web, 2017-12-21:
Tbh, I don't really see the issue regarding the block diagram. It basically answers all your questions, doesn't it? The Master -> Slave connectivity is indicated by the arrow head. And you also properly added your module. So maybe I am not getting your question here.

The added AXI Module looks fine. Are you sure that your module is behaving correct? Or did you use a Xilinx Module?  Do you have an RTL simulation set-up?

FrancescoConti, 2017-12-21:
Let's assume that the RTL is not "wrong" in the sense that even when the 5th AXI is added, it's still fully  synthesizable with no particular warnings/issues (you should check this).
Moreover, from the code you have above it looks like the 4-AXI case already realizes what you  have in the diagram (one additional master and one additional slave), while the 5-AXI case would add an entirely new external module with its own slave and master ports. If it is so, then the names `AXI_MASTER_MEM_*` and `AXI_SLAVE_MEM_*` for the memory region parameters are a bit misleading. If this is not what you want, then check your code.

With these assumptions, the behavior you report could be explained as follows.
Since the standard output happens through a peripheral mapped in the `0x1a10_0000` - `0x1a20_0000` region, at the end of the day its basic transactions will pass through the interconnect in question - so a problem with one of the AXI M/S modules connected to it which causes the interconnect to deadlock will also freeze execution when you do a `printf`.

A few suggestions to check what is happening:
1. as @be4web suggested, set up an RTL simulation and if you have more info to share, we'll be happy to help
2. check the AXI interface handshakes for issues ( AR -> R ; AW -> W -> B )
3. check the AXI IDs, which are a  typical source of headaches

Hope this helps!

amohamme, 2017-12-21:
Thanks @be4web and @FrancescoConti for your response.

I have checked all that and had a code review again. The issue is not related to the connected external module actually. As it doesn't make any transactions at all now to get the system up. So, the point for me was more related to the settings of the AXI Interconnect related to the Master addition.

@FrancescoConti all your points are valid and the issue is related to it:
1. I have RTL simulation setup 
2. I had to hard wire the R and B responses (the ready) signals to one pulse 1 not to get the AXI hanging waiting for handshaking.
3. The issue was mainly related to the AXI ID used. I have to set the AXI_SLAVE_ID_WIDTH to 5. To accommodate for the 5th added AXI module (Master/Slave).

So, the issue is resolved now after the fixes on 2 and 3.

I have a point here, would it better to get those ID set automatically and calculated inside the code from the NB_SLAVE/NB_MASTER numbers ?

Appreciating your reply and support. This was helpful.

Thanks
Ahmed

f2013619, 2018-01-17:
@amohamme I am also trying to connect an external module to the axi bus. Can you tell me how you did it? Specifically how you updated the files. I have tried doing it but I am getting the following error:
 ** Note: (vsim-3812) Design is being optimized...
# ** Fatal: Unexpected signal: 11.
# ** Error: Vopt Compiler exiting
# Error loading design
It would be a great help. Thanks in advance.

FrancescoConti, 2018-01-17:
Your error means your design cannot be linked, most likely because of some error. Try to add the `-novopt` flag to your `vsim` call; this disables all optimizations, which makes the simulation very slow, but can typically turn opaque errors such as the one you get into much clearer messages.

be4web, 2018-01-18:
Unfortunately, this is the type of error you get if you mess up with SV interfaces. Most probably because you have an unconnected interface port (which is not possible with Questa).

amohamme, 2018-01-18:
@f2013619 Please check @FrancescoConti  and @be4web feedback on the error logged. You first need to fix the simulation environment, scripts, system verilog files and connections and make sure that the simulation is loading. After that we can check the AXI interface and I will advise for what changes are needed to get it working.

FrancescoConti, 2018-01-18:
@amohamme by the way I never replied to this question:
> I have a point here, would it better to get those ID set automatically and calculated inside the code from the NB_SLAVE/NB_MASTER numbers ?
I actually agree with you, that's an idea we have been considering for future releases.

f2013619, 2018-01-18:
Hi, with the .-novopt command I am getting the following error: Error: (vsim-3839) /home/choudhary/pulpino/vsim/..//rtl/axi2apb_wrap.sv(113): Variable '/tb/top_i/slaves[3]/aw_ready', driven via a port connection, is multiply driven. See /home/choudhary/pulpino/vsim/..//rtl/axi_aes_wrap.sv(83).
#         Region: /tb/top_i/peripherals_i/axi2apb_i/genblk1/axi2apb_i
# ** Error: (vsim-3839) /home/choudhary/pulpino/vsim/..//rtl/axi2apb_wrap.sv(113): Variable '/tb/top_i/slaves[3]/w_ready', driven via a port connection, is multiply driven. See /home/choudhary/pulpino/vsim/..//rtl/axi_aes_wrap.sv(83).
Can you help me with how to remove this error?

FrancescoConti, 2018-01-18:
Yep, as we imagined there's a problem with how you're connecting your block to the interconnect. As you see, `-novopt` makes it possible to understand the source of the problem more clearly: you are driving two `ready` signals incorrectly from your own `axi_aes_wrap` block (as suggested at the end of the error message). In the AXI protocol, `ready`s for the AW, AR, and W channels have to be driven from the slave to the master; the ones for the R and B channels from the master to the slave.

f2013619, 2018-01-18:
Hi, The simulation is loading now  but I am not able to get helloworld in console or vsim. Can you please tell me what I should check to correct it?
Thanks in advance

amohamme, 2018-01-18:
Could you please clarify more what is the simulation status ? What modifications have been added ?
a screenshot will be fine as well.

f2013619, 2018-01-18:
I have added a core with a wrapper to the AXI bus. The following are the screenshots of the simulation. I have changed the number of AXI slaves to 4 and updated the address.
[image]
[image]

f2013619, 2018-01-18:
The modelsim transcript gives the following message :There has been a data request but the address is unknown

FrancescoConti, 2018-01-18:
@f2013619 I see quite a lot of warnings for port sizes that do not match with connection sizes related to your module. Try to solve those preliminarily.

f2013619, 2018-01-22:
Hi, I have removed the port size mismatches. Now its shows some new kind of warnings as follows:Error: (vsim-3852) Invalid override of parameter '/tb/TEST' whose value may no longer be modified.
#         Region: /tb
# ** Warning: (vsim-3770) Failed to find user specified function 'mem_init' in DPI C/C++ source files.
#    Time: 0 ps  Iteration: 0  Instance: /tb File: /home/choudhary/pulpino/vsim/..//tb/tb.sv
# ** Warning: (vsim-3770) Failed to find user specified function 'mem_poll' in DPI C/C++ source files.
#    Time: 0 ps  Iteration: 0  Instance: /tb File: /home/choudhary/pulpino/vsim/..//tb/tb.sv
# ** Warning: (vsim-3770) Failed to find user specified function 'mem_push' in DPI C/C++ source files.
#    Time: 0 ps  Iteration: 0  Instance: /tb File: /home/choudhary/pulpino/vsim/..//tb/tb.sv
# ** Warning: (vsim-3770) Failed to find user specified function 'mem_push' in DPI C/C++ source files.
#    Time: 0 ps  Iteration: 0  Instance: /tb File: /home/choudhary/pulpino/vsim/..//tb/tb.sv
Can you please help me out here? Thanks

FrancescoConti, 2018-01-22:
Mmmm, it seems you are changing the value of the `TEST` parameter in the script calling `vsim`. Can you post it here?

f2013619, 2018-01-22:
The following are the contents of run.tcl -
#!/bin/bash
# \
exec vsim -novopt -64 -do "$0"

set TB_TEST $::env(TB_TEST)
set VSIM_FLAGS    "-GTEST=\"$TB_TEST\""

set TB            tb
set MEMLOAD       "PRELOAD"

source ./tcl_files/config/vsim.tcl
And in the tb.sv file -
module tb;
  timeunit      1ns;
  timeprecision 1ps;

  // +MEMLOAD= valid values are "SPI", "STANDALONE" "PRELOAD", "" (no load of L2)
  parameter  SPI            = "QUAD";    // valid values are "SINGLE", "QUAD"
  parameter  BAUDRATE       = 781250;    // 1562500
  parameter  CLK_USE_FLL    = 0;  // 0 or 1
  parameter  TEST           = ""; //valid values are "" (NONE), "DEBUG"
  parameter  USE_ZERO_RISCY = 0;
  parameter  RISCY_RV32F    = 0;
  parameter  ZERO_RV32M     = 1;
  parameter  ZERO_RV32E     = 0;

  int           exit_status = `EXIT_ERROR

f2013619, 2018-01-23:
One of my colleague was also getting the sam etype of error but when he did run -all helloworld is printing. For me it says the following when I execute run-all : There has been a data request but the address is unknown. @amohamme for address what else changes you did other than mentioned above in your comment. It will be a great help. Thanks

FrancescoConti, 2018-01-23:
Try to remove the `set VSIM_FLAGS    "-GTEST=\"$TB_TEST\""` line in the `run.tcl`.

amohamme, 2018-01-23:
Below are the changes added to the pulpino_top.sv file for adding more AXI interfaces:
The important ones are the AXI_ID_SLAVE_WIDTH and address ranges to the AXI interconnect node

`
`define AXI_ADDR_WIDTH         32
`define AXI_DATA_WIDTH         32
`define AXI_ID_MASTER_WIDTH     2
`define AXI_ID_SLAVE_WIDTH      5
`define AXI_USER_WIDTH          1
`define NB_MASTER               5
`define NB_SLAVE                5

/////////////////////////////////////////////////
// Memory Map - Refer to datasheet
parameter [31:0]    INST_MEM_START       = 32'h0000_0000,
                    INST_MEM_END         = 32'h000F_FFFF,
                    DATA_MEM_START       = 32'h0010_0000,
                    DATA_MEM_END         = 32'h001F_FFFF,
                    PERIPH_MEM_START     = 32'h1A10_0000,
                    PERIPH_MEM_END       = 32'h1A1F_FFFF,
                    AXI_MASTER_MEM_START = 32'h1A20_0000,
                    AXI_MASTER_MEM_END   = 32'h1A20_FFFF,
					AXI_SLAVE_MEM_START  = 32'h1A30_0000,
                    AXI_SLAVE_MEM_END    = 32'h1A30_FFFF;

////////////////////////////////////////////////////////////////////////

  //----------------------------------------------------------------------------//
  // Axi node
  //----------------------------------------------------------------------------//

  axi_node_intf_wrap
  #(
    .NB_MASTER      ( `NB_MASTER           ),
    .NB_SLAVE       ( `NB_SLAVE            ),
    .AXI_ADDR_WIDTH ( `AXI_ADDR_WIDTH      ),
    .AXI_DATA_WIDTH ( `AXI_DATA_WIDTH      ),
    .AXI_ID_WIDTH   ( `AXI_ID_MASTER_WIDTH ),
    .AXI_USER_WIDTH ( `AXI_USER_WIDTH      )
  )
  axi_interconnect_i
  (
    .clk       ( clk_int    ),
    .rst_n     ( rstn_int   ),
    .test_en_i ( testmode_i ),

    .master    ( slaves     ),
    .slave     ( masters    ),

    .start_addr_i ( { AXI_SLAVE_MEM_START, AXI_MASTER_MEM_START, PERIPH_MEM_START, DATA_MEM_START, INST_MEM_START } ),
    .end_addr_i   ( { AXI_SLAVE_MEM_END  , AXI_MASTER_MEM_END  , PERIPH_MEM_END  , DATA_MEM_END  , INST_MEM_END   } )
	
  );
`

f2013619, 2018-01-23:
@amohamme As told I have updated the NB_SLAVE and NB_MASTER in axi_node_intf_wrap and in module pulpino_top.sv. I have added a crypto aes core as AXI Slave and have written the axi_aes_wrap for the same. Still the same thing I am getting after run that is---
# There has been a data request but the address is unknown           2834220000
I am not able to understand what I am doing wrong. Can you throw some insights on this as to what might be the resaon for this?

f2013619, 2018-01-24:
Hi,
The simulations are running on the platform now. @amohamme Can you tell me now how to check if the module I have added to the platform is working properly as a slave to the platform.

amohamme, 2018-01-24:
I am not sure about the function of the module you are connecting to the CPU. However, it is working if you are able to do writes/reads through the AXI interface.

f2013619, 2018-01-24:
Thanks for the quick reply. To test that if it is able to do writes/reads through AXI interface what changes should I do in the test module. Can you guide me a little in this. I am new to this actually. It will be a great help. Thanks in advance.

f2013619, 2018-01-29:
Hi,
I am getting helloworld in the console but when I try it in modelsim and add uart parameters to the wave I am not able to see helloworld. Can you suggest something for this.
Thanks in advance.
[image]
[image]

f2013619, 2018-02-01:
@amohamme How do I test if its taking read and write from AXI bus. I am working with a crypto module and the testbench for the module is working properly but  how do I test if the module is connected to AXI bus and working properly. Which files should I edit and how should be the testbench. Can you please help me with this. It will be a great help.

closed 2018-02-25 by FrancescoConti

JeffreyHJF, 2023-11-01:
> Thanks @be4web and @FrancescoConti for your response.
>
> I have checked all that and had a code review again. The issue is not related to the connected external module actually. As it doesn't make any [...]

hello, could you pls tell me how to calculate the ID_Width from the NB?
I do appreciate it, if you could reply that