pulp-platform/pulpino #143: pulpino as SPI slave and master

WaleedOsama, 2017-12-14:
### i have several questions 
1) can i synthesize pulpino using altera tools rather than xillinix ?

2) can i verify and watch signals of the peripherals in modelsim ?

3)  the slave in the memory map does not have registers , so how it works ? can pulpino work as 
    slave SPI with another pulpino(SPI master) ?

2)  what is the flow of Pulpino to work as SPI master ?

here is how i get it and please correct me if i am wrong , using the spi library you provide , i would make this sequence of functions in case of master write

`
spi_setup_master()`
`spi_set_datalen()`
`spi_setup_dummy()`
`spi_setup_cmd_addr()`
`spi_start_transaction()`
`spi_write_fifo()`

so it will not leave before sending number of bits in SPILEN , for cmd ,adr and data .

#### and for slave 

i can not find a function except
'spi_setup_slave()'

and all other function act on registers in memory segment of MASTER SPI , according to datasheet
**so how slave works ?**

FrancescoConti, 2017-12-14:
1. Yes, but we do not provide any support / scripts for Altera devices at the moment.
2. Sure. Why not?
3. The SPI slave gets configured in a slave fashion, i.e. by an external master. So yes, it is possible to connect two PULPino's (one as a master, one as a slave).
4. I'm not an expert on using PULPino's SPI master, but I believe you are correct.  The slave needs no configuration, it is just a receiver.

WaleedOsama, 2017-12-14:
@FrancescoConti  thank you very much

closed 2017-12-14 by WaleedOsama