pulp-platform/pulpino #144: riscv_core module not found while compiling the RTL files using Vivado 2015.1

3adawy, 2017-12-20:
I'm working on my graduation project and I will use pulpino as my open source controller , but I'm facing problems while trying to compile and synthesize the RTL files using Vivado 2015.1 on windows.

and I'm getting the errors in the core_regions file (pic is attached for the error)
[image]

quangdaovu, 2017-12-21:
Seem like you are missing some source file. Check ips_src_files.tcl for more details.

3adawy, 2017-12-21:
I checked my files and this repo. files and I couldn't find a file has this name.
Also I have tried to import the original riscv core from this [repo.]( https://github.com/pulp-platform/riscv) but I found it different somehow.

quangdaovu, 2017-12-21:
Surely you forgot to run ./update-ips.py after clone the repo.

FrancescoConti, 2017-12-21:
@quangdaovu is correct. Additionally, I should add that Windows is not a supported platform for PULPino - we have no idea of which scripts work and which ones don't. For example, it's possible that `./update-ips.py` breaks on Windows.

closed 2017-12-21 by FrancescoConti

3adawy, 2017-12-21:
Thanks for you both.
I have already run the update-ips.py after cloning the repo. but I couldn't find the file too.

can you provide me with a working link for Vivado 2015.1 linux version cuz I can only find the windows version ?!!!

quangdaovu, 2017-12-21:
You can clone the repo and run update-ips.py on linux side and then move all to Windows. That's what I did actually.

You can find Linux version of Vivado here: https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools/archive.html

3adawy, 2017-12-21:
Thanks very much.
but I have tried it a week ago and it didn't allow me to download any version of the SDK due to what they have called "failed export compliance verification".

FrancescoConti, 2017-12-21:
@3adawy that likely means that Xilinx is not allowed to "export" the SDK product to your country. But @quangdaovu's suggestion is actually good: run only update-ips.py on Linux - it's based on open-source software, it won't give you any problem of that nature.

3adawy, 2017-12-21:
Okay.... I did as you told me and after running ./update-ips.py ...
that is the progress on terminal :

> Using remote git server https://github.com, remote is https://github.com/pulp-platform
Cloning into 'ipstools'...
remote: Counting objects: 293, done.
remote: Total 293 (delta 0), reused 0 (delta 0), pack-reused 293
Receiving objects: 100% (293/293), 45.63 KiB | 52.00 KiB/s, done.
Resolving deltas: 100% (196/196), done.
Checking connectivity... done.
Traceback (most recent call last):
  File "./update-ips.py", line 86, in <module>
    import ipstools
  File "/mnt/c/pulp/pulpino/ipstools/__init__.py", line 15, in <module>
    from .IPDatabase import *
  File "/mnt/c/pulp/pulpino/ipstools/IPDatabase.py", line 16, in <module>
    import yaml
ImportError: No module named yaml

then I  copied the repo. files from linux to windows , started a new project on vivado and the same error again :) .

quangdaovu, 2017-12-21:
ImportError: No module named yaml

First you will need python, and after that yaml. Install yaml by using pip install pyyaml.

3adawy, 2017-12-22:
Thanks @quangdaovu .
I did it and it worked.
I have done it totally wrong , I was building the project without using its tcl files.
Now I'm using the tcl files but facing a problem with directories (as I'm not familiar with tcl).
that is the last one I couldn't deal with

>  source tcl/ips_inc_dirs.tcl
>couldn't read file "tcl/ips_inc_dirs.tcl": no such file or directory
 >   while executing
>"source tcl/ips_inc_dirs.tcl"
>    (file "run.tcl" line 62)

quangdaovu, 2017-12-22:
That error happens because of current Vivado working folder is not correct. Considering pulpino repo is cloned to D:/pulpino and you're trying to build xilinx_clock_manager. Type cd "D:/pulpino/fpga/ips/xilinx_clock_manager" in Vivado TCL console. Then you can start to run source tcl/run.tcl. Make can be run on Vivado console by using "exec make", but of course some part of make script will be broken. I think "source tcl/run.tcl" is more than enough, the rest part is easy (copy generated file to ip folder for example).

In case you want to work on other development board, environment variables (like XILINX_PART, XILINX_BOARD...) can be set as Windows environment variables.

3adawy, 2017-12-23:
First of all thanks for all your generous help, and excuse me if I ask too much as I spent about 2 weeks trying to open the project only.
It worked and opened the project but with no source files at. what do you mean by the generated file to be copied  , and about the " ip folder " did you mean that one at main directory of the repo. or what ?!!.

is that process is documented somewhere that I can follow it to build and synthesize this project ?!!

quangdaovu, 2017-12-23:
Better contact me via email. My email is azkan@bka.vn.