pulp-platform/pulpino #145: make vcompile error

saranprasad, 2017-12-22:
Hi,

on executing make vcompile on newly cloned pulpino, the below error is shown.

```
Compiling component:  apb_uart 

** Error: Library std not found.
** Error: VHDL Compiler exiting

make[3]: *** [CMakeFiles/vcompile] Fehler 1
make[2]: *** [CMakeFiles/vcompile.dir/all] Fehler 2
make[1]: *** [CMakeFiles/vcompile.dir/rule] Fehler 2
make: *** [vcompile] Fehler 2
prasar00@i80pc125:~/project/pulpino_postsyn/pulpino/sw/
```

saranprasad, 2017-12-22:
hi,

copied the modelsim.ini from other place and the issue is solved. Closing the issue.

closed 2017-12-22 by saranprasad

riteish4141, 2018-01-09:
@saranprasad Hi,
I am having the same problem, can you please explain what you did to solve the problem ? copied modelsim.ini from which place and where ?