pulp-platform/pulpino #148: Regarding Advanced Debug Unit interface

thotaraghava, 2018-01-04:
HI Designer,

   I could see below line of commented text just above adv_dbg_if instantiation in "core_region.sv" module.
   // TODO: remove the debug connections to the core

  Q1. What does it mean to design.
  Q2. What is this adv_dbg_if mean to. 
  Q3. I could see "debug" interface between "peripheral" module and "core_region" module.  I think this is actual debug interface. I am trying to replace cortex-m3 processor in my design with PULPino. I suppose my design debug interface should be directly connected to this core_region debug interface.

   I request your patience to clarify all the above queries.

Thanks,
Raghav

quangdaovu, 2018-01-04:
As I'm currently working on making JTAG works with PULPino, I think I can answer some of your questions. And please correct me if I'm wrong.

A1. adv_dbg_if has 2 interfaces. One is AXI4 interface that connects to AXI4 bus of PULPino, the other one used to connect to CPU debug port. In previous version of PULPino, the core itself is connected to adv_dbg_if through debug ports. But since current release, all of core's register is memory mapped, so debug port is not needed anymore. We can easily read and write to any register using AXI4 interface.
You can refer to this: https://github.com/pulp-platform/pulpino/issues/39

A2. Of course it's debug interface.
A3. I have no idea about that debug if. Guess it's currently in use in current FPGA system, since debug_bridge use SPI interface to debug the core.

thotaraghava, 2018-01-04:
HI,

  Thanks for your response. Its regarding your answer A1.
  adv_dbg_if is having two interfaces, one is AXI4 and other is JTAG. So, I should use this JTAG for debugging my core and the other one I was mentioning between core and peripheral is not for debug purpose.  

Thanks

quangdaovu, 2018-01-04:
Hi,

You misunderstood me. Adv_dbg_if has only one input interface, and it's JTAG as you can see on HDL code. On the other side (output), it can connect to AXI bus and CPUs, that's 2 interfaces I refered to. You can refer to adv_dbg_if manual on ips/adv_dbg_if/doc for better understanding about this IP. 

Surely if you want to replace your CM3 with PULPino then you have to either use SPI (as the same as current FPGA system) or JTAG (through adv_dbg_if) to debug PULPino core. 

Regard,

be4web, 2018-01-04:
Just adding to Q3: The `debug` bus you saw is basically just a request, grant protocol similar to that used in the core's memory interfaces (for more details you can see the core's [datasheet](http://www.pulp-platform.org/documentation/) - LSU section). The AXI transactions issued by the `adv_dbg_if` are translated to this more simple protocol (just not to have a full blown AXI on the core just for debug ;-)).

thotaraghava, 2018-01-04:
Hi, 

 Right, adv_dbg_if is having two interfaces namely cpu and axi. But cpu ports are either left open or tied to constants. What does it mean.
 Second thing is regarding your latest update (thanks for that !!), I understand that axi interface of adv_dbg_if has to be mapped to this simple req/gnt protocol and connect them.

Thanks

thotaraghava, 2018-01-04:
And I don't see anything in ips directory.

quangdaovu, 2018-01-05:
Hi,

Yes it's related to #39, and it's your Q1. And yes you may forgot to run ./update-ips.py after clone the repo.

thotaraghava, 2018-01-05:
Hi,
  When I am trying to do ./update-ips.py, getting below error.
fatal: Not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
File "./update-ips.py", line 65, in <module>
[server, group, remote] = find_server()
File "./update-ips.py", line 53, in find_server
print tcolors.ERROR + "ERROR: could not find remote server." + tcolors.END
NameError: global name 'tcolors' is not defined
------------

Your help is appreciated highly.

Regards,
Raghav

thotaraghava, 2018-01-05:
Hi,

  Regarding one of the interface of adv_dbg_if, which is CORE, can I still connect directly to core_region for debugging purpose. I understand, since all registers are memory mapped, we can connect through AXI if.
But if I want still direct, will it works.

Thanks

thotaraghava, 2018-01-10:
[image]

Hi @quangdaovu , 
      In the above image, core to adv_dbg_if interface is highlighted by arrow mark.
     Can we still use this connection for debugging is my query.

Thanks,

thotaraghava, 2018-01-10:
@quangdaovu , any comment

be4web, 2018-01-10:
No, you can't use it anymore.

thotaraghava, 2018-01-10:
thank you very much

closed 2018-01-10 by be4web