pulp-platform/pulpino #149: Regarding PULPino RTL files

thotaraghava, 2018-01-11:
HI all,

    I do not have internet access for my VWP,  hence I need to download to Windows and then transfer.
    With regards to RTL files, all files are here in this web browser or else we need to update in Linux environment to get IPs.

   Please help in this regard.
Thanks,

be4web, 2018-01-11:
I am really sorry but I have no idea what you are actually asking for... You can always checkout the repo, update the ips and transfer the whole sourcecode.

closed 2018-01-11 by be4web

thotaraghava, 2018-01-11:
How can I update repo in Windows machine after cloning through GitHub Desktop.

any idea.

be4web, 2018-01-11:
We do not support windows. You might be able to just run `./update-ips.py` without modifications.

thotaraghava, 2018-01-11:
Does this command generates any more RTL files for IPs.

be4web, 2018-01-11:
Yes, this command checks out all necessary IPs from github.

thotaraghava, 2018-01-11:
ohh ok. Thanks