pulp-platform/pulpino #150: make vcompile issue

riteish4141, 2018-01-11:
Hello,
I am facing error when I do make vcompile:

/home/riteish/2_OpenRISC/pulpino-master/vsim/..//vsim/vcompile/vcompile_ips.csh: No such file or directory.
CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 1
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2

I had a look at other issues and found that vcompile_ips.csh is generated after ./update-ips.py and ./generate-scripts.py
So I did ./update-ips.py and found this error:

riteish@riteish-ThinkPad-X1-Carbon-2nd:~/2_OpenRISC/pulpino-master$ ./update-ips.py 
Traceback (most recent call last):
  File "./update-ips.py", line 65, in <module>
    [server, group, remote] = find_server()
  File "./update-ips.py", line 53, in find_server
    print tcolors.ERROR + "ERROR: could not find remote server." + tcolors.ENDC
NameError: global name 'tcolors' is not defined

So i commented the line 53 in update-ips.py file and it was fine.
Then when I did ./generate-scripts.py file i had the issue:

riteish@riteish-ThinkPad-X1-Carbon-2nd:~/2_OpenRISC/pulpino-master$ ./generate-scripts.py 
Cloning into 'ipstools'...
git@iis-git.ee.ethz.ch's password: 

I learnt from other closed issues that I need not password for it.
Can someone please help me with this issue ?

Another thing I learnt from other closed issues was that i must reference the path to Modelsim in the configure scipts, so I changed the line to 
#VSIM=`which vsim`
VSIM=/home/riteish/altera/13.1/modelsim_ase/bin

in all my configure scripts inside the /pulpino/sw/build folder
is it fine what i am doing or some other way exists ?

riteish4141, 2018-01-12:
Hello,

I saw somewhere in the closed issues where it ws asked did you clone or download the repo.
So i cloned it and then the problem was gone.
But now I am having another problem when doing make vcompile:

riteish@riteish-ThinkPad-X1-Carbon-2nd:~/exp/pulpino/build$ sudo "PATH=$PATH" make vcompile
[sudo] password for riteish: 

--> Compiling PULPino Platform... 

--> Compiling PULP IPs libraries... 
--> Compiling axi_node... 
Modifying modelsim.ini
Compiling component:  axi_node 

--> axi_node compilation complete! 
--> Compiling apb_node... 
Modifying modelsim.ini
Compiling component:  apb_node 

--> apb_node compilation complete! 
--> Compiling axi_mem_if_DP... 
Modifying modelsim.ini
Compiling component:  axi_mem_if_DP 

--> axi_mem_if_DP compilation complete! 
--> Compiling axi_spi_slave... 
Modifying modelsim.ini
Compiling component:  axi_spi_slave 

--> axi_spi_slave compilation complete! 
--> Compiling axi_spi_master... 
Modifying modelsim.ini
Compiling component:  axi_spi_master 

--> axi_spi_master compilation complete! 
--> Compiling apb_uart_sv... 
Modifying modelsim.ini
Compiling component:  apb_uart_sv 

--> apb_uart_sv compilation complete! 
--> Compiling apb_gpio... 
Modifying modelsim.ini
Compiling component:  apb_gpio 

--> apb_gpio compilation complete! 
--> Compiling apb_event_unit... 
Modifying modelsim.ini
Compiling component:  apb_event_unit 

--> apb_event_unit compilation complete! 
--> Compiling apb_spi_master... 
Modifying modelsim.ini
Compiling component:  apb_spi_master 

--> apb_spi_master compilation complete! 
--> Compiling fpu... 
Modifying modelsim.ini
Compiling component:  fpu 

--> fpu compilation complete! 
--> Compiling apb_pulpino... 
Modifying modelsim.ini
Compiling component:  apb_pulpino 

--> apb_pulpino compilation complete! 
--> Compiling apb_fll_if... 
Modifying modelsim.ini
Compiling component:  apb_fll_if 
[... 180 lines cut ...]

--> apb2per compilation complete! 
--> Compiling PULPino... 
Modifying modelsim.ini
Compiling component:  PULPino 

 Compiling for RISCV core 
** Warning: /home/riteish/exp/pulpino/vsim/..//rtl/random_stalls.sv(72): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Warning: /home/riteish/exp/pulpino/vsim/..//rtl/random_stalls.sv(116): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
** Error: /home/riteish/exp/pulpino/vsim/..//rtl/random_stalls.sv(183): Questa has encountered an unexpected internal error: ../../../src/vlog/vgenstmt.c(935) <disable_entry_labelOf(block)>. Please contact Questa support at http://supportnet.mentor.com/

CMakeFiles/vcompile.dir/build.make:57: recipe for target 'CMakeFiles/vcompile' failed
make[3]: *** [CMakeFiles/vcompile] Error 1
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/vcompile.dir/all' failed
make[2]: *** [CMakeFiles/vcompile.dir/all] Error 2
CMakeFiles/Makefile2:175: recipe for target 'CMakeFiles/vcompile.dir/rule' failed
make[1]: *** [CMakeFiles/vcompile.dir/rule] Error 2
Makefile:168: recipe for target 'vcompile' failed
make: *** [vcompile] Error 2

Can anyone please help me in this as I am not able to understand this issue at all.

FrancescoConti, 2018-01-12:
The error  reported by ModelSim is not especially helpful, but the error might bwe due the various warnings on randomization, which are not well managed and trigger an internal bug in the simulator.

```
** Warning: /home/riteish/exp/pulpino/vsim/..//ips/riscv/riscv_controller.sv(855): (vlog-2186) SystemVerilog testbench feature
(randomization, coverage or assertion) detected in the design.
These features are only supported in Questasim.
****** Error: /home/riteish/exp/pulpino/vsim/..//rtl/random_stalls.sv(183): Questa has encountered an unexpected internal error: ../../../src/vlog/vgenstmt.c(935) <disable_entry_labelOf(block)>. Please contact Questa support at http://supportnet.mentor.com/
```

Which version of ModelSim are you using?

riteish4141, 2018-01-15:
Hello,
Thank you for your prompt reply.
I am using 13.1 modelsim.
Due to this error I am unable to get the desired result on make vcompile.

FrancescoConti, 2018-01-18:
Mmmm, maybe you are using a version of Modelsim distributed by Altera? As far as I know, the most recent versions of Modelsim should be around version 10.6 (and the student version here https://www.mentor.com/company/higher_ed/modelsim-student-edition is 10.4a).
In any case, your specific version seems not to be supporting the randomization features, which causes the error above. What you can try to do is to comment them out, as they are used only for verification purposes and they are non-essential. Let us know!

closed 2018-01-18 by FrancescoConti

riteish4141, 2018-01-18:
Hello,
The Modelsim student version is for windows i guess.
I am using Modelsim distributed by Altera.
Is there any other way to download the modelsim for ubuntu ?

saurabhgplus, 2019-09-24:
Hi,
As Ritesh suggested, if I commented line number 53 in update-ips.py it start giving below error.

**fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).**

If I run ./generate-scripts.py with or without commenting 53 line nuber in update-ips.py it is asking for a password.
If I just don’t give password and do ctrl+c its giving below error:
**git@iis-git.ee.ethz.ch's password: Traceback (most recent call last):
  File "./generate-scripts.py", line 33, in <module>

    execute("git pull", silent=True)
  File "./generate-scripts.py", line 22, in execute
    return subprocess.call(cmd.split(), stdout=stdout)
  File "/usr/lib/python2.7/subprocess.py", line 522, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 1376, in wait
    pid, sts = _eintr_retry_call(os.waitpid, self.pid, 0)
  File "/usr/lib/python2.7/subprocess.py", line 476, in _eintr_retry_call
    return func(*args)
KeyboardInterrupt**

1) Could you please help me for this mention problem ?
2) Could you give me a password if require?

Regards
Saurabh