pulp-platform/pulpino #151: Updating IPs by running update-ips.py

thotaraghava, 2018-01-12:
Hello all,

     I have downloaded PULPino database onto Windows and trying to update/clone all IPs by running update-ips.py file through WINDOWS command prompt.
    I ended up by getting below error at line 53.
    Error:  Syntax error. Missing parathesis in call to 'print. Did you mean print(int tcolors------------)

Please help.

Thanks

FrancescoConti, 2018-01-12:
1. you are using python3, while the script is meant for python2.
2. I remind you that Windows is not a supported platform for PULPino, so you might anyways end up having other problems.

Just as a side note, this specific error is not PULPino-related and it literally only takes a simple Google search copy-pasting the error to understand that it’s related to python: [https://www.google.it/search?q=Error:++Syntax+error.+Missing+parathesis+in+call+to+%27print.+Did+you+mean+print(int+tcolor&ie=UTF-8&oe=UTF-8&hl=en](https://www.google.it/search?q=Error:++Syntax+error.+Missing+parathesis+in+call+to+%27print.+Did+you+mean+print(int+tcolor&ie=UTF-8&oe=UTF-8&hl=en). Please do not pollute the issue tracker with this kind of questions, it is meant to be used for reporting bugs, querying about the hardware etc.

closed 2018-01-12 by FrancescoConti

thotaraghava, 2018-01-12:
Thank you very much

thotaraghava, 2018-01-12:
Hi @FrancescoConti ,

     My question is simple. All PULPino related RTL files are already here or  some needs to be gerated by running ./updates-ips.py file.

Thanks,

FrancescoConti, 2018-01-12:
You have already been answered in issue #150 : RTL files for IPs are downloaded (not generated) by running that script. You can also download them by hand if you really want by looking at the content of `ips_list.yml` which lists all of them. It will be painful, and take care that the version hashes (e.g. `a1dfc3e`) or tags (e.g. `tag/version1`) are correct or *nothing* will work.