pulp-platform/pulpino #155: Regarding boot_code.sv and boot_ram_wrap.sv files

thotaraghava, 2018-01-19:
Hi ,

  If I wanted to use PULPino core in place of Cortex M3 core in some application, should I use this boot loader.
  Though it is simple question, please respond.

Thanks,

FrancescoConti, 2018-01-19:
I am actually unsure what the question is.
The `boot_code.sv` is a behavioral model of a boot ROM generated by a script so that it can be used for the simulation. The actual source of the boot code is here: https://github.com/pulp-platform/pulpino/blob/master/sw/ref/crt0.boot.S

You are not bound to using specifically this boot code (even if you target PULPino), and probably if you want to take RI5CY and plug it in your own platform in place of a Cortex-M3, you will have to modify the boot code. The boot code is specific to the PULPino platform, in the sense that any boot code makes assumptions about the architecture *around* your core, not only the architecture *of* your core.

thotaraghava, 2018-01-19:
@FrancescoConti  @gautschimi , please respond.
In my applications, I have instruction ROM seperately .

FrancescoConti, 2018-01-19:
I already responded... perhaps I do not understand your question, and I do not understand what "I have instruction ROM separately".

Please rephrase your question so that it is a bit clearer...

FrancescoConti, 2018-01-20:
Given your answer to the other issue #156, I think you are using Riscy separately, not as a part of the PULPino platform. So in that case, as I said above, you will have to develop your own boot code (you can take inspiration on ours, of course, but most probably it won't be identical).

closed 2018-01-20 by FrancescoConti