pulp-platform/pulpino #156: Reg: riscv_core.sv download

thotaraghava, 2018-01-19:
HI @FrancescoConti  @gautschimi ,

    Sorry for multiple questions but I am clueless.

    As I had mentioned in other #, I have to do WINDOWS download.
    Whatever the .sv files available in this github website under rtl directory, I downloaded.
    From where can I download riscv_core.sv file and its sub-modules.

Thanks,

FrancescoConti, 2018-01-19:
I copy paste my answer to issue #151:

> You have already been answered in issue #149 : RTL files for IPs are downloaded (not generated) by running that script. You can also download them by hand if you really want by looking at the content of `ips_list.yml` which lists all of them. It will be painful, and take care that the version hashes (e.g. a1dfc3e) or tags (e.g. tag/version1) are correct or nothing will work.

I will try to be more thorough, but please note that this procedure is anyways unsupported. It is probably better to run the `update-ips.py` script on Windows and then copy stuff.

Each IP corresponds to a repository in our group on GitHub: https://github.com/pulp-platform ; e.g. for a couple of random IPs chosen from the `ips_list.yml`:
```
apb/apb_node:
  commit: pulpinov1
riscv:
  commit: pulpinov1
```
you have to download the IPs https://github.com/pulp-platform/apb_node/tree/pulpinov1 and https://github.com/pulp-platform/riscv/tree/pulpinov1. You have to place the IPs under a subdirectory `ips` in the root directory of PULPino, under a subdirectory given by the keys in the `ips_list.yml`; e.g. for the two IPs above it would be `pulpino/ips/apb/apb_node` and `pulpino/ips/riscv`.

Then you should run the `./generate-scripts.py`. If everything was correct, the build scripts related to the various IPs should be created correctly.

closed 2018-01-19 by FrancescoConti

thotaraghava, 2018-01-20:
Hi @FrancescoConti ,

     I understood everything above you had mentioned, but I am unsure about running ./generate-scripts.py.
    If I download manually all sub IPs, then is it necessary to run this command. Why I am asking this because I am facing issues in running python scripts in windows.
  I request your patience.

Thanks

FrancescoConti, 2018-01-20:
This script has to be run on Linux, to generate the simulation scripts/environment. If you cannot access the internet from your Linux machine, then you will have to download this https://github.com/pulp-platform/IPApproX in a directory in the root of PULPino named `ipstools`.

Then you should remove the lines in `generate-scripts.py` that mention downloading these tools.

thotaraghava, 2018-01-20:
That is fine. But the scripts in the link above for what.
 They are nothing to do with RTL right. I think they will be used for standalone pulpino simulation.
 If this is the case, I may not need them in my case.
Please opine.

Thanks

FrancescoConti, 2018-01-20:
I will try to answer to your question, but first a preliminary question on my side. If you don't need the "standalone" PULPino platform, can't you just download the RTL sources you need for whatever IP you require (e.g. Riscy)? I don't get it.

So I will assume you actually *do need* the full PULPino, as implied by the fact that you are asking all these questions. In that case, the `generate-scripts.py` does not create any new RTL, but it collects information from all the IPs collected in the previous step and it generates simulation scripts for Modelsim. If you want to write these scripts on your own manually you can do it of course. But 1) it will take time and effort to get them working, 2) you won't get any support here about that.

thotaraghava, 2018-01-20:
I will try to answer to your question, but first a preliminary question on my side. If you don't need the "standalone" PULPino platform, can't you just download the RTL sources you need for whatever IP you require (e.g. Riscy)? I don't get it.
>>>>My case comes under this category. Thanks for confirming. I just need pulpino RTL. I will download it.
>>>> Thanks alot for your patience.