pulp-platform/pulpino #157: xilinx_mem_8192x32 not found

bnkla, 2018-01-20:
Hi
I am trying to run the pulpino FPGA port on my zedboard, but I get following error in the "make -C pulpino clean all" step:
```
ERROR: [Synth 8-439] module 'xilinx_mem_8192x32' not found [/pulpino/rtl/sp_ram_wrap.sv:33]
WARNING: [Synth 8-693] zero replication count - replication ignored [/pulpino/rtl/sp_ram_wrap.sv:42]
ERROR: [Synth 8-285] failed synthesizing module 'sp_ram_wrap' [/pulpino/rtl/sp_ram_wrap.sv:13]
ERROR: [Synth 8-285] failed synthesizing module 'instr_ram_wrap' [/pulpino/rtl/instr_ram_wrap.sv:14]
ERROR: [Synth 8-285] failed synthesizing module 'core_region' [/pulpino/rtl/core_region.sv:15]
ERROR: [Synth 8-285] failed synthesizing module 'pulpino_top' [/pulpino/rtl/pulpino_top.sv:20]
ERROR: [Synth 8-285] failed synthesizing module 'pulpino' [/pulpino/fpga/rtl/pulpino_wrap.v:11]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:11 ; elapsed = 00:00:12 . Memory (MB): peak = 1004.422 ; gain = 224.738 ; free physical = 214 ; free virtual = 19357
---------------------------------------------------------------------------------
RTL Elaboration failed
203 Infos, 5 Warnings, 0 Critical Warnings and 7 Errors encountered.
```
make -C ips/xilinx_mem_8192x32/ clean all works without problems. Any idea?

3adawy, 2018-01-23:
I think you just need to refer to the file where the module has been instantiated using 
 ` include "{DIR}\xilinx_mem_8192x32_stub.v" ` ...
in my case it was at  "C:\pulpino\fpga\ips\xilinx_mem_8192x32\ip\xilinx_mem_8192x32_stub.v"

closed 2018-01-25 by bnkla

kingweio, 2019-04-26:
hi，@bnkla I have encountered the same problem as you. Have you finally solved this problem? How was it solved?