pulp-platform/pulpino #158: master spi stops after sending address and command

WaleedOsama, 2018-01-28:
i am trying to run SPI test , and here is my code 
```
#include <stdio.h>
#include <stdlib.h>
#include "spi.h"

int main (){

  int message_bits_no;
  int  spi_ptr[2];
  spi_ptr[0] = 10;
  spi_ptr[1] = 20 ;

  message_bits_no = 64; 

  // setup as master , and set 1 chip only , connected at CSN0
  spi_setup_master(1);

  spi_set_datalen(message_bits_no);

  spi_setup_cmd_addr(SPI_CMD_WR , 1 ,0x0000000A, 32);
  //spi_setup_dummy (2,2);
  spi_write_fifo( spi_ptr , message_bits_no);
  spi_start_transaction(SPI_CMD_WR , SPI_CSN0);
  return 0;
}
```
and this is the waveform on modelsim of signals (from top to bottom):
- master clock
- chip select 0
- mode (2 bits )
- sdo0
- sdi0

[image]

nothing happens after that , the chip select turns to 1 , till it breaks .
the spi is connected to the interface ```spi_slave spi_master()```

now i have some questions :
- the command is write , so why it does not continue to send the data ?
- does the master has to be connected to slave that will send back anything to start sending data ?because the interface does nothing back to the master

all i want is to see the data sent over spi (which in this case is 10 and 20), to further connect it with another pulpino , i did this with uart but i am stuck here . 

i think the sequence of command then address then data would benefit in case of connecting with memory , but in my case i just want array of data to be sent to slave .

any information or work around about this issue would be really appreciated . thanks

quangdaovu, 2018-01-29:
Hi,

I think I can help with something.
1. This happens because you messed up the order. The flow should be:
   1st: spi_setup_cmd_addr
   2nd: spi_set_datalen
   3rd: spi_start_transaction
   4th: spi_write_fifo

2. I don't think so.
3. SPI is 0 and GPIO is 1. It's correct.

Best,
Quang Dao.

WaleedOsama, 2018-01-29:
@quangdaovu THANK YOU ! , i made the sequence you told and it is working now .

closed 2018-01-29 by WaleedOsama

ravi02011993, 2018-03-06:
Hi, @WaleedOsama , @quangdaovu  can you tell me what changes need to be done on the tb.sv to check the output of the array? 
I did not understand what do you mean by "the spi master is connected to the interface spi_slave spi_master()". Can you explain this a little bit?

Thanks in advance

WaleedOsama, 2018-03-06:
Spi_slave is an implemented interface , if you see in tb.sv you will find it connected to spi master of pulpino , it is like a dummy slave , a place holder i guess .

To check the output of pulpino's spi master just write your code and watch spi master pins you will see the waveform

But this interface will not print spi output for you so you have to see it on waveform.. it is not like uart_bus module for example as it prints the sent message over uart.

ravi02011993, 2018-03-06:
Yes,

Thanks for the reply.
For instance, consider that I run the above code.
1. In which pin could I see the value 10 or 20 in the waveform? because sd0 is 1 bit sized. but spi_ptr is an array and it stores 64-bit data.
2.So, should I make some changes in tb?

Thanks in advance

WaleedOsama, 2018-03-06:
You do not need to make a change in tb.sv 
And to see the waveform , just watch sdo0 , spi is serial communication protocol , it sends the data on one line (and maybe 4 if you want but it is 1 by default) 
Its outer interface is clk , cs , sdo , sdi  , so this us what you can watch to see spi working.
So you will see on sdo somethong like the screenshot ,  command , address and then you will see the sent data.

ravi02011993, 2018-03-06:
Ok, Thank you. I got it now.