pulp-platform/pulpino #160: ERROS in the FPGA Synthesis

merchantf, 2018-01-31:
Hi,

I am getting followng error during synthesis.

ERROR: [Synth 8-439] module 'xilinx_mem_8192x32' not found [/net/home/merchantf/Downloads/pulpino/rtl/sp_ram_wrap.sv:33]
WARNING: [Synth 8-693] zero replication count - replication ignored [/net/home/merchantf/Downloads/pulpino/rtl/sp_ram_wrap.sv:42]
ERROR: [Synth 8-285] failed synthesizing module 'sp_ram_wrap' [/net/home/merchantf/Downloads/pulpino/rtl/sp_ram_wrap.sv:13]
ERROR: [Synth 8-285] failed synthesizing module 'instr_ram_wrap' [/net/home/merchantf/Downloads/pulpino/rtl/instr_ram_wrap.sv:14]
ERROR: [Synth 8-285] failed synthesizing module 'core_region' [/net/home/merchantf/Downloads/pulpino/rtl/core_region.sv:15]
ERROR: [Synth 8-285] failed synthesizing module 'pulpino_top' [/net/home/merchantf/Downloads/pulpino/rtl/pulpino_top.sv:20]
ERROR: [Synth 8-285] failed synthesizing module 'pulpino' [/net/home/merchantf/Downloads/pulpino/fpga/rtl/pulpino_wrap.v:11]

How do I solve these errors?

Will the emulation work without solving these errors?

I have not made any changes in the scripts that are already there. 

Help will be appreciated.

Thanks & Regards,

Farhad

merchantf, 2018-01-31:
> The error means that the command make ips to create the memory models (which are not inferred) failed in some way. So:
>
>   1.  did you download the IPs and generate the scripts by update-ips.py first? Neglecting to do this could cause the error above.
>   2.  if the scripts are correctly generated, then what is the full output of make clean ips?

Hi,

I have ran update-ips.py script already.

Full output of make clean ips (in fpga directory) is as follows:

make -C ips/xilinx_clock_manager/ clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_clock_manager'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_clock_manager.xpr
rm -rf xilinx_clock_manager.hw
rm -rf xilinx_clock_manager.data
rm -rf xilinx_clock_manager.runs
rm -rf xilinx_clock_manager.srcs
rm -rf xilinx_clock_manager.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
rm -rf .Xil
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_clock_manager'
make -C ips/xilinx_mem_8192x32/ clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_mem_8192x32'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_mem_8192x32.xpr
rm -rf xilinx_mem_8192x32.hw
rm -rf xilinx_mem_8192x32.data
rm -rf xilinx_mem_8192x32.runs
rm -rf xilinx_mem_8192x32.srcs
rm -rf xilinx_mem_8192x32.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_mem_8192x32'
make -C ips/xilinx_mem_32768x32_dp/ clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_mem_32768x32_dp'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_mem_32768x32_dp.xpr
rm -rf xilinx_mem_32768x32_dp.hw
rm -rf xilinx_mem_32768x32_dp.data
rm -rf xilinx_mem_32768x32_dp.runs
rm -rf xilinx_mem_32768x32_dp.srcs
rm -rf xilinx_mem_32768x32_dp.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_mem_32768x32_dp'
make -C ips/xilinx_fp_fma/ clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_fp_fma'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_fp_fma.xpr
rm -rf xilinx_fp_fma.hw
rm -rf xilinx_fp_fma.data
rm -rf xilinx_fp_fma.runs
rm -rf xilinx_fp_fma.srcs
rm -rf xilinx_fp_fma.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_fp_fma'
make -C pulpino clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/pulpino'
rm -rf pulpino.xpr
rm -rf pulpino.data
rm -rf pulpino.runs
rm -rf pulpino.srcs
rm -rf pulpino.sim
rm -rf pulpino.cache
rm -rf pulpino.ioplanning
rm -rf pulpino.hw
rm -rf component.xml
rm -rf vivado*
rm -rf .Xil
rm -rf xgui
rm -rf pulpino.edn
rm -rf pulpino_stub.v
rm -rf pulpino_funcsim.v
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/pulpino'
make -C pulpemu clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/pulpemu'
rm -rf pulpemu.hw
rm -rf pulpemu.xpr
rm -rf pulpemu.data
rm -rf pulpemu.runs
rm -rf pulpemu.srcs
rm -rf pulpemu.sim
rm -rf pulpemu.sdk
rm -rf pulpemu.cache
rm -rf pulpemu.ioplanning
rm -rf pulpemu.hwdef
rm -rf *.log
rm -rf ulpsoc.txt
rm -rf ulpcluster.txt
rm -rf component.xml
rm -rf vivado*
rm -rf xgui
rm -rf .Xil
rm -rf ps_clock*
rm -rf pulpemu_top*
rm -rf pulpemu.edf
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/pulpemu'
make -C sw clean
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/sw'
rm -rf u-boot-xlnx
make -C linux-xlnx clean
make[2]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/sw/linux-xlnx'
  CLEAN   .
  CLEAN   drivers/tty/vt
  CLEAN   kernel/time
  CLEAN   kernel
  CLEAN   lib
  CLEAN   usr
  CLEAN   .tmp_versions
make[2]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/sw/linux-xlnx'
rm -f linux-xlnx/.config
rm -f buildroot/.config
make -C buildroot clean
make[2]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot'
rm -rf /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/target /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/images /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/host \
    /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/build /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/staging \
    /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/legal-info /net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot/output/graphs
make[2]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/sw/buildroot'
rm -f hsi/pulpemu_top.sysdef
rm -f sd_image/*
make[1]: Leaving directory `/net/home/merchantf/Downloads/pulpino/fpga/sw'
make -C ips/xilinx_clock_manager/ clean all
make[1]: Entering directory `/net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_clock_manager'
rm -rf ip/*
mkdir -p ip
rm -rf xilinx_clock_manager.xpr
rm -rf xilinx_clock_manager.hw
rm -rf xilinx_clock_manager.data
rm -rf xilinx_clock_manager.runs
rm -rf xilinx_clock_manager.srcs
rm -rf xilinx_clock_manager.cache
rm -rf component.xml
rm -rf vivado*.jou
rm -rf vivado*.log
rm -rf vivado*.str
rm -rf xgui
rm -rf .Xil

After this, it starts with

vivado -mode batch -source tcl/run.tcl

which is re-running of synthesis.

Thanks & Regards,

Farhad

merchantf, 2018-02-01:
Any solution to this problem?

FrancescoConti, 2018-02-01:
Hi @merchantf , actually the part of the log which is really important is what happens after the clean. It is the synthesis of the memory and clocking macros, which is what has apparently failed in the problem you see above.

merchantf, 2018-02-01:
No..It has not failed. I just sent you the clean part. In a few minutes, I will upload the whole log file.

FrancescoConti, 2018-02-01:
👍

merchantf, 2018-02-01:
Hi,

I have attached two files

make all &> log_1.txt
make clean ips &> log_make_clean_ips.txt

I have following queries:

1) I am not able to get why it is not synthesizing xilinx_mem_32788x32

2) In the fpga directory I type "make all" and it performs all the tasks. I copy contents in the sd_image to the SD Card, but it does not work.

3) Do I strictly need to follow the partitioning scheme that is given in http://www.wiki.xilinx.com/Prepare+Boot+Medium

I have tried with SDK and it works fine. I am not able to understand, why I it is not logging into linux that is running on ARM core.

Any help in this will be really appreciated.

Thanks & Regards,

Farhad

👍

****** Vivado v2015.1 (64-bit)
  **** SW Build 1215546 on Mon Apr 27 19:07:21 MDT 2015
  **** IP Build 1209967 on Tue Apr 21 11:39:20 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
# if { ![info exists ::env(XILINX_PART)] } {
#   set ::env(XILINX_PART) "xc7z020clg484-1"
# }
# if { ![info exists ::env(XILINX_BOARD)] } {
#   set ::env(XILINX_BOARD) "em.avnet.com:zynq:zed:c"
# }
# set partNumber $::env(XILINX_PART)
# set boardName  $::env(XILINX_BOARD)
# set ila_name xilinx_clock_manager
# create_project $ila_name . -part $partNumber
# set_property board $boardName [current_project]
WARNING: [Common 17-599] Property 'board' is deprecated for object type 'project'. Use 'board_part'.
# create_ip -name clk_wiz -vendor xilinx.com -library ip -module_name $ila_name
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/net/sw/xilinx/Vivado/2015.1/data/ip'.
# set_property -dict [list CONFIG.INTERFACE_SELECTION {Enable_AXI} CONFIG.USE_DYN_RECONFIG {true} CONFIG.PRIM_IN_FREQ {50.000} CONFIG.PRIMARY_PORT {clk50_i} CONFIG.CLK_OUT1_PORT {clk_o} CONFIG.CLKOUT1_REQUESTED_OUT_FREQ {5.000} CONFIG.LOCKED_PORT {rst_no} CONFIG.RESET_TYPE {ACTIVE_LOW} CONFIG.RESET_PORT {rst_ni}] [get_ips xilinx_clock_manager]
WARNING: [IP_Flow 19-3374] An attempt to modify the value of disabled parameter 'RESET_TYPE' from 'ACTIVE_HIGH' to 'ACTIVE_LOW' has been ignored for IP 'xilinx_clock_manager'
# generate_target {instantiation_template} [get_files ./$ila_name.srcs/sources_1/ip/$ila_name/$ila_name.xci]
INFO: [IP_Flow 19-1686] Generating 'Instantiation Template' target for IP 'xilinx_clock_manager'...
# generate_target all [get_files  ./$ila_name.srcs/sources_1/ip/$ila_name/$ila_name.xci]
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'xilinx_clock_manager'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'xilinx_clock_manager'...
INFO: [IP_Flow 19-1686] Generating 'Implementation' target for IP 'xilinx_clock_manager'...
INFO: [IP_Flow 19-1686] Generating 'Change Log' target for IP 'xilinx_clock_manager'...
# create_ip_run [get_files -of_objects [get_fileset sources_1] ./$ila_name.srcs/sources_1/ip/$ila_name/$ila_name.xci]
# launch_run -jobs 8 ${ila_name}_synth_1
[Thu Feb  1 19:08:16 2018] Launched xilinx_clock_manager_synth_1...
Run output will be captured here: /net/home/merchantf/Downloads/pulpino/fpga/ips/xilinx_clock_manager/xilinx_clock_manager.runs/xilinx_clock_manager_synth_1/runme.log
# wait_on_run ${ila_name}_synth_1
[Thu Feb  1 19:08:16 2018] Waiting for xilinx_clock_manager_synth_1 to finish...

*** Running vivado
    with args -log xilinx_clock_manager.vds -m64 -mode batch -messageDb vivado.pb -notrace -source xilinx_clock_manager.tcl

****** Vivado v2015.1 (64-bit)
  **** SW Build 1215546 on Mon Apr 27 19:07:21 MDT 2015
  **** IP Build 1209967 on Tue Apr 21 11:39:20 MDT 2015
    ** Copyright 1986-2015 Xilinx, Inc. All Rights Reserved.
[... 709 lines cut ...]
|xilinx_clock_manager_mmcm_drp      | rom        | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp      | rom__1     | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp      | rom__2     | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp      | rom__3     | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp__GB1 | rom        | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp__GB1 | rom__4     | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp__GB1 | rom__5     | 2x39          | LUT            | 
|xilinx_clock_manager_mmcm_drp__GB1 | rom__6     | 2x39          | LUT            | 
+-----------------------------------+------------+---------------+----------------+

---------------------------------------------------------------------------------
Finished ROM, RAM, DSP and Shift Register Reporting
---------------------------------------------------------------------------------
---------------------------------------------------------------------------------
Start Area Optimization
---------------------------------------------------------------------------------
INFO: [Synth 8-3333] propagating constant 1 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][0] )
INFO: [Synth 8-3333] propagating constant 0 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][16] )
INFO: [Synth 8-3333] propagating constant 1 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][1] )
INFO: [Synth 8-3333] propagating constant 0 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][17] )
INFO: [Synth 8-3333] propagating constant 1 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][2] )
INFO: [Synth 8-3333] propagating constant 0 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][18] )
INFO: [Synth 8-3333] propagating constant 1 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][3] )
INFO: [Synth 8-3333] propagating constant 0 across sequential element (inst/mmcm_drp_insti_2/\ram_reg[23][19] )
INFO: [Synth 8-3333] propagating const

FrancescoConti, 2018-02-12:
Hi, you should attach the files in GitHub directly (not as email attachements). I can't see them.

FrancescoConti, 2018-02-25:
Closing the issue as no additional info was provided.

closed 2018-02-25 by FrancescoConti