pulp-platform/pulpino #163: Error during the finalisation of the linux image

Dracauflammenkuche, 2018-02-06:
Hello,

I'm using Ubuntu 16.04 64 bits in a virtual machine, the pulpino folder is located in an external hard drive. There was no error reported during the synthesis. 

I'm getting the following error when trying to execute make all in the pulpino/fpga directory. The error occurs after the busybox building and installing to target.

```
>>>   Finalizing target directory
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/luarocks
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/5.20.2/pod
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/5.20.2/arm-linux/CORE
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/ -name 'extralibs.ld' -print0 | xargs -0 rm -f
find: ‘/media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/’: No such file or directory
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/ -name '*.bs' -print0 | xargs -0 rm -f
find: ‘/media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/’: No such file or directory
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/ -name '.packlist' -print0 | xargs -0 rm -f
find: ‘/media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/perl5/’: No such file or directory
grep -q '^console$' /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/securetty || echo 'console' >> /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/securetty
mkdir -p /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc
echo "buildroot" > /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/hostname
/bin/sed -i -e '$a \127.0.1.1\tbuildroot' -e '/^127.0.1.1/d' /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/hosts
mkdir -p /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc
echo "Welcome to Buildroot" > /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/issue
mkdir -p /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/network/
( echo "# interface file auto-generated by buildroot"; echo ; echo "auto lo"; echo "iface lo inet loopback"; ) > /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/network/interfaces
[ -n "" ] && TARGET_GENERIC_ROOT_PASSWD_HASH=$(/media/rootux/HDD/pulpino/fpga/sw/buildroot/output/host/usr/bin/mkpasswd -m "md5" ""); /bin/sed -i -e "s,^root:[^:]*:,root:$TARGET_GENERIC_ROOT_PASSWD_HASH:," /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/shadow
ln -sf /bin/busybox /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/bin/sh
/bin/sed -i -e '/# GENERIC_SERIAL$/s~^.*#~console::respawn:/sbin/getty -L  console 0 vt100 #~' /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/inittab
/bin/sed -i -e '/^#.*-o remount,rw \/$/s~^#\+~~' /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/etc/inittab
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/include /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/aclocal \
	/media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/pkgconfig /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/pkgconfig \
	/media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib/cmake /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/cmake
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/{lib,share}/ -name '*.cmake' -print0 | xargs -0 rm -f
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/lib \( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/lib \( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/gdb
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/man /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/man
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/info /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/info
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/doc /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/doc
rm -rf /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share/gtk-doc
rmdir /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/usr/share 2>/dev/null
Makefile:547: recipe for target 'target-finalize' failed
make[2]: [target-finalize] Error 1 (ignored)
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target -type f \( -perm /111 -o -name '*.so*' \) -not \( -name 'libpthread*.so*' -o -name '*.ko' \) -print0 | xargs -0 /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/host/usr/bin/arm-xilinx-linux-gnueabi-strip --remove-section=.comment --remove-section=.note 2>/dev/null || true
if test -d /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/lib/modules; then \
	find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/lib/modules -type f -name '*.ko' -print0 | \
	xargs -0 -r /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/host/usr/bin/arm-xilinx-linux-gnueabi-strip --remove-section=.comment --remove-section=.note --strip-unneeded; fi
find /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/target/lib -type f -name 'libpthread*.so*' | \
	xargs -r /media/rootux/HDD/pulpino/fpga/sw/buildroot/output/host/usr/bin/arm-xilinx-linux-gnueabi-strip --remove-section=.comment --remove-section=.note --strip-debug
Makefile:547: recipe for target 'target-finalize' failed
make[2]: *** [target-finalize] Error 123
make[2]: Leaving directory '/media/rootux/HDD/pulpino/fpga/sw/buildroot'
Makefile:79: recipe for target 'buildroot/output/images/rootfs.tar' failed
make[1]: *** [buildroot/output/images/rootfs.tar] Error 2
make[1]: Leaving directory '/media/rootux/HDD/pulpino/fpga/sw'
Makefile:51: recipe for target 'sw' failed
make: *** [sw] Error 2
```

How can I solve this error ? Where does it come from ?
Can it come from a bad configuration of environment variables before the synthesis ?
I have not made any changes in the scripts that are already there.

Thank you in advance
Kind regards,
Dracauflammenkuche

FrancescoConti, 2018-02-06:
It seems you are in a rather special case; the error anyways happens when building the root filesystem for the Zynq board, so technically it doesn't have much to do with PULPino itself.
You probably have to go through the buildroot scripts in detail to understand what goes wrong...

closed 2018-02-16 by FrancescoConti

kingweio, 2019-04-29:
@FrancescoConti @Dracauflammenkuche Hello，I have the same problem as you, but I have two differences with you:
1. I also got the error "Makefile:294: recipe for target 'check-config' failed"
2. I can successfully generate the rootfs file system, but when I boot from the sd card, I found that I could not enter the expected "pulp", but entered the "buildroot login".

I don't know how to solve this problem. Have you finally solved this problem? How was it solved?

saranyabala92, 2021-01-11:
Hi,
I have the similar problem but it is ignored and image generated successfully. But I couldn't boot the zynq from the SD card.

make[3]: /root/pulpino/fpga/sw/buildroot/output/target/usr/sbin/sshd: Command not found
Makefile:294: recipe for target 'check-config' failed
make[3]: [check-config] Error 127 (ignored)
Makefile:547: recipe for target 'target-finalize' failed
make[2]: [target-finalize] Error 1 (ignored)
How to solve this problem?

zzlzzlwithzzl, 2021-01-12:
@kingweio have you solved this problem? i am facing the same problem right now