pulp-platform/pulpino #164: Problem while running  "cmake_configure.riscv.gcc.sh" file

3adawy, 2018-02-07:
Hey there

I'm facing a problem while running cmake_configure.riscv.gcc.sh

here is the log : 

~/pulpino/sw/build$ sh ./cmake_configure.riscv.gcc.sh
System is unknown to cmake, create:
Platform/Linux-CXX to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please send that file to cmake@www.cmake.org.
-- Loaded CMakeASMInformation - ASM support is still experimental, please report issues
-- GCC_MARCH= IMXpulpv2
-- USE_ZERO_RISCY= 0
-- RISCY_RV32F= 1
-- ZERO_RV32M= 0
-- ZERO_RV32E= 0
-- PL_NETLIST= 
-- Configuring done
CMake Error at CMakeLists.txt:115 (add_library):
  Cannot find source file "OBJECT".  Tried extensions .c .C .c++ .cc .cpp
  .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

CMake Error in apps/helloworld/CMakeLists.txt:
  Cannot find source file "$<TARGET_OBJECTS:crt0>".  Tried extensions .c .C
  .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

CMake Error in apps/Arduino_tests/GPIO_test/CMakeLists.txt:
  Cannot find source file "$<TARGET_OBJECTS:crt0>".  Tried extensions .c .C
  .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx
  
.
.
.
.
.
.

CMake Error in apps/imperio_tests/testI2C/CMakeLists.txt:
  Cannot find source file "$<TARGET_OBJECTS:crt0>".  Tried extensions .c .C
  .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

CMake Error in apps/imperio_tests/testSPIMaster/CMakeLists.txt:
  Cannot find source file "$<TARGET_OBJECTS:crt0>".  Tried extensions .c .C
  .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp .hxx .in .txx

-- Build files have been written to: /home/abdelrahman/pulpino/sw/build

rajatssr835, 2018-02-11:
I am also getting the same issue.

LilachDery, 2018-02-19:
I am also getting this problem

FrancescoConti, 2018-02-19:
The error is quite generic. Are you running a supported version of Ubuntu?

PS: I assumed Ubuntu for brevity.

LilachDery, 2018-02-21:
Yes. I have 16.04 version

FrancescoConti, 2018-02-21:
From this stackexchange answer https://stackoverflow.com/questions/24907916/cmake-how-to-properly-setup-new-toolchain-to-avoid-system-is-unknown-to-cmake it looks like it could be a toolchain related issue. Is the RISCV toolchain correctly in your path?

LilachDery, 2018-02-22:
after some attempts, this is what I have got:

-- The C compiler identification is unknown
CMake Error at CMakeLists.txt:20 (enable_language):
No CMAKE_C_COMPILER could be found.

Tell CMake where to find the compiler by setting either the environment
variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
the compiler, or to the compiler name if it is in the PATH.

 -- Configuring incomplete, errors occurred!
 See also "/home/pulp1234/Desktop/pulpino-
 master/sw/build/CMakeFiles/CMakeOutput.log".
 See also "/home/pulp1234/Desktop/pulpino-
 master/sw/build/CMakeFiles/CMakeError.log".

quangdaovu, 2018-02-22:
You don't have RISCV toolchain. Get it here, build and add it to your PATH then you're good to go.

https://github.com/pulp-platform/ri5cy_gnu_toolchain

closed 2018-02-25 by FrancescoConti

LilachDery, 2018-03-01:
Hi,
How to "add it to your PATH" ?

LilachDery, 2018-03-02:
after running the following command as mentioned in README file at ri5cy_gnu_toolchain-master/toolchain 
### Installation (Linux)

To build the Linux cross-compiler, pick an install path.  If you choose,
say, `/opt/riscv`, then add `/opt/riscv/bin` to your `PATH` now.  Then, simply
run the following command:

    ./configure --prefix=/opt/riscv
    make linux
###################################################

this what i got:

/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/src/binutils/bfd/elfxx-riscv.c:838:5: error: ‘BFD_RELOC_RISCV_REL12’ undeclared here (not in a function)
   { BFD_RELOC_RISCV_REL12, R_RISCV_REL12 },
     ^
/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/src/binutils/bfd/elfxx-riscv.c:839:5: error: ‘BFD_RELOC_RISCV_RELU5’ undeclared here (not in a function)
   { BFD_RELOC_RISCV_RELU5, R_RISCV_RELU5 },
     ^
/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/src/binutils/bfd/elfxx-riscv.c:840:5: error: ‘BFD_RELOC_RISCV_12_I’ undeclared here (not in a function)
   { BFD_RELOC_RISCV_12_I, R_RISCV_12_I },
     ^
/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/src/binutils/bfd/elfxx-riscv.c:841:5: error: ‘BFD_RELOC_RISCV_12_S’ undeclared here (not in a function)
   { BFD_RELOC_RISCV_12_S, R_RISCV_12_S },
     ^
Makefile:1607: recipe for target 'elfxx-riscv.lo' failed
make[5]: *** [elfxx-riscv.lo] Error 1
make[5]: Leaving directory '/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/build-binutils-linux/bfd'
Makefile:1649: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/build-binutils-linux/bfd'
Makefile:1134: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/build-binutils-linux/bfd'
Makefile:2600: recipe for target 'all-bfd' failed
make[2]: *** [all-bfd] Error 2
make[2]: Leaving directory '/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/build-binutils-linux'
Makefile:832: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/lilach/Desktop/ri5cy_gnu_toolchain-master/toolchain/build-binutils-linux'
Makefile:89: recipe for target 'stamps/build-binutils-linux' failed
make: *** [stamps/build-binutils-linux] Error 2

kostarnov, 2018-05-10:
I've got the same error.