pulp-platform/pulpino #165: SPI slave to process received data

WaleedOsama, 2018-02-09:
i want to know the best choice i have to make SPI slave PULPino node aware that data is completely received to begin processing it , and here is what i did.
i simulated on modelsim two PULPinos connected like this 
[image]

so what happens in the previous figure is that master can write and read data from slave PULPino memory , and in order for slave to know that master finished sending over SPI it waits for character over UART . 

but this is not what i want , i want SPI slave does not depend on another peripheral to know that master finished , like this 
[image]

so i think optimal way would be if there is some interrupt at slave to tell it that transaction finished and at some known addresses i find the address and size of received block of data . 

but i do not think this interrupt exists here , what i am thinking as closest to that is that i will define these addresses in memory and use SPI master to write in some known places in memory of slave side:

- that it is done
- address it wrote to
- size of block it wrote

and make the slave side by polling wait for done place to be set and go to see the address and size of block of data .
but this takes me to another issue , i do not know how memory management works here , but will it over write this places ?

so my questions are :

1.  is there a better way than what i suggested (better than slave polling on place)?
2.  is there some interrupt that can do what i want at slave SPI side ?
3.  will these places be over written ? , i will use a pointer and write in  memory directly 

thank you

FrancescoConti, 2018-02-12:
As both PULPino's will be running bare-metal code, there is no memory management essentially :) if you write your code carefully (you statically allocate the variables in the data memory, and you don't introduce races in your code), you shouldn't have problems.

As for a better solution, I think you can set an interrupt on GPIO - however I am not 100% sure whether this is doable in PULPino.

closed 2018-02-12 by FrancescoConti

WaleedOsama, 2018-02-12:
@FrancescoConti  yes it is doable , i just imlpemented it , thanks for your feedback , i think this is the best solution.

ravi02011993, 2018-03-15:
@WaleedOsama 
Can you tell me what are the changes required in tb to run the simulation in Modelsim for  gpio_demo example?