pulp-platform/pulpino #166: make vcompile

rajatssr835, 2018-02-11:
In build director after executing command "make vcompile"

I am getting the following error:
make: *** No rule to make target `vcompile'.  Stop.

bnkla, 2018-02-11:
did you first executed e.g. cmake_configure.riscv.gcc.sh in the build directory?

rajatssr835, 2018-02-12:
Yes I executed first the script in the build directory using the command:
source ./cmake_configure.riscv.gcc.sh

rajatssr835, 2018-02-12:
Yes I executed first the script in the build directory using the command: 
sh ./cmake_configure.riscv.gcc.sh after which I got error:
CMake Error at CMakeLists.txt:115 (add_library):
  Cannot find source file:

    OBJECT

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

FrancescoConti, 2018-02-12:
Which platform are you running on?

rajatssr835, 2018-02-12:
my machine details:
Linux bubbly1 3.5.0-54-generic #81~precise1-Ubuntu SMP Tue Jul 15 04:02:22
UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
OS details:
Distributor ID:    Ubuntu
Description:    Ubuntu 12.04.5 LTS
Release:    12.04
Codename:    precise
Cmake version: 2.8.7

FrancescoConti, 2018-02-12:
Technically CMake 2.8.7 should be supported by PULPino. However, I still strongly suspect a version compatibility issue somewhere.
I strongly suggest that you update to an operating system more recent than Ubuntu 12.04, which is 6 years old (basically an eternity in computing).
By the way, you should do it anyways because Ubuntu 12.04 does not even receive security updates any more: http://releases.ubuntu.com/12.04/

closed 2018-02-12 by FrancescoConti