pulp-platform/pulpino #167: Error in helloworld.sim

bnkla, 2018-02-11:
Hi,
I am trying to build the helloworld.vsim example. make vcompile worked:
```
--> PULPino compilation complete! 
--> Compiling work.tb... 
Compiling component:  work.tb 

--> work.tb compilation complete! 

--> PULPino platform compilation complete! 

Built target vcompile
```
But now on make helloworld.vsim I get this:
```
make helloworld.vsim
[  0%] Built target bench
[  0%] Built target crt0
[  0%] Built target string
[ 33%] Built target sys
[ 33%] Built target Arduino_core
[ 66%] Built target Arduino_separate
[ 66%] Built target helloworld.elf
[ 66%] Generating vectors/stim.txt
[ 66%] Built target helloworld.stim.txt
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
Scanning dependencies of target helloworld.vsim
[100%] Running helloworld in ModelSim
env: ‘-do’: No such file or directory
apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make:57: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim' failed
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
CMakeFiles/Makefile2:1754: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
CMakeFiles/Makefile2:1761: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:675: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2
```
Any suggestions?

FrancescoConti, 2018-02-12:
It looks like ModelSim is not in your PATH. Can you check with `which vsim`?

who93, 2018-02-12:
I have a symbolic link to vsim (ln -s /home/user/intelFPGA/17.1/modelsim_ase/bin/vsim /usr/bin/vsim) and `which vsim` returns therefore `/usr/bin/vsim`

bnkla, 2018-02-12:
I have a symbolic link to vsim (ln -s /home/user/intelFPGA/17.1/modelsim_ase/bin/vsim /usr/bin/vsim) and `which vsim` returns therefore `/usr/bin/vsim`

FrancescoConti, 2018-02-12:
Ok, that means that CMake should be able to see the `vsim` binary.
If I remember correctly, running `make VERBOSE=1 helloworld.vsim` should yield a bit more information. Can you try that?

bnkla, 2018-02-12:
here is the output:
```
make VERBOSE=1 helloworld.vsim
/usr/bin/cmake -H/home/user/Documents/pulpino/sw -B/home/user/Documents/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 helloworld.vsim
make[1]: Entering directory '/home/user/Documents/pulpino/sw/build'
/usr/bin/cmake -H/home/user/Documents/pulpino/sw -B/home/user/Documents/pulpino/sw/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/user/Documents/pulpino/sw/build/CMakeFiles 3
make -f CMakeFiles/Makefile2 apps/helloworld/CMakeFiles/helloworld.vsim.dir/all
make[2]: Entering directory '/home/user/Documents/pulpino/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/depend
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
cd /home/user/Documents/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Documents/pulpino/sw /home/user/Documents/pulpino/sw/libs/bench_lib /home/user/Documents/pulpino/sw/build /home/user/Documents/pulpino/sw/build/libs/bench_lib /home/user/Documents/pulpino/sw/build/libs/bench_lib/CMakeFiles/bench.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
make -f libs/bench_lib/CMakeFiles/bench.dir/build.make libs/bench_lib/CMakeFiles/bench.dir/build
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/bench_lib/CMakeFiles/bench.dir/build'.
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
[  0%] Built target bench
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/depend
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
cd /home/user/Documents/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Documents/pulpino/sw /home/user/Documents/pulpino/sw /home/user/Documents/pulpino/sw/build /home/user/Documents/pulpino/sw/build /home/user/Documents/pulpino/sw/build/CMakeFiles/crt0.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
make -f CMakeFiles/crt0.dir/build.make CMakeFiles/crt0.dir/build
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
make[3]: Nothing to be done for 'CMakeFiles/crt0.dir/build'.
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
[  0%] Built target crt0
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/depend
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
cd /home/user/Documents/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Documents/pulpino/sw /home/user/Documents/pulpino/sw/libs/string_lib /home/user/Documents/pulpino/sw/build /home/user/Documents/pulpino/sw/build/libs/string_lib /home/user/Documents/pulpino/sw/build/libs/string_lib/CMakeFiles/string.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
make -f libs/string_lib/CMakeFiles/string.dir/build.make libs/string_lib/CMakeFiles/string.dir/build
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/string_lib/CMakeFiles/string.dir/build'.
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
[  0%] Built target string
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/depend
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
cd /home/user/Documents/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Documents/pulpino/sw /home/user/Documents/pulpino/sw/libs/sys_lib /home/user/Documents/pulpino/sw/build /home/user/Documents/pulpino/sw/build/libs/sys_lib /home/user/Documents/pulpino/sw/build/libs/sys_lib/CMakeFiles/sys.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
make -f libs/sys_lib/CMakeFiles/sys.dir/build.make libs/sys_lib/CMakeFiles/sys.dir/build
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/sys_lib/CMakeFiles/sys.dir/build'.
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
[ 33%] Built target sys
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/depend
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
cd /home/user/Documents/pulpino/sw/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/user/Documents/pulpino/sw /home/user/Documents/pulpino/sw/libs/Arduino_lib/core_libs /home/user/Documents/pulpino/sw/build /home/user/Documents/pulpino/sw/build/libs/Arduino_lib/core_libs /home/user/Documents/pulpino/sw/build/libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
make -f libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build.make libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build
make[3]: Entering directory '/home/user/Documents/pulpino/sw/build'
make[3]: Nothing to be done for 'libs/Arduino_lib/core_libs/CMakeFiles/Arduino_core.dir/build'.
make[3]: Leaving directory '/home/user/Documents/pulpino/sw/build'
[ 33%] Built target Arduino_core
make -f libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/build.make libs/Arduino_lib/separate_libs/CMakeFiles/Arduino_separate.dir/depend
[... cut ...]
CMakeFiles/Makefile2:1754: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
make[2]: Leaving directory '/home/user/Documents/pulpino/sw/build'
CMakeFiles/Makefile2:1761: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
make[1]: Leaving directory '/home/user/Documents/pulpino/sw/build'
Makefile:675: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2
```

bnkla, 2018-02-13:
I switched now to CentOS release 6.8 (Final) 64bit and ModelSim 16.0.0.211 and its working without any problems. If somebody has same issues, here is what I did:
```
Toolchain:
git clone https://github.com/pulp-platform/ri5cy_gnu_toolchain.git
cd ri5cy_gnu_toolchain/
make

Pulpino:
git clone https://github.com/pulp-platform/pulpino.git
cd pulpino
./update-ips.py
cd sw && mkdir build
cp cmake_configure.riscv.gcc.sh build/ && cd build/
#set export PATH=${PATH}:/home/user/Documents/ri5cy_gnu_toolchain/install/bin in cmake_configure.riscv.gcc.sh
./cmake_configure.riscv.gcc.sh
cd ../../..
sed -e 's/-64//' pulpino/vsim/vcompile/rtl/vcompile_tb.sh  > pulpino/vsim/vcompile/rtl/vcompile_tb.sh.tmp
mv pulpino/vsim/vcompile/rtl/vcompile_tb.sh.tmp pulpino/vsim/vcompile/rtl/vcompile_tb.sh

sed -e 's/-64//' pulpino/sw/apps/CMakeSim.txt > pulpino/sw/apps/CMakeSim.txt.tmp
mv pulpino/sw/apps/CMakeSim.txt.tmp pulpino/sw/apps/CMakeSim.txt

sed -e 's/-64//' pulpino/vsim/tcl_files/run_debug.tcl > pulpino/vsim/tcl_files/run_debug.tcl.tmp
mv pulpino/vsim/tcl_files/run_debug.tcl.tmp pulpino/vsim/tcl_files/run_debug.tcl

sed -e 's/-64//' pulpino/vsim/tcl_files/run_mem_dpi.tcl > pulpino/vsim/tcl_files/run_mem_dpi.tcl.tmp
mv pulpino/vsim/tcl_files/run_mem_dpi.tcl.tmp pulpino/vsim/tcl_files/run_mem_dpi.tcl

sed -e 's/-64//' pulpino/vsim/tcl_files/run_memdpi.tcl > pulpino/vsim/tcl_files/run_memdpi.tcl.tmp
mv pulpino/vsim/tcl_files/run_memdpi.tcl.tmp pulpino/vsim/tcl_files/run_memdpi.tcl

sed -e 's/-64//' pulpino/vsim/tcl_files/run_spi.tcl > pulpino/vsim/tcl_files/run_spi.tcl.tmp
mv pulpino/vsim/tcl_files/run_spi.tcl.tmp pulpino/vsim/tcl_files/run_spi.tcl

sed -e 's/-64//' pulpino/vsim/tcl_files/run.tcl > pulpino/vsim/tcl_files/run.tcl.tmp
mv pulpino/vsim/tcl_files/run.tcl.tmp pulpino/vsim/tcl_files/run.tcl

cd pulpino/sw/build
make vcompile
make helloworld.vsim
```

closed 2018-02-13 by bnkla

wooolf0455, 2018-12-12:
I have some questions like yours ,could you help me ?#264

tresafebina, 2021-01-21:
Error while running the helloworld.vsim
 #make helloworld.vsim
[  0%] Built target bench
[  0%] Built target crt0
[  0%] Built target string
[ 33%] Built target sys
[ 33%] Built target Arduino_core
[ 66%] Built target Arduino_separate
[ 66%] Built target helloworld.elf
[ 66%] Generating vectors/stim.txt
[ 66%] Built target helloworld.stim.txt
[100%] Generating modelsim.ini
[100%] Built target helloworld.links
[100%] Built target helloworld.slm.cmd
[100%] Running helloworld in ModelSim
Failed to open executable /root/pulpino/altera/16.0/modelsim_ase/linuxaloem/../linux_x86_64pe/vish in execute mode needed for the option -64.
execv: No such file or directory
** Fatal: Unable to exec the GUI /root/pulpino/altera/16.0/modelsim_ase/linuxaloem/../linux_x86_64pe/vish.
apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make:57: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim' failed
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 1
CMakeFiles/Makefile2:1759: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
CMakeFiles/Makefile2:1766: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:675: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2

Kindly look into this error.
Thanks in advance.

shivanandchidri, 2021-01-22:
while running the helloworld.vsim

pulp/sw/app/cmakesim.txt
remove -64

Nagatejapallapu, 2021-09-06:
env: ‘-do’: No such file or directory
apps/helloworld/CMakeFiles/helloworld.vsim.dir/build.make:70: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim' failed
make[3]: *** [apps/helloworld/CMakeFiles/helloworld.vsim] Error 127
CMakeFiles/Makefile2:4720: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/all' failed
make[2]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/all] Error 2
CMakeFiles/Makefile2:4727: recipe for target 'apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule' failed
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.vsim.dir/rule] Error 2
Makefile:680: recipe for target 'helloworld.vsim' failed
make: *** [helloworld.vsim] Error 2