pulp-platform/pulpino #168: error ninja vcompile

lfernand4, 2018-02-13:
Hey,
I am trying to build the "ninja vcompile" :

[image]

Any suggestions?

zarubaf, 2018-02-13:
It seems that you don't have ModelSim installed (or in your path): `vlib` command not found.

lfernand4, 2018-02-13:
What can i do in order to use verilator ?

lfernand4, 2018-02-13:
In the README from Jun 2017 we can found :
[image]

zarubaf, 2018-02-13:
We currently do not support Verilator with PULPino. You are more then welcome though to add support for it and make a PR ;-)

closed 2018-02-13 by zarubaf