pulp-platform/pulpino #170: problem in running cmake_configure.riscv.gcc.sh

thotaraghava, 2018-02-15:
Hi @FrancescoConti ,

     I am getting below error while running it.

CMake Error at libs/string_lib/CMakeLists.txt:12 (add_library):
Cannot find source file:
inc/string_lib.h
Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

  Where can I find these files.

Thanks,
Raghav

thotaraghava, 2018-02-15:
As per the README.md file,  > make vcompile compiles RTL. Is n't it?

WaleedOsama, 2018-02-15:
put the path to folder of modelsim which contains vlog in PATH , then try it again , note that you run this command "make vcompile" , in same directory you ran the script in

thotaraghava, 2018-02-15:
Yes, I am running this command in /sw/build/ path only.
I will try.
Thanks,

FrancescoConti, 2018-02-15:
The `vsim` command must be in your [`PATH` environment variable ](https://superuser.com/questions/284342/what-are-path-and-other-environment-variables-and-how-can-i-set-or-use-them).
To append the ModelSim path to the `PATH` variable, you can use

```
export PATH=/path/to/vsim/bin:$PATH
```
assuming your shell is Bash.

thotaraghava, 2018-02-16:
@FrancescoConti ,
        I understand as below.
        I should run below command in linux terminal.
               export PATH=/path/to/vsim/bin:$PATH
        Can you guide me, how to find /path/to/vsim/ here.

Thanks

thotaraghava, 2018-02-16:
Hi @FrancescoConti ,

      We do not use Modelsim here, rather we use ncSim. Can we change script for this.
      Please guide me in this regard.

Thanks,

FrancescoConti, 2018-02-16:
Why did you have us lose our time in answering questions on ModelSim if you don't use it in the end???

Anyways. We **do not support** NCSim in PULPino. We know it would work because other people have done it, butyou have to

closed 2018-02-16 by FrancescoConti

lfernand4, 2018-02-16:
thotaraghava, I want to use NCSim to, if you found how to do it, explain me.

thotaraghava, 2018-02-16:
Hi @FrancescoConti ,

     My interest is riscv only (not zeroriscy), so I should run 'cmake_configure.riscv.gcc.sh' only right in /sw/build.

     I ran only this and when I am  trying to run 'make helloworld', I am getting below errors.
Building C object libs/sys_lib/CMakeFiles/sys.dir/src/exceptions.c.o
cc1: error: unrecognized command line option "-fdiagnostics-color=always
/sw/libs/sys_lib/src/exceptions.c:1: error: bad value (RV32I) for -march= switch
/sw/libs/sys_lib/src/exceptions.c:1: error: bad value (RV32I) for -mtune= switch
make[3]: *** [libs/sys_lib/CMakeFiles/sys.dir/src/exceptions.c.o] Error 1
make[2]: *** [libs/sys_lib/CMakeFiles/sys.dir/all] Error 2
make[1]: *** [apps/helloworld/CMakeFiles/helloworld.dir/rule] Error 2

Can you look into this.

Thanks

thotaraghava, 2018-02-16:
Did I miss any settings in cmake_configure_riscy.gcc.sh.

Quick response would be highly appreciated

quangdaovu, 2018-02-16:
All the details you need is on the README file. So PLEASE read it carefully before make a question, the amount time that it takes to ask a question is enough for you to read it one time. 

This time, your problem lies on the very first part of README. 

**Requirements**
PULPino has the following requirements

riscv-toolchain, specifically you need riscv32-unknown-elf-gcc compiler and friends. There are two choices for this toolchain: Either using the official RISC-V toolchain supported by Berkeley or the custom RISC-V toolchain from ETH. The ETH versions supports all the ISA extensions that were incorporated into the RI5CY core as well as the reduced base instruction set for zero-riscy. Please make sure you are using the newlib version of the toolchain.

thotaraghava, 2018-02-17:
One thing, I am missing in this platform.
After installing ri5cy_gnu_toolchain, how this should be linked to pulpino.
May be I am not understanding this part thoroughly.

Thanks,

quangdaovu, 2018-02-17:
Yet again your binary folder of compiled toolchain must be on your PATH.

thotaraghava, 2018-02-19:
Thanks for responding.
In PATH means, how. Can you please elaborate.
@FrancescoConti , could you please reopen this ticket.

Thanks,

FrancescoConti, 2018-02-19:
http://lmgtfy.com/?q=what+is+PATH+on+Linux

quangdaovu, 2018-02-19:
I suggest you take a simple course about Linux first.
And as Francesco already answered 4 days ago:

export PATH=/path/to/built_toolchain/bin:$PATH

thotaraghava, 2018-02-19:
Ok, ok. I understood. :)

thotaraghava, 2018-02-19:
Hi all,

  I am still in the process of installing toolChain and hence, I did not reach to the level to experiment.. Yet, I wanted to make following questions ahead.

      Q1. make vcompile will compile the RTL + simulates and creates waveform dumps. README says just compiles. Please confirm.
      Q2. waveform dumps will be created in /vsim/waves/. Is it ?

thotaraghava, 2018-02-20:
Hi all,

   Is this should be in my PATH.
  /PATH/ri5cy_gnu_toolchain.

   Under this, I have toolchain and build directories also.

Thanks

quangdaovu, 2018-02-20:
Have you ever look at Francesco's previous answer about PATH? PATH is not a directory. Please study about it first.

Then, considering your ri5cy_gnu_toolchain directory is /somewhere/ri5cy_gnu_toolchain/, and of course you correctly built it, then you should include this on your PATH: /somewhere/ri5cy_gnu_toolchain/install/bin

I don't mean to be rude, but you really have to study before making question.

thotaraghava, 2018-02-20:
I have installed ri5cy_gnu_toolchain in some other machine and copied to my working (which does not have internet) and running.
Should I run "make" again  here, to make sure installation proper.
 Its getting halted with some error.

Thanks

thotaraghava, 2018-02-20:
Even after setting /somewhere/ri5cy_gnu_toolchain/install/bin in my PATH, I am still getting the above errors. Any clue ? I am clueless here.

quangdaovu, 2018-02-20:
Are you really put **somewhere** there??? Could you give me the output of 
`echo $PATH`

thotaraghava, 2018-02-20:
path1:path2:/user/<org>/USERS/raghav.thota/ri5cy_gnu_toolchain/install/bin

quangdaovu, 2018-02-20:
I have no idea how could your system works with that kind of PATH. I'm speechless.

FrancescoConti, 2018-02-20:
@quangdaovu thanks!!!!!!!!!!

@thotaraghava , your setup is clearly completely different from any "normal" supported Linux system. I mean, not only you have a system where you cannot access the internet from Linux, but also a strongly customized Linux environment, very different from "normal" setups. This means that you are likely working in a big industrial organization and you have a system administrator. **Ask for his help**. It's impossible for us to help you solve problems for such an environment.

thotaraghava, 2018-02-20:
Ok, thanks, I will. Let me ask one question before that.
When I am running cmake_configure_riscv.gcc.sh, I have got below message

CMake Error: your C compiler: "CMAKE_C_COMPILER-NOTFOUND" was not found Please set CMAKE_C_COMPILER to a valid compiler path or name

I guess, I need to set cmake compiler in path variable. Is not it.

Thanks,

thotaraghava, 2018-02-20:
Hi all,

   After running cmake_configure_riscv.gcc.sh, I am running "make helloworld" for generating hex file.
   Correct me if I am wrong.

Thanks,

thotaraghava, 2018-02-20:
Any clue will be highly appreciated.

thotaraghava, 2018-02-20:
Hi,

  My cmake_configure.riscv.gcc.sh settings are as below.

TARGET_C_FLAGS="-O3 -g"
#ratands TARGET_C_FLAGS="-O3 -m32 -g"
#TARGET_C_FLAGS="-O2 -g -falign-functions=16  -funroll-all-loops"
RVC=0
USE_ZERO_RISCY=0
RISCY_RV32F=0
ZERO_RV32M=0
ZERO_RV32E=0
GCC_MARCH="IMXpulpv2"
ARDUINO_LIB=1

I am interested in riscv core (4stage).
Can you please confirm my above settings.

Thanks,

thotaraghava, 2018-02-20:
How USE_RISCY variable in sw/CMakeLists.txt will be decided.
I guess, somehow its taking 0.
I hope you people can answer this.

thotaraghava, 2018-02-21:
Any clue is highly appreciated :)

quangdaovu, 2018-02-21:
For the first question, everything you need is listed in README section. **Read it**. The following part comes from sw's README.

> CMake uses out-of-source builds which means you will need a separate build folder for the software, e.g. `build`
>     `mkdir build`
> Then switch to the build folder and copy the cmake template configuration script there which resides in the `sw` folder. The name of template follows the following naming scheme: cmake-configure.{or1k/riscv}.{gcc/llvm}.sh Choose, copy, modify and then execute this script. It will setup the build
> environment for you.
> 

So, cmake_configure.riscv.gcc.sh means it'll setup for RISCY core. 

For the second question, According to cmake_configure.riscv.gcc.sh, the first part it sets the variables value and the last part of it calls cmake with these values. Note that non-included parameters will be 0. So, if you're using cmake_configure.riscv.gcc.sh then USE_RISCY will be 0.

I suggest again that you have to study. It's the third time already. If you have time to ask a question and wait for others to answer, why can you just give a little more time to learn? You can open cmake_configure.riscv.gcc.sh to copy its content (in first question), why didn't you read all the file? Why didn't you try to google "HOW CMAKE WORKS"? To be honest, I've lost all of my willing to help. Good luck on your proj.

thotaraghava, 2018-02-21:
thanks @quangdaovu

thotaraghava, 2018-02-22:
To be honest, I strictly followed the README file and compiling (make helloworld).
I am still stuck with below error.

cc1: error: unrecognized command line option "-fdiagnostics-color=always"
/user/USERS/raghav.thota/pulpino_raghav/sw/libs/bench_lib/src/bench.c:1: error: bad value (RV32I) for -march= switch

My cmake version :- cmake version 2.8.12.2 , README file mentioned as below.
""CMake >= 2.8.0, versions greater than 3.1.0 recommended due to support for ninja""
To some extent I suspect my cmake version.

    gcc version        :-gcc (GCC) 4.4.7 20120313 (Red Hat 4.4.7-11)
And I am using the toolchain -> https://github.com/pulp-platform/ri5cy_gnu_toolchain

I am unable to corner out the where exactly issue.

Regards,

thotaraghava, 2018-02-22:
HI all,

    What is the hex file name and extension, which will be generated after compiling "make helloworld" and its path. 
   Somebody generated and given the database. I am looking for this file.

Thanks

thotaraghava, 2018-02-22:
flash_stim.slm
I2_ram.slm
I2_ram_cut0_hi.slm
I2_ram_cut0_lo.slm
I2_stim.slm
spi_stim.txt
tcdm_bank0.slm.

 Which files of these are considered for simulation.

Regards,
Raghav

thotaraghava, 2018-02-22:
Hello @FrancescoConti ,

    I may need information about this.
Tahnks

thotaraghava, 2018-02-22:
Hello anybody,

     I am clueless on this. Can someone break the silence.
Thanks in advance,

FrancescoConti, 2018-02-22:
No one of that files is a ``simulator'' and if you ask this, it means that you have no idea of what a simulator is. ModelSim is a simulator, QuestaSim is a simulator, Cadence Incisive is a simulator, Synopsys VCS is a simulator. Those files host **stimuli** generated by the SDK when you compile and run your program, **as stated in the README**.

FrancescoConti, 2018-02-22:
@thotaraghava it is clear to me you are proceeding blindly without much idea of what you are doing. I really suggest you do your own personal experiments and ***study, study, study***. PULPino is not a particularly complicated system, but it seems it is well beyond your current knowledge of how to handle an RTL platform.

As a person who has followed several student projects, some of them good others less, let me give you one piece of advice: *if you don't spend a few days on a problem **completely on your own**, you won't learn*. I personally learned that the hard way, and I think this is true for 99% of engineers in the world (at least those in this field). In real engineering ***problems are hard*** and most of the time if you're the person on a problem, it means that ***you*** are the most competent and skilled person that there is for that specific problem - so there's literally nobody that can solve the thing for you. Take these as occasions to grow your problem-solving skills.

thotaraghava, 2018-02-22:
absolutely brilliant message.

thotaraghava, 2018-02-22:
My intention was to ask like this
  >> Which files of these are considered as s"t"imulators for simulation  (this sentence is from my previous post)
        Typing mistake ("t" was missed).
        But its good that I got confirmed and got some good message.

Thanks

thotaraghava, 2018-02-23:
I tried  my best to find out which file of the below files should be used as stimuli.

flash_stim.slm
I2_ram.slm
I2_ram_cut0_hi.slm
I2_ram_cut0_lo.slm
I2_stim.slm
spi_stim.txt
tcdm_bank0.slm

Not able to find out. Any clue on where can I get.

Regards,

gautschimi, 2018-02-23:
there are different methods on booting in pulpino.

either you preload all memories (fast) or you load the memory content 
over spi (slow)

option 1: preload:
l2_ram.slm = content of the instruction memory
tcdm_bank0.slm = content of the data memory

option 2: load over spi:
spi_stim.txt is used

you will see in tb.sv and its packages which stimuli is used under what 
conditions

thotaraghava, 2018-02-23:
@gautschimi , Excellent !!!.

Let me work on myself .

Thanks,

thotaraghava, 2018-02-24:
Hi @gautschimi ,

      For "helloworld" testcase, I generated slm files .
      Interestingly, I could not see any data in l2_ram.slm, but as expected, data exists in tcdm_bank0.slm.
      I came to know that, I should input scatter file during compilation which I did not.
      Was this the reason for the situation.
      And if I have to mention scatter file, where I need to.

      PS: This is my first processor related project.

Thanks

quangdaovu, 2018-02-24:
> you will see in tb.sv and its packages which stimuli is used under what conditions
I doubt that you really have no intent to read source code. 

gautschimi already mentioned about tb.sv and its package, and the method that tb used to preload memory. You should have read and trace it back.

https://github.com/pulp-platform/pulpino/blob/0cdfddd93531acb5291d8ed4a449a5173a02826f/tb/tb.sv#L246-L250

If PRELOAD then execute mem_preload(), which lies on tb_mem_pkg.sv. And from there you will be able to find this:

https://github.com/pulp-platform/pulpino/blob/05f0dbeeb8823e215a8fc526145af24ad18417ab/tb/tb_mem_pkg.sv#L38-L48

The files that you need are slm_files/l2_stim.slm for instruction mem and slm_files/tcdm_bank0.slm for data mem.

thotaraghava, 2018-02-24:
HI all,

   In utils/s19toslm.py, line no-181, I wanted to print just data only (no address required) and that too in 32 bit binary format.
I tried as below, but not working.
  l2_stim.write("%32B\n" %(data)).

  Could you please correct me.

Regards

thotaraghava, 2018-02-27:
HI all,

    I am able to generate both instr and data files.
    In my system, we have only one ROM in which both instruction and data should sit.
    So, I put instruction data on top of the ROM and data just below that. 
    What could be the value of "boot_addr_i" of core_region module to drive in this case.
    
    Please correct me if something is wrong in this procedure.

Regards,

thotaraghava, 2018-02-27:
Should it be 0x0000_0000.