pulp-platform/pulpino #171: My case: Replace cortex-m3 with PULPino processor and run system level simulations

thotaraghava, 2018-02-15:
Hi @FrancescoConti ,

    As I had mentioned in earlier tickets about my CASE, I replaced Cortex-M3 RTL in my SYSTEM with just PULPino RTL and trying to simulate in system level.
     In this case, with respect to software, what data from stand alone PULPino, needs to be re-used in my case and simulate.
   My requirement is to run PULPino processor in my system and observe waveform only.
   To make processor run, apart from RTL, what else I need to use and HOW?
  I request your patience.

Thanks,
Raghav

thotaraghava, 2018-02-16:
Hi @FrancescoConti  and @gautschimi ,

   Any comment on this.

Thanks

FrancescoConti, 2018-02-16:
Riscy supports the RVI32MC ISA. Apart from that constraint, which kind of software you run on it depends entirely on your system. I have no idea what that is in your case.

closed 2018-02-16 by FrancescoConti