pulp-platform/pulpino #172: FPGA Debug

bnkla, 2018-02-16:
Hi
I am running the pulpino on the ZedBoard and I wonder, if there is any possibility to debug the core? I successfully simulated my changes in the design with ModelSim and executed a sample program, however the program is not running when executing on the FPGA board.
Best wishes

bnkla, 2018-02-21:
hm I think the debug interface isn't working any more (see #148)?

quangdaovu, 2018-02-21:
It still works. Provided debug_bridge wowrks if you're working on Zedboard, you'll need to modify things if it's pure FPGA (other board without PS).

bnkla, 2018-02-21:
I created the executable debug_bridge with `make pulpino=1`, uploaded it to the ZedBoard and tried to execute it with ./debug_bridge, but I get this:
```
FPGA SPI device opened!
Unknown platform, trying auto-detect
Detected PULP
No debug interface available! Exiting now
```

quangdaovu, 2018-02-21:
I don't work with Zedboard, so I can't help with this. I have my PULPino running on Arty board, and it doesn't have PS so I manage to debug software by develop another version of debug_bridge. If you have any JTAG cable (mine is Olimex ARM-USB-OCD-H), you can give it a try with this:

https://github.com/quangdaovu/pulpino_jtag_bridge

Just uploaded it yesterday and I don't have time to fill out the README section. You will need libftdi1 (https://www.intra2net.com/en/developer/libftdi/download.php), build it, put it somewhere and change the location on Makefile, run make and it will be the same as debug_bridge.

closed 2018-02-24 by bnkla