pulp-platform/pulpino #173: Error while running SPI load

merchantf, 2018-02-21:
./spiload -t20 spi_stim.txt 
Device has been reset
Sending block addr 00000000 with 256 entries
Sending block addr 000003FC with 256 entries
Sending block addr 000007F8 with 256 entries
Sending block addr 00000BF4 with 256 entries
Sending block addr 00000FF0 with 256 entries
Sending block addr 000013EC with 237 entries
Sending block addr 00100000 with 256 entries
Sending block addr 001003FC with 73 entries
Starting device
Waiting for EOC...
Timeout reached!
Stopped after 20.264

Can someone help me with this?

Farhad

who93, 2018-02-21:
what is the program supposed to do? If its the led_demo example, the leds on the board should be flashing, if its an example with uart, a text should be displayed before timeout reached (if the timeout is long enough)

merchantf, 2018-02-21:
It is led_demo example. I also tried gpio_demo, but what every I do there isone LED glowing on the board which is LD0 (T22). I have using zeroriscy core and also using corresponding script which is cmake_configure.zeroriscy.gcc.sh to generate the configuration. I have set USE_ZERO_RISCY and ZERO_RV32M to 1.

who93, 2018-02-21:
are you using the ZedBoard? I tried the example on this board with RISCY and it worked OOTB. Have you tried to use the spi_stim.txt file as an input for ModelSim to simulate it?

merchantf, 2018-02-21:
I am using Zedboard. I simulated in modelsim with zero riscy and it seems it does not work. Then, I tried for RISCY in modelsim and it worked in simulations.  Now, I am synthesizing it for RISCY and will put it on FPGA once it is done. Will get back to you with the update once the synthesis is over.

merchantf, 2018-02-21:
It is working fine. I have a small question. Is RISCY core running FreeRTOS ?

FrancescoConti, 2018-02-25:
Hi @merchantf . There is a port performed some time ago by @zarubaf, but it's on an old version of the platform. I don't think we support FreeRTOS at the moment - however there is no technical issue in porting it here, it's only a matter of interest (or lack of) on our side.

closed 2018-02-25 by FrancescoConti