pulp-platform/pulpino #175: testGPIO.c is not working

ravi02011993, 2018-02-22:
[image]

Hii , 
I am very new to this platform. I ran the hello world and its working fine. But when I run testGPIO.c it shows error as attached in the picture.

Can somebody help?

quangdaovu, 2018-02-22:
You should include spi.h to testGPIO.c.

`#include "spi.h"`

ravi02011993, 2018-02-22:
Hi, I tried that. Now i am getting an error as below.
error: 'FUNC_EXT2' undeclared (first use in this function)
   set_pin_function(PIN_MSPI_CSN1, FUNC_EXT2);

quangdaovu, 2018-02-22:
Eventhough FUNC_EXT2 doesn't exist but I believe its value is 0. You can just replace it with 0, or add it to sw/libs/sys_lib/inc/gpio.h, under FUNC_GPIO.

closed 2018-02-22 by ravi02011993