pulp-platform/pulpino #176: Problem running  cmake_configure.zeroriscy.gcc.sh

SumanthKK, 2018-02-22:
Hi,

To evaluate the pulp processor cores for our projects, I am trying to generate the PULPino zeroriscy without any customization (as I do not have access to ETH compiler) processor core by running the "cmake_configure.zeroriscy.gcc.sh" as per the instructions mentioned on your GitHub page. I am getting an error as the compiler broken, below is the error message snippet for your reference. 

[image]

 I have installed the toolchain suite correctly by running the below commands as mentioned in the RISC-V software toolchain page. I am not sure if the error is because of the incorrect tool installation or with zeroriscy configure script. Can you please take a look at the below tool installation commands and let me know if I missed anything or added anything extra. 
	
```
export TOP=$(pwd)
git clone https://github.com/riscv/riscv-tools.git
cd $TOP/riscv-tools
git submodule update --init --recursive
export RISCV=$TOP/riscv
export PATH=$PATH:$RISCV/bin
Installed the g++ compiler at it is missing
Execute the ./build-spike-only.sh command, this will install the risc-V components.
cd $TOP/riscv-tools/
./configure --prefix=$RISCV --with-arch=rv32gc --with-abi=ilp32d
make linux
make report linux 
```

Let me know if you need any further information. I appreciate your help.

Thanks,
Sumanth.

quangdaovu, 2018-02-22:
You can get ETH toolchain here:

https://github.com/pulp-platform/ri5cy_gnu_toolchain

Remember to build it with make ZERORISCY=1.

SumanthKK, 2018-02-22:
I have already installed this toolchain as well. But I am getting the same error. Not sure if we need to anything to the linux PATH, as I see nothing has been mentioned on the page.

quangdaovu, 2018-02-22:
Make sure that your PATH contains only ETH path.

SumanthKK, 2018-02-22:
Thanks much for the prompt response! 

Whats the ETH you want me to set? I remember the webpage says that ETH compiler is not needed for zeroriscy core and also the compiler is not released to the public. Can you please elaborate on this?

Thanks!

quangdaovu, 2018-02-22:
riscv-toolchain, specifically you need riscv32-unknown-elf-gcc compiler and friends. There are two choices for this toolchain: Either using the official RISC-V toolchain supported by Berkeley or the custom RISC-V toolchain from ETH. The ETH versions supports all the ISA extensions that were incorporated into the RI5CY core as well as the reduced base instruction set for zero-riscy. Please make sure you are using the newlib version of the toolchain.

Actually it's needed. And I don't really get your question. What do you mean by set ETH?

SumanthKK, 2018-02-22:
Sorry for the confusion! I meant by the ETH compiler path. There are a lot of sub-directories inside the ri5cy_gnu_toolchain  directory, which directory path should I add to the Linux path? Is it the one highlighted in the below figure?

[image]

From your comment, what I understand is that it's better to install only the ri5cy_gnu_toolchain tool suite. Am I correct?

quangdaovu, 2018-02-22:
Yep you're correct. Remove whatever related to official RISCV toolchain in your PATH, leave only ETH things. 
For ETH toolchain, if you follow the guide on README then you should include ~/R_work/pulpino/ri5cy_gnu_toolchain/install/bin

SumanthKK, 2018-02-22:
Great! It worked! I am assuming that I can use any of the cores by using your tail chain and setting respective variables, correct me if I am wrong.
And also, I was with the idea that the scripts generate the zeroriscy core as well. Do I need to clone the core separately?

quangdaovu, 2018-02-23:
I haven't try other cores, mine is ri5cy core and its corresponding toolchain works fine with zeroriscy script, even with compile helloworld (of course my custom "helloworld"). So, if you can then clone the toolchain with different build option is the best, but if you're too lazy to wait (for the build process) then I bet you can give a try.

FrancescoConti, 2018-02-25:
@SumanthKK if you refer to the `update-ips.py` / `generate-scripts.py`, those are used to download the RTL both for RISCY and zero-riscy. BTW, I see no reason for which the zero-riscy toolchain should not work on RISCY - zero-riscy supports a strict subset of instructions AFAIK.

closed 2018-02-25 by FrancescoConti

SumanthKK, 2018-02-28:
@FrancescoConti , The GitHub PULPino repository page mentions that installation of the other cores except Zero-riscy needs the ETH compiler. So I was wondering if the zero-risky tool chain supports the other cores. Can you please clarify me if the ETH compiler is released to public and part of your tool-chain?