pulp-platform/pulpino #177: Understanding boot code

mrproton, 2018-02-22:
I want to understand, how the RISCY core boots on the FPGA. For this reason I am examining the helloworld example with modelsim. when generating the spi_stim.txt file for helloworld, the first line is 00000000_00000013 - this means first instruction is NOP. But when I am looking in modelsim, the first instruction is 00C0006F, so at 0x80, why it is not starting at address 0x00?

How is booting done on the real FPGA platform? With spiload, I am loading spi_stim.txt to the RISCV core. With an reset and fetch enable triggered by spiload, the core starts fetching the first instruction from 0x80? is there any bootcode used? when yes, when and why is it loaded?

FrancescoConti, 2018-02-22:
Hi @mrproton, you can find the boot code here https://github.com/pulp-platform/pulpino/blob/master/sw/ref/crt0.riscv.S . The corresponding linker script is here https://github.com/pulp-platform/pulpino/blob/master/sw/ref/link.common.ld . Honestly I don't know why the reset vector is located at 0x80, probably @haugoug does.

quangdaovu, 2018-02-23:
There is different between normal application and boot code. E.g. in helloworld CMakeLists.txt, add_application is used while boot_code's CMakelist uses add_boot_code. The difference between them are:
- Normal application takes https://github.com/pulp-platform/pulpino/blob/master/sw/ref/crt0.riscv.S as its entry/initialization routine; uses https://github.com/pulp-platform/pulpino/blob/master/sw/ref/link.common.ld as its linker.
- Boot_code takes https://github.com/pulp-platform/pulpino/blob/master/sw/ref/crt0.boot.S as its entry/initialization routine; uses https://github.com/pulp-platform/pulpino/blob/master/sw/ref/link.boot.ld as its linker.

Most notable point is their memory region. Normal application instructions lies on Instruction RAM (start from 0x0) while boot_code's instruction lies on ROM (start from 0x8000). So in your case, you was exercise with normal app (helloworld), not boot_code.

If you look at https://github.com/pulp-platform/pulpino/blob/master/tb/tb.sv, you can see that the boot_address is overwriten via JTAG at line 243 if MEMLOAD != STANDALONE. The first executed instruction will always be the one at address boot_address + 0x80 (as Francesco points out 0x80 is the core reset vector). By default MEMLOAD is PRELOAD, so in order to test boot_code, you can change MEMLOAD on https://github.com/pulp-platform/pulpino/blob/master/vsim/tcl_files/run.tcl, from PRELOAD to STANDALONE. By doing this, the default value of boot_address is 0x8000, which is pointed to start address of ROM (set in RTL code).

On real FPGA platform, the core always tries to boot from ROM other than directly boot from instruction RAM (since that RAM is empty at power on. spiload overwrite boot_address via SPI.

haugoug, 2018-02-23:
When the core starts executing the first instruction it jumps to the reset exception vector which is located at 0x80.
The reason is that there are the entries for all interrupts between 0 and 0x80 (for 32 interrupts).
I must admit this was not very clever to do it this way, we kept it for historical reasons. We will probably change that in the future to be more compliant with the riscv privileged specifications.

closed 2018-02-25 by FrancescoConti

mrproton, 2018-02-25:
when using an FPGA platform, spiload loads the stim file to the instruction memory, resets the core and sets the boot address to boot from instruction ram? then, the spi_stim file from 0x00000000 is read out and executed?

so, in this scenario no boot code is used?

haugoug, 2018-02-25:
yes indeed